verilog-to-routing/vtr-verilog-to-routing #258: Specifying a subset of pins in a pin location and causes a segfault

mithro, 2017-12-05:
I'm trying to describe the dual-port ram in the iCE40 (as part of #245). The dual port ram takes up two vertical tiles in the design which you can see described here;
[image]
```xml
  <pb_type name="RAM" height="2">
   <!-- Read port -->
   <output name="RDATA" num_pins="16" equivalent="false"/>
   <input  name="RADDR" num_pins="11" equivalent="false"/>
   <input  name="RE"    num_pins="1"  equivalent="false"/>
   <input  name="RCLKE" num_pins="1"  equivalent="false"/>
   <clock  name="RCLK"  num_pins="1"  equivalent="false"/>
   <!-- Write port -->
   <input  name="WDATA" num_pins="16" equivalent="false"/>
   <input  name="MASK"  num_pins="16" equivalent="false"/>
   <input  name="WADDR" num_pins="11" equivalent="false"/>
   <input  name="WE"    num_pins="1"  equivalent="false"/>
   <input  name="WCLKE" num_pins="1"  equivalent="false"/>
   <clock  name="WCLK"  num_pins="1"  equivalent="false"/>
  </pb_type>
```

The [IceStorm docs for the ram tile](http://www.clifford.at/icestorm/ram_tile.html) then describes the "pin layout" as shown here;
[image]

When I describe the architecture with the following, I get a segfault;
```xml
   <interconnect>
    <!-- Read port -->
    <direct name="RDATA" input="SB_RAM.RDATA[15:0]" output="RAM.RDATA[15:0]"/>
    <direct name="RADDR" input="RAM.RADDR[7:0]"  output="SB_RAM.RADDR[7:0]" />
    <direct name="RE"    input="RAM.RE"          output="SB_RAM.RE" />
    <direct name="RCLKE" input="RAM.RCLKE"       output="SB_RAM.RCLKE" />
    <direct name="RCLK"  input="RAM.RCLK"        output="SB_RAM.RCLK" />
    <!-- Write port -->
    <direct name="WDATA" input="RAM.WDATA[15:0]" output="SB_RAM.WDATA[15:0]"/>
    <direct name="MASK"  input="RAM.MASK[15:0]"  output="SB_RAM.MASK[15:0]"/>
    <direct name="WADDR" input="RAM.WADDR[7:0]"  output="SB_RAM.WADDR[7:0]" />
    <direct name="WE"    input="RAM.WE"          output="SB_RAM.WE" />
    <direct name="WCLKE" input="RAM.WCLKE"       output="SB_RAM.WCLKE" />
    <direct name="WCLK"  input="RAM.WCLK"        output="SB_RAM.WCLK" />
   </interconnect>

   <fc default_in_type="frac" default_in_val="0.0" default_out_type="frac" default_out_val="0.0">
    <!-- Connect all the ports to 2 local tracks -->
    <fc_override fc_type="abs" fc_val="2" port_name="RDATA" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RADDR" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RE"    segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RCLKE" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RCLK"  segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WDATA" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="MASK"  segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WADDR" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WE"    segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WCLKE" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WCLK"  segment_name="local"/>
   </fc>

   <pinlocations pattern="custom">
    <!-- RAMB Tile -->
    <loc side="right" xoffset="0" yoffset="0">
     RAM.RDATA[7:0]
     RAM.WADDR[10:0]
     RAM.MASK[7:0]
     RAM.WDATA[7:0]
     RAM.WCLKE
     RAM.WCLK
     RAM.WE
    </loc>
    <!-- RAMT Tile -->
    <loc side="right" xoffset="0" yoffset="1">
     RAM.RDATA[15:8]
[... 153 lines cut ...]
Warning 94: in check_rr_node: rr_node 3131 OPIN at (4,5) side=RIGHT, block_type=RAM ptc=67 has no out-going edges.     
Warning 95: in check_rr_node: rr_node 3132 OPIN at (4,5) side=RIGHT, block_type=RAM ptc=68 has no out-going edges.                                                                                                                            
Warning 96: in check_rr_node: rr_node 3133 OPIN at (4,5) side=RIGHT, block_type=RAM ptc=69 has no out-going edges.     
Warning 97: in check_rr_node: rr_node 3134 OPIN at (4,5) side=RIGHT, block_type=RAM ptc=70 has no out-going edges.   
Warning 98: in check_rr_node: rr_node 3135 OPIN at (4,5) side=RIGHT, block_type=RAM ptc=71 has no out-going edges.     
Warning 99: in check_rr_node: rr_node 3136 OPIN at (4,5) side=RIGHT, block_type=RAM ptc=72 has no out-going edges.                                  
-----
Cannot route from PIO.inpad[0] (rr_node: 1 type: SOURCE ptc: 1 xlow: 1 ylow: 2) to RAM.RADDR[0] (rr_node: 2836 type: SINK ptc: 0 xlow: 4 ylow: 2) to -- no possible path                                                                      
Warning 436: Unable to route between blocks at (1,2) and (4,2) to characterize delay (setting to inf)                  
Warning 437: Empty heap occurred in get_heap_head.                                                                   
Warning 438: Some blocks are impossible to connect in this architecture.                                               
Cannot route from PIO.inpad[0] (rr_node: 1 type: SOURCE ptc: 1 xlow: 1 ylow: 2) to RAM.RADDR[0] (rr_node: 2836 type: SINK ptc: 0 xlow: 4 ylow: 2) to -- no possible path                                                                      
Warning 439: Unable to route between blocks at (1,2) and (4,3) to characterize delay (setting to inf)                  
Warning 440: Empty heap occurred in get_heap_head.                                                                    
Warning 441: Some blocks are impossible to connect in this architecture.                                               
Cannot route from PIO.inpad[0] (rr_node: 1 type: SOURCE ptc: 1 xlow: 1 ylow: 2) to RAM.RADDR[0] (rr_node: 2988 type: SINK ptc: 0 xlow: 4 ylow: 4) to -- no possible path                                                                      
Warning 442: Unable to route between blocks at (1,2) and (4,4) to characterize delay (setting to inf)                  
Warning 443: Empty heap occurred in get_heap_head.                                                                    
```

#### Your Environment

* VTR revision used: 7d838604f0be248836a19410a9d18559d187136d
* Operating System and version: Linux Debian Testing
* Compiler version: gcc (Debian 6.3.0-18) 6.3.0 20170516

mithro, 2017-12-05:
[image]

mithro, 2017-12-05:
So it seems `<fc_override fc_type="abs" fc_val="2" port_name="RDATA[7:0]" segment_name="local"/>`  and `<loc side="right" xoffset="0" yoffset="0">` don't take partial pins, you can only specify the full port name.
I've reworked the code to;
```xml
  <pb_type name="RAM" height="2">
   <!-- Read port -->
   <output name="RDATAT" num_pins="8"  equivalent="false"/>
   <output name="RDATAB" num_pins="8"  equivalent="false"/>
   <input  name="RADDR"  num_pins="11" equivalent="false"/>
   <input  name="RE"     num_pins="1"  equivalent="false"/>
   <input  name="RCLKE"  num_pins="1"  equivalent="false"/>
   <clock  name="RCLK"   num_pins="1"  equivalent="false"/>
   <!-- Write port -->
   <input  name="WDATAT" num_pins="8"  equivalent="false"/>
   <input  name="WDATAB" num_pins="8"  equivalent="false"/>
   <input  name="MASKT"  num_pins="8"  equivalent="false"/>
   <input  name="MASKB"  num_pins="8"  equivalent="false"/>
   <input  name="WADDR"  num_pins="11" equivalent="false"/>
   <input  name="WE"     num_pins="1"  equivalent="false"/>
   <input  name="WCLKE"  num_pins="1"  equivalent="false"/>
   <clock  name="WCLK"   num_pins="1"  equivalent="false"/>

   <fc default_in_type="frac" default_in_val="0.0" default_out_type="frac" default_out_val="0.0">
    <fc_override fc_type="abs" fc_val="2" port_name="RDATAT" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RDATAB" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RADDR"  segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RE"     segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RCLKE"  segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="RCLK"   segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WDATAT" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WDATAB" segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="MASKT"  segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="MASKB"  segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WADDR"  segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WE"     segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WCLKE"  segment_name="local"/>
    <fc_override fc_type="abs" fc_val="2" port_name="WCLK"   segment_name="local"/>
   </fc>

   <pinlocations pattern="custom">
    <!-- RAMB Tile -->
    <loc side="right" xoffset="0" yoffset="0">
     RAM.RDATAB
     RAM.WADDR
     RAM.MASKB
     RAM.WDATAB
     RAM.WCLKE
     RAM.WCLK
     RAM.WE
    </loc>
    <!-- RAMT Tile -->
    <loc side="right" xoffset="0" yoffset="1">
     RAM.RDATAT
     RAM.RADDR
     RAM.MASKT
     RAM.WDATAT
     RAM.RCLKE
     RAM.RCLK
     RAM.RE
    </loc>
   </pinlocations>
   <switchblock_locations pattern="external_full_internal_straight"/>
  </pb_type>
  <!-- End RAM -->
```

But now I'm hitting the following;
`vtr-verilog-to-routing/vpr/src/route/rr_graph2.cpp:900 is_cblock: Assertion 'ofs < length' failed.`

mithro, 2017-12-05:
```
Type: "RAM"
	capacity: 1
	width: 1
	height: 2
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 0 1 2 3 4 5 6 7 8 9 10
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 11
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 12
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 13 14 15 16 17 18 19 20
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 21 22 23 24 25 26 27 28
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 29 30 31 32 33 34 35 36
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 37 38 39 40 41 42 43 44
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 45 46 47 48 49 50 51 52 53 54 55
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 56
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 57
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 58 59 60 61 62 63 64 65
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 66 67 68 69 70 71 72 73
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 74
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: global pins: 75
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 0 1 2 3 4 5 6 7 8 9 10
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 11
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 12
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 13 14 15 16 17 18 19 20
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 21 22 23 24 25 26 27 28
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 29 30 31 32 33 34 35 36
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 37 38 39 40 41 42 43 44
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 45 46 47 48 49 50 51 52 53 54 55
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 56
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 57
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 58 59 60 61 62 63 64 65
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 66 67 68 69 70 71 72 73
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 74
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span12 pins: 75
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 0 1 2 3 4 5 6 7 8 9 10
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 11
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 12
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 13 14 15 16 17 18 19 20
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 21 22 23 24 25 26 27 28
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 29 30 31 32 33 34 35 36
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 37 38 39 40 41 42 43 44
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 45 46 47 48 49 50 51 52 53 54 55
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 56
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 57
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 58 59 60 61 62 63 64 65
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 66 67 68 69 70 71 72 73
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 74
fc_value_type: FRACTIONAL fc_value: 0.000000 segment: span4 pins: 75
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 0 1 2 3 4 5 6 7 8 9 10
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 11
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 12
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 13 14 15 16 17 18 19 20
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 21 22 23 24 25 26 27 28
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 29 30 31 32 33 34 35 36
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 37 38 39 40 41 42 43 44
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 45 46 47 48 49 50 51 52 53 54 55
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 56
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 57
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 58 59 60 61 62 63 64 65
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 66 67 68 69 70 71 72 73
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 74
fc_value_type: ABSOLUTE fc_value: 2.000000 segment: local pins: 75
	num_drivers: 16
	num_receivers: 58
```

mithro, 2017-12-06:
@kmurray - Any idea what is going on here? 

I hit my head against it a bit yesterday but was unable to figure out the correct fix.

FYI When we fix this, then I should have blockrams on the iCE40. If you need more information (like the full arch.xml) I'm happy to provide.

kmurray, 2017-12-11:
Full architecture files would help debug the issue.  VPR should not crash on invalid input.

A quick look through the code indicates that `<fc_override>` only supports a single override per port.

I haven't looked at the `<loc>` handling code, but I think it should support per-pin location specifications.

closed 2025-05-15 by github-actions[bot]