verilog-to-routing/vtr-verilog-to-routing #259: Add tags for the vtr release versions

mithro, 2017-12-05:
At the moment this repository doesn't have any tags which means things like `git describe` fail with `fatal: No names found, cannot describe anything.`.

Would it be possible to add tags for the vtr releases which can be found at http://docs.verilogtorouting.org/en/latest/vtr/get_vtr/ in the git repository.

Looks like there have been the following releases;
 * 8.0??? -- Can't find a download link to that. --
 From http://docs.verilogtorouting.org/en/latest/vtr/get_vtr/#release 
 * 7.0 -- From http://www.eecg.utoronto.ca/vtr/vtr_release_7.0_full.tar.gz
 * 1.0 -- From http://www.eecg.utoronto.ca/vtr/vtr_release_1.0_full.tar.gz

kmurray, 2017-12-06:
This is something we plan to do for the next officially release.

VTR 8.0 will be the next release, but it has not been released yet.

Sadly no one kept track of the specific revisions used for the VTR 1.0/7.0 releases so we can't tag them retrospectively with any certainty.

mithro, 2017-12-06:
I'll do some archaeology and see if I can figure out what versions they came from.

Could we tag the very first commit in the repo with `v0.0` then? (It looks like you are using `X.Y` revision numbers?)

kmurray, 2017-12-11:
If you find a commit which we are confident matches the VTR 7 release that could be tagged.

I'd probably lean against tagging a `v0.0`, since the initial commit in the repo history is not really v0, but something > VPR 4.3 and < VTR 7. I suspect it is also > VPR 6 as well, but I'm not sure. It was before my time, and that initial import did not maintain the previous project history.

closed 2025-05-15 by github-actions[bot]