verilog-to-routing/vtr-verilog-to-routing #262: Colorize the tiles in the viewer based on the tile type

mithro, 2017-12-06:
It would be really awesome if we could colorize each of the tiles in the vpr `--disp` a given color.

This makes it easy to see which files are in which locations quickly. This is important when you have many different tile types in a non-regular pattern.

Example would be something like this;
[image]

Would also mean the viewer matches the pictures in your docs like this one -> 
[image]

(In fact we could probably generate these from the *actual* architecture description :-)

kmurray, 2017-12-11:
VPR should assign unique colours to each block/tile type.

With the latest version on my machine with: 

`vpr /project/trees/vtr/vtr_flow/arch/titan/stratixiv_arch.timing.xml /project/trees/vtr/vtr_flow/benchmarks/titan_other_blif/ucsb_152_tap_fir_stratixiv_arch_timing.blif --disp on`

I get:
[image]

Do you getting something different?

mithro, 2017-12-11:
This is what my vpr output on my architecture looks like;

[image]

The columns which have empty squares in it are different tile types to the other ones. The tiles around the edge are IO. Will give a test on your command line above...

vpr commit - 45b852bbe2d2428794f940ab00423625bf1055ca

mithro, 2017-12-11:
So running your example I get the same colors you do. See below;
[image]

Could it be the fact that almost all my cells have no logic assigned to them? (I'm currently testing with a single flip-flip blif;

```blif
.model top
.inputs clk di
.outputs D1
.subckt SB_DFF C=clk D=di Q=D1
.end

.model SB_DFF
.inputs C D
.outputs Q
.blackbox
.end
```

Does white mean something special in vpr?

closed 2017-12-12 by kmurray

mithro, 2017-12-12:
Confirmed as working!

[image]

commit 3b9009c9 (2017-12-12): vpr: Improve colouring of block locations in graphics

Previously, the first 3 block types were always gray (due to legacy
assumptions about empty blocks and IO).

We also now programmatically lighten the colours of blocks with no logic
assigned so they are easier to differentiate.

Closes #262.