verilog-to-routing/vtr-verilog-to-routing #263: Not able to install vtr on mac OS X

manikjindal91, 2017-12-09:
I am trying to install vtr on my mac. But, installation errors out running the make command. I have attached the log file. Please look into it and feedback.

* VTR revision used: 7.0
* Operating System and version: Mac OS X
[make_error.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1544419/make_error.txt)

jeanlego, 2017-12-09:
Try Changing libc++ to libstdc++

There is good explanation for that right here. 
https://stackoverflow.com/questions/19821041/g-ld-symbols-not-found-for-architecture-x86-64

Odin is written in C but compiled using c++ due to dependency. Odin need some library only found in libstdc++ as opposed to libc++ (the default one in xcode)

manikjindal91, 2017-12-10:
Hi Jeanlego,

Thanks for the response. I tried changing it.  But still, I am stuck with same error. 

g++ by default takes libstdc++. For gcc, I additionally gave the command line argument.

Attached is the log file. Please let me know if there is any other solution.

[error_log.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1545367/error_log.txt)

kmurray, 2017-12-11:
The VTR 7.0 release was never tested on Mac as far as I know. If you want to stick with VTR 7.0 I'd suggest trying Ubuntu instead.

Another option would be to try the latest VTR development code from here on Github (but be aware it may not be entirely stable). We have made efforts to make the latest VTR code more portable (e.g. it compiles with LLVM), but it is still untested on Mac.