verilog-to-routing/vtr-verilog-to-routing #264: If you specify a fixed_layout with width or height smaller than column specifications you get negative timing information.

mithro, 2017-12-10:
With an `arch.xml` file which has the following;
```xml
<fixed_layout name="XXX" width="5" height="5">
   <row     type="EMPTY" starty="5" priority="31"/>
</fixed_layout>
```

You get the following error;
```
Build routing resource graph took 11.5614 seconds                                                                                                         
Computing delta delay lookup matrix, may take a few seconds, please wait...                                                                               
Placement took 31.39 seconds                                                                                                                              
Error 1:                                                                                                                                                  
Type: Placement                                                                                                                                           
File: /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/place/timing_place_lookup.cpp                                               
Line: 566                                                                                                                                                   
Message: Found negative delay -1 for delta (2,4), but this is a valid delta between non-EMPTY blocks                                                        
```

Maybe the layout code needs to check things are within bounds when using a `fixed_layout` description?

kmurray, 2017-12-11:
For timing-driven placement VPR profiles the delta delays between blocks using the timing-driven router.

In the layout you described there are no blocks (everything is empty) so there are no valid delta delays to calculate, which leaves the delta delays at their invalid initial values.

The temporary work-around is to ensure that you have a reasonably 'full' device layout so all the delta delays get initialized to sane values.

Longer term this is code that I've been working to improve the robustness of (I encountered similar issues when working on another feature). So this should be improved in the near future.

kmurray, 2017-12-15:
Should be fixed in a15f5cf.

closed 2017-12-15 by kmurray