verilog-to-routing/vtr-verilog-to-routing #269: ABC overflow in MvcCoverStruct nWords

gorski123, 2017-12-19:
I am currently working with some FPGA designs that have large memories (1024 cells). When mapping these to hard memory blocks I have no problems, however when mapping them to soft logic I ran into issues with ABC. Memories up to 512 cells were OK, but over this limit caused ABC to crash. Tracing the problem back during debugging pointed me to an overflow in the nWords variable in the MvcCoverStruct, changing the variable type from char to unsigned char appears to fix the problem, but perhaps an even larger data type would be a better choice. I understand that I am working with some very large designs (8500+ nets and 4500+ clbs) and perhaps many people never hit these limits, but I figured perhaps you might want to be aware the limit is there.

kmurray, 2017-12-19:
Thanks for the bug report! I do recall encountering this issue in the past.

VTR uses a rather old version of ABC (ABC development occurs [here](https://bitbucket.org/alanmi/abc)), so this bug may be fixed in newer versions of ABC.

@KennethKent is working on upgrading the version of ABC which VTR uses to a more recent version which may fix the issue.

If you could verify whether new versions of ABC fix the issue, it would further motivate upgrading VTR's copy of ABC.

gorski123, 2017-12-19:
No problem!

Sure, I will try with the new version of ABC, but having a quick look at
the latest source on the site I would think it will almost for sure fix my
problem as it looks like they are now using an int datatype instead of a
char for nWords in the struct:

// the cover data structurestruct MvcCoverStruct{    int
nWords;       // the number of machine words    int
nUnused;      // the number of unused bits in the last word    int
        nBits;        // the number of used data bits in the cube
Mvc_List_t      lCubes;       // the single-linked list of cubes
Mvc_Cube_t **   pCubes;       // the array of cubes (for sorting)
int             nCubesAlloc;  // the size of allocated storage    int
*           pLits;        // the counter of lit occurrances in cubes
 Mvc_Cube_t *    pMask;        // the multipurpose mask
Mvc_Manager_t * pMem;         // the memory manager};

Jason

aarongraham9, 2017-12-19:
Yes **KennethKent** and I are currently working on upgrading the version of ABC. There are still some issues getting it to work in our current flow. We hope to have it working and pushed to the repository **_"soon"_**.

aarongraham9, 2018-04-16:
ABC has been updated in #312 and #314 so this issue should be fixed. Can you give it a try @gorski123?

gorski123, 2018-04-16:
Sure, Ill give it a try and report back.
Thanks
Jason

gorski123, 2018-04-20:
Hey Aaron,

That appears to have fixed my issue, thanks!

Jason

aarongraham9, 2018-04-23:
Excellent! Glad to help!

closed 2018-04-23 by aarongraham9