verilog-to-routing/vtr-verilog-to-routing #270: Can't "make" gcc (Ubuntu/Linaro 6.3.0-18ubuntu2~16.04) 6.3.0 20170519 system

reeshad, 2017-12-22:
In the root file of VTR when I try to build with "make" it gives me this:
flex: can't open Volume/VTR/vtr-verilog-to-routing/build/libs/EXTERNAL/libsdcparse/sdc_lexer.gen.hpp
libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/build.make:68: recipe for target 'libs/EXTERNAL/libsdcparse/sdc_lexer.gen.cpp' failed
make[2]: *** [libs/EXTERNAL/libsdcparse/sdc_lexer.gen.cpp] Error 1
CMakeFiles/Makefile2:828: recipe for target 'libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/all' failed
make[1]: *** [libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
#### Expected Behaviour

#### Current Behaviour

#### Possible Solution

#### Steps to Reproduce (for bugs)

1. 
2. 
3. 
4. 

#### Context

#### Your Environment

* VTR revision used: 7.0
* Operating System and version: gcc (Ubuntu/Linaro 6.3.0-18ubuntu2~16.04) 6.3.0 20170519
* Compiler version: gcc (Ubuntu/Linaro 6.3.0-18ubuntu2~16.04) 6.3.0 20170519

jeanlego, 2017-12-22:
I can see you are running on embedded platform. Although this is not supported and may break, I managed to get it to run using the following guidelines (they were taken from the docker file so some of it is useless) . https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/Running_On_ARM(raspberry_pi).txt 
Don't expect anything fantastic on embedded devices since the flow need a decent amount of memory.

Try this and if you have further issue I can try to help more. Make sure you take the latest version of vtr from GitHub on master.

reeshad, 2017-12-22:
@jeanlego Thanks! I will get back to you if there is any further issue. Thanks again :)

reeshad, 2017-12-22:
new issue. after applying this : 
apt-get install -y clang time supervisor build-essential g++ zip clang curl libssl-dev git libxml2-dev libx11-dev libxft-dev fontconfig libcairo2-dev gcc automake git cmake flex bison ctags gdb perl valgrind

it gives: Note, selecting 'exuberant-ctags' instead of 'ctags'
bison is already the newest version (2:3.0.4.dfsg-1build1).
build-essential is already the newest version (12.1ubuntu2).
cmake is already the newest version (3.7.2-1).
exuberant-ctags is already the newest version (1:5.9~svn20110310-11).
flex is already the newest version (2.6.1-1.3).
fontconfig is already the newest version (2.11.94-0ubuntu2).
g++ is already the newest version (4:6.3.0-2ubuntu1).
gcc is already the newest version (4:6.3.0-2ubuntu1).
time is already the newest version (1.7-25.1).
zip is already the newest version (3.0-11).
curl is already the newest version (7.52.1-4ubuntu1.4).
gdb is already the newest version (7.12.50.20170314-0ubuntu1.1).
git is already the newest version (1:2.11.0-2ubuntu0.3).
libssl-dev is already the newest version (1.0.2g-1ubuntu11.4).
perl is already the newest version (5.24.1-2ubuntu1.1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 clang : Depends: clang-4.0 (>= 4.0~) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

kmurray, 2018-01-08:
Continued in #271

closed 2018-01-08 by kmurray