verilog-to-routing/vtr-verilog-to-routing #271: flex: can't open Volume/VTR/vtr-verilog-to-routing/build/libs/EXTERNAL/libsdcparse/sdc_lexer.gen.hpp

reeshad, 2017-12-23:
flex: can't open Volume/VTR/vtr-verilog-to-routing/build/libs/EXTERNAL/libsdcparse/sdc_lexer.gen.hpp
libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/build.make:68: recipe for target 'libs/EXTERNAL/libsdcparse/sdc_lexer.gen.cpp' failed
make[2]: *** [libs/EXTERNAL/libsdcparse/sdc_lexer.gen.cpp] Error 1
CMakeFiles/Makefile2:791: recipe for target 'libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/all' failed
make[1]: *** [libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

gcc --version
gcc (Ubuntu 4.8.5-4ubuntu2) 4.8.5
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

g++ --version
g++ (Ubuntu 5.4.0-6ubuntu1~16.04.5) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

reeshad, 2017-12-23:
after this....

sudo make
[sudo] password for tasnim: 
[  1%] Built target libpugixml
[  1%] Built target liblog
[  1%] Built target version
[  1%] Generating vtr_version.cpp
-- VTR Version: 8.0.0-dev+ef6f2a1-dirty
Scanning dependencies of target libvtrutil
[  1%] Building CXX object libs/libvtrutil/CMakeFiles/libvtrutil.dir/vtr_version.cpp.o
[  1%] Linking CXX static library ../../../libs/libvtrutil/libvtrutil.a
[  2%] Built target libvtrutil
[  2%] Built target libpugiutil
[  4%] Built target libarchfpga
[  4%] Linking CXX executable ../../../libs/libarchfpga/read_arch
[  4%] Built target read_arch
[  4%] Built target libcatch
[  4%] Linking CXX executable test_vtrutil
[  4%] Built target test_vtrutil
[  5%] Built target test_log
[  6%] Built target libeasygl
[  6%] Built target libargparse
[  6%] [FLEX][SdcLexer] Building scanner with flex 2.6.0
flex: can't open Volume/VTR/vtr-verilog-to-routing/build/libs/EXTERNAL/libsdcparse/sdc_lexer.gen.hpp
libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/build.make:68: recipe for target 'libs/EXTERNAL/libsdcparse/sdc_lexer.gen.cpp' failed
make[2]: *** [libs/EXTERNAL/libsdcparse/sdc_lexer.gen.cpp] Error 1
CMakeFiles/Makefile2:791: recipe for target 'libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/all' failed
make[1]: *** [libs/EXTERNAL/libsdcparse/CMakeFiles/libsdcparse.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

kmurray, 2018-01-03:
Can you run the following commands, and provide the full output?

$ make distclean
$ make

reeshad, 2018-01-08:
@kmurray Thank you very much for your reply.... I followed your instruction and then I found this:

tasnim@tasnim-X540YA:~$ cd '/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master' 
tasnim@tasnim-X540YA:/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master$ make distclean
cd ./build && /usr/bin/cmake -DCMAKE_BUILD_TYPE=release -G 'Unix Makefiles'  .. 
CMake Error: The current CMakeCache.txt directory /media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master/build/CMakeCache.txt is different than the directory /cygdrive/d/VTR/vtr-verilog-to-routing-master/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
-- CMAKE_BUILD_TYPE: release
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2 - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wimplicit-fallthrough=0
-- Performing Test CXX_COMPILER_SUPPORTS_-Wimplicit-fallthrough=0 - Failed
-- CMAKE_CXX_FLAGS:  -std=c++14  -Wall -Wextra -Wpedantic -Wcast-qual -Wcast-align -Wshadow -Wformat=2 -Wno-format-nonliteral -Wlogical-op -Wmissing-declarations -Wmissing-include-dirs -Wredundant-decls -Wswitch-default -Wundef -Wunused-variable -Wdisabled-optimization -Wnoexcept -Woverloaded-virtual -Wctor-dtor-privacy -Wnon-virtual-dtor   
-- EasyGL: UNIX detected, looking for X11 and font libraries
-- EasyGL: graphics enabled
-- Tatum will support parallel analysis
-- Configuring incomplete, errors occurred!
See also "/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master/build/CMakeFiles/CMakeOutput.log".
See also "/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master/build/CMakeFiles/CMakeError.log".
Makefile:60: recipe for target 'build/Makefile' failed
make: *** [build/Makefile] Error 1
tasnim@tasnim-X540YA:/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master$ sudo make distclean
[sudo] password for tasnim: 
cd ./build && /usr/bin/cmake -DCMAKE_BUILD_TYPE=release -G 'Unix Makefiles'  .. 
-- CMAKE_BUILD_TYPE: release
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2 - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wimplicit-fallthrough=0
-- Performing Test CXX_COMPILER_SUPPORTS_-Wimplicit-fallthrough=0 - Failed
-- CMAKE_CXX_FLAGS:  -std=c++14  -Wall -Wextra -Wpedantic -Wcast-qual -Wcast-align -Wshadow -Wformat=2 -Wno-format-nonliteral -Wlogical-op -Wmissing-declarations -Wmissing-include-dirs -Wredundant-decls -Wswitch-default -Wundef -Wunused-variable -Wdisabled-optimization -Wnoexcept -Woverloaded-virtual -Wctor-dtor-privacy -Wnon-virtual-dtor   
-- EasyGL: UNIX detected, looking for X11 and font libraries
-- EasyGL: graphics enabled
-- Tatum will support parallel analysis
-- Configuring done
-- Generating done
-- Build files have been written to: /media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master/build
Cleaning build...
Removing build system files..
tasnim@tasnim-X540YA:/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master$ make
cd ./build && /usr/bin/cmake -DCMAKE_BUILD_TYPE=release -G 'Unix Makefiles'  .. 
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_BUILD_TYPE: release
-- Performing Test CXX_COMPILER_SUPPORTS_CXX14_FLAG
-- Performing Test CXX_COMPILER_SUPPORTS_CXX14_FLAG - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wall
-- Performing Test CXX_COMPILER_SUPPORTS_-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wextra
-- Performing Test CXX_COMPILER_SUPPORTS_-Wextra - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wpedantic
-- Performing Test CXX_COMPILER_SUPPORTS_-Wpedantic - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-qual
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-qual - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-align
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-align - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wshadow
-- Performing Test CXX_COMPILER_SUPPORTS_-Wshadow - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2 - Success
[... 120 lines cut ...]
     const char* VCS_REVISION = "fatal: Not a git repository (or any parent up t
                                ^
/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master/build/libs/libvtrutil/vtr_version.cpp:16:5: error: missing terminating " character
     const char* VCS_REVISION = "fatal: Not a git repository (or any parent up t
     ^
/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master/build/libs/libvtrutil/vtr_version.cpp:17:75: warning: missing terminating " character
 Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).";
                                                                           ^
/media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master/build/libs/libvtrutil/vtr_version.cpp:17:1: error: missing terminating " character
[... cut ...]
libs/libvtrutil/CMakeFiles/libvtrutil.dir/build.make:306: recipe for target 'libs/libvtrutil/CMakeFiles/libvtrutil.dir/vtr_version.cpp.o' failed
make[3]: *** [libs/libvtrutil/CMakeFiles/libvtrutil.dir/vtr_version.cpp.o] Error 1
CMakeFiles/Makefile2:332: recipe for target 'libs/libvtrutil/CMakeFiles/libvtrutil.dir/all' failed
make[2]: *** [libs/libvtrutil/CMakeFiles/libvtrutil.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:45: recipe for target 'all' failed
make: *** [all] Error 2

reeshad, 2018-01-08:
I have successfully built it but new problem popes up...when I tries to  Run "./run_quick_test.pl" to check that the build worked or not it gives me this: 
tasnim@tasnim-X540YA:~/vtr-verilog-to-routing$ ./run_quick_test.pl
=============================================
    Verilog-to-Routing Regression Testing    
=============================================

Running regression test... 
scripts/run_vtr_task.pl -l /home/tasnim/vtr-verilog-to-routing/vtr_flow/tasks/regression_tests/vtr_reg_basic/task_list.txt 

regression_tests/vtr_reg_basic/basic_no_timing
-----------------------------------------
Wide character in print at scripts/run_vtr_task.pl line 414.
Current time: জানু-08 05:20 অপরাহ্ণ.  Expected runtime of next benchmark: 1 seconds
k4_N10_memSize16384_memData64/ch_intrinsics...OK
Wide character in print at scripts/run_vtr_task.pl line 414.
Current time: জানু-08 05:21 অপরাহ্ণ.  Expected runtime of next benchmark: Unknown
k4_N10_memSize16384_memData64/diffeq1...OK

regression_tests/vtr_reg_basic/basic_timing
-----------------------------------------
Wide character in print at scripts/run_vtr_task.pl line 414.
Current time: জানু-08 05:24 অপরাহ্ণ.  Expected runtime of next benchmark: 2 seconds
k6_N10_mem32K_40nm/ch_intrinsics...OK
Wide character in print at scripts/run_vtr_task.pl line 414.
Current time: জানু-08 05:27 অপরাহ্ণ.  Expected runtime of next benchmark: Unknown
k6_N10_mem32K_40nm/diffeq1...OK

regression_tests/vtr_reg_basic/basic_timing_no_sdc
-----------------------------------------
Wide character in print at scripts/run_vtr_task.pl line 414.
Current time: জানু-08 05:35 অপরাহ্ণ.  Expected runtime of next benchmark: 16 seconds
k6_N10_mem32K_40nm/mkPktMerge...OK

Parsing test results... 
scripts/parse_vtr_task.pl -l /home/tasnim/vtr-verilog-to-routing/vtr_flow/tasks/regression_tests/vtr_reg_basic/task_list.txt  

Checking test results... 

Calculating QoR results... 
scripts/parse_vtr_task.pl -l /home/tasnim/vtr-verilog-to-routing/vtr_flow/tasks/regression_tests/vtr_reg_basic/task_list.txt -check_golden -calc_geomean  
regression_tests/vtr_reg_basic/basic_no_timing...[Fail] 
 k4_N10_memSize16384_memData64.xml/ch_intrinsics.v pack_time: golden = 0.258723 result = 7.56
[Fail] 
 k4_N10_memSize16384_memData64.xml/ch_intrinsics.v place_time: golden = 0.241872 result = 28.56
[Fail] 
 k4_N10_memSize16384_memData64.xml/ch_intrinsics.v min_chan_width_route_time: golden = 0.606091 result = 53.10
[Fail] 
 k4_N10_memSize16384_memData64.xml/diffeq1.v pack_time: golden = 0.357711 result = 10.77
[Fail] 
 k4_N10_memSize16384_memData64.xml/diffeq1.v place_time: golden = 0.400464 result = 40.76
[Fail] 
 k4_N10_memSize16384_memData64.xml/diffeq1.v min_chan_width_route_time: golden = 2.97256 result = 110.62
regression_tests/vtr_reg_basic/basic_timing...[Fail] 
 k6_N10_mem32K_40nm.xml/ch_intrinsics.v pack_time: golden = 0.282112 result = 7.90
[Fail] 
 k6_N10_mem32K_40nm.xml/ch_intrinsics.v place_time: golden = 0.663885 result = 55.06
[Fail] 
 k6_N10_mem32K_40nm.xml/ch_intrinsics.v min_chan_width_route_time: golden = 0.899631 result = 71.86
[Fail] 
 k6_N10_mem32K_40nm.xml/ch_intrinsics.v crit_path_route_time: golden = 0.176981 result = 7.41
[Fail] 
 k6_N10_mem32K_40nm.xml/diffeq1.v pack_time: golden = 0.365102 result = 9.73
[Fail] 
 k6_N10_mem32K_40nm.xml/diffeq1.v place_time: golden = 1.18001 result = 91.06
[Fail] 
 k6_N10_mem32K_40nm.xml/diffeq1.v min_chan_width_route_time: golden = 8.14421 result = 352.33
[Fail] 
 k6_N10_mem32K_40nm.xml/diffeq1.v crit_path_route_time: golden = 0.748761 result = 41.02
regression_tests/vtr_reg_basic/basic_timing_no_sdc...[Fail] 
 k6_N10_mem32K_40nm.xml/mkPktMerge.v pack_time: golden = 0.640585 result = 22.69
[Fail] 
 k6_N10_mem32K_40nm.xml/mkPktMerge.v place_time: golden = 1.85706 result = 140.01
[Fail] 
 k6_N10_mem32K_40nm.xml/mkPktMerge.v min_chan_width_route_time: golden = 11.6591 result = 682.10
[Fail] 
 k6_N10_mem32K_40nm.xml/mkPktMerge.v crit_path_route_time: golden = 1.40032 result = 70.50

Test 'vtr_reg_basic' had 18 qor test failures

Test 'vtr_reg_basic' had 0 run failures

Test complete

Error: 18 tests failed!

kmurray, 2018-01-08:
This at the end of your build log indicates that the build failed:
```
Makefile:45: recipe for target 'all' failed
make: *** [all] Error 2
```

It appears that you have the source code outside of a GIT repository, which was causing this build issue:
```
media/tasnim/New Volume/VTR/vtr-verilog-to-routing-master/build/libs/libvtrutil/vtr_version.cpp:17:1: error: missing terminating " character
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).";
```

I've fixed it so it should build properly even if the code is not in a git repo (see 8121315). Try updating the source tree and building again.

reeshad, 2018-01-22:
When I am trying to run the test(./run_quick_test.pl) it gives me this:
Calculating QoR results... 
scripts/parse_vtr_task.pl -l /home/tasnim/Downloads/vtr-verilog-to-routing-master/vtr_flow/tasks/regression_tests/vtr_reg_basic/task_list.txt -check_golden -calc_geomean  
regression_tests/vtr_reg_basic/basic_no_timing...[Fail] 
 k4_N10_memSize16384_memData64.xml/ch_intrinsics.v min_chan_width_route_time: golden = 0.606091 result = 7.77
regression_tests/vtr_reg_basic/basic_timing...[Pass]
regression_tests/vtr_reg_basic/basic_timing_no_sdc...[Fail] 
 k6_N10_mem32K_40nm.xml/mkPktMerge.v min_chan_width_route_time: golden = 11.6591 result = 123.17
[Fail] 
 k6_N10_mem32K_40nm.xml/mkPktMerge.v crit_path_route_time: golden = 1.40032 result = 14.69

Test 'vtr_reg_basic' had 3 qor test failures

Test 'vtr_reg_basic' had 0 run failures

Test complete

Error: 3 tests failed!

Please Help!

kmurray, 2018-01-22:
> k6_N10_mem32K_40nm.xml/mkPktMerge.v min_chan_width_route_time: golden = 11.6591 result = 123.17

The above indicates that this particular benchmark took 123.17 seconds to find the minimum channel width on your computer, while our 'golden' reference took 11.65 seconds.

Most likely this is just a difference in computer speed. The golden reference is run on a high-performance desktop-class x86 CPU. I believe you are using a mobile-class ARM-based system which will be quite slow in comparison. As a result I wouldn't worry about these errors.

closed 2018-01-22 by kmurray