verilog-to-routing/vtr-verilog-to-routing #276: Empty Truth Table for Buffer/Passthrough LUT(s)

brettgradyeng, 2018-01-23:
LUT(s) that are inserted into the physical netlist as passthroughs do _not_ have truth tables in the atom netlist, nor do they have children LUTs.

#### Expected Behaviour

- pb blocks for LUTs should have a primitive child_pb "lut" created for them.
- All LUTs used to implement the netlist should have an associated atom block id and truth table.

#### Current Behaviour

- Passthrough LUTs do _not_ have any block_id or truth table associated in the atom netlist.
- Passthrough LUT t_pb(s) do _not_ have any primitive "lut" pb as children (child_pbs is NULL).

#### Possible Solution

When passthrough LUTs are created generate corresponding truth table entries in the atom netlist.

#### Steps to Reproduce (for bugs)

1. Use gdb or another debugger
2. Finish all routing and placement (_place_and_route.cpp:372_ is a good place to put a breakpoint)
3. Find the fifth FLE in CLB_2_1 in the device grid (x=2,y=1).
4. Look inside hierarchy till you find a pb_type "flut5" with the name _"top^z_4~1_FF_NODE"_
5. The child of the aforementioned block has type "lut5", but has no child_pb and has no associated truth table. The following code however shows that there is an atom net associated with one of the inputs to the LUT pb (where "**pb**" is the "lut5" physical block). This demonstrates the block _is_ used, but does not have a truth table.

```
int p_pin_count = pb->pb_graph_node->input_pins[0][j].pin_count_in_cluster;
p_net = toplevel_pb->pb_route[p_pin_count].atom_net_id;
```

#### Context

This issue seems to be not so much a bug but an unstated convention, wherein extra LUTs used as routing are not inserted into the netlist . This affects generation of bitstreams from VTR/VPR as truth tables typically determine the configuration memory for LUTs. Reverse engineering the correct configuration for routing is messy if you do not have a truth table.

#### Your Environment

* VTR revision used: **VTR 8.0**
* Operating System and version: **Linux (4.14.13-1-ARCH)**
* Compiler version: **g++ (7.2.1)**

kmurray, 2018-01-24:
Thanks for the detailed report. 

Would it be possible to attach the architecture and netlist you used?

brettgrady, 2018-01-25:
I updated the description with testcase files and a (hopefully) working method to replicate the problem.

closed 2025-05-15 by github-actions[bot]