verilog-to-routing/vtr-verilog-to-routing #277: Long tall FPGA seems to have issues causing unconnected routing

mithro, 2018-02-01:
With the architecture file [described here](https://github.com/mithro/symbiflow-arch-defs/blob/b7a1149144e4787f7ac188532d029397eabe4b9e/artix7/arch/xc7a50t-roi-virt/merged.xml) I'm getting some weird errors about tiles I'm expecting to have outgoing edges not haing them, see below;
```
Warning 42: in check_rr_node: RR node: 66 type: OPIN location: (1,7) pin: 1 pin_name: PAD.inpad[0] has no out-going edges.                             
Warning 43: in check_rr_node: RR node: 67 type: OPIN location: (1,7) pin: 1 pin_name: PAD.inpad[0] has no out-going edges.                             
Warning 44: in check_rr_node: RR node: 69 type: OPIN location: (1,7) pin: 1 pin_name: PAD.inpad[0] has no out-going edges.                             
Warning 45: in check_rr_node: RR node: 1456 type: OPIN location: (3,7) pin: 68 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[0] has no out-going edges.   
Warning 46: in check_rr_node: RR node: 1457 type: OPIN location: (3,7) pin: 69 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[1] has no out-going edges.   
Warning 47: in check_rr_node: RR node: 1458 type: OPIN location: (3,7) pin: 70 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[2] has no out-going edges.   
Warning 48: in check_rr_node: RR node: 1459 type: OPIN location: (3,7) pin: 71 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[3] has no out-going edges.   
Warning 49: in check_rr_node: RR node: 1460 type: OPIN location: (3,7) pin: 72 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[4] has no out-going edges.   
Warning 50: in check_rr_node: RR node: 1461 type: OPIN location: (3,7) pin: 73 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[5] has no out-going edges.   
Warning 51: in check_rr_node: RR node: 1462 type: OPIN location: (3,7) pin: 74 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[6] has no out-going edges.   
Warning 52: in check_rr_node: RR node: 1463 type: OPIN location: (3,7) pin: 75 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[7] has no out-going edges.   
Warning 53: in check_rr_node: RR node: 1464 type: OPIN location: (3,7) pin: 76 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[8] has no out-going edges.   
Warning 54: in check_rr_node: RR node: 1465 type: OPIN location: (3,7) pin: 77 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[9] has no out-going edges.   
Warning 55: in check_rr_node: RR node: 1466 type: OPIN location: (3,7) pin: 78 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[10] has no out-going edges.  
Warning 56: in check_rr_node: RR node: 1467 type: OPIN location: (3,7) pin: 79 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[11] has no out-going edges.  
Warning 57: in check_rr_node: RR node: 1468 type: OPIN location: (3,7) pin: 80 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[12] has no out-going edges.  
Warning 58: in check_rr_node: RR node: 1469 type: OPIN location: (3,7) pin: 81 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[13] has no out-going edges.  
Warning 59: in check_rr_node: RR node: 1470 type: OPIN location: (3,7) pin: 82 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[14] has no out-going edges.  
Warning 60: in check_rr_node: RR node: 1471 type: OPIN location: (3,7) pin: 83 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[15] has no out-going edges.                                             
Warning 61: in check_rr_node: RR node: 1472 type: OPIN location: (3,7) pin: 84 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[16] has no out-going edges.  
Warning 62: in check_rr_node: RR node: 1473 type: OPIN location: (3,7) pin: 85 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[17] has no out-going edges.  
Warning 63: in check_rr_node: RR node: 1474 type: OPIN location: (3,7) pin: 86 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[18] has no out-going edges.  
Warning 64: in check_rr_node: RR node: 1475 type: OPIN location: (3,7) pin: 87 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[19] has no out-going edges.  
Warning 65: in check_rr_node: RR node: 1476 type: OPIN location: (3,7) pin: 88 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[20] has no out-going edges.  
Warning 66: in check_rr_node: RR node: 1477 type: OPIN location: (3,7) pin: 89 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[21] has no out-going edges.  
Warning 67: in check_rr_node: RR node: 1478 type: OPIN location: (3,7) pin: 90 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[22] has no out-going edges.  
Warning 68: in check_rr_node: RR node: 1479 type: OPIN location: (3,7) pin: 91 pin_name: BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L[23] has no out-going edges.  
Warning 69: in check_rr_node: RR node: 1480 type: OPIN location: (3,7) pin: 92 pin_name: BLOCK_CLBLL_L_INT_L.COUT_N[0] has no out-going edges.         
Warning 70: in check_rr_node: RR node: 1481 type: OPIN location: (3,7) pin: 93 pin_name: BLOCK_CLBLL_L_INT_L.COUT_N[1] has no out-going edges.    
```

The error occurs when using a fixed_layout with `width="8" height="16"`
```xml
  <fixed_layout name="xc7a50t-test" width="8" height="16">

   <col type="EMPTY" startx="0" priority="10"/>
   <col type="PAD" startx="1" priority="10"/>
   <col type="EMPTY" startx="2" priority="10"/>
   <col type="BLOCK_CLBLL_L_INT_L" startx="3" priority="10"/>
   <!--<col type="BLOCK_CLBLL_L_INT_L" startx="4" priority="10" /> -->
   <col type="EMPTY" startx="5" priority="10"/>
   <col type="PAD" startx="6" priority="10"/>
   <col type="EMPTY" startx="7" priority="10"/>

   <row type="EMPTY" starty="0" priority="11"/>
   <row type="EMPTY" starty="15" priority="11"/>
  </fixed_layout>
```

If I change the width to be 16, the error messages go away (IE see below);
```xml
  <fixed_layout name="xc7a50t-test" width="16" height="16">

   <col type="EMPTY" startx="0" priority="10"/>
   <col type="PAD" startx="1" priority="10"/>
   <col type="EMPTY" startx="2" priority="10"/>
   <col type="BLOCK_CLBLL_L_INT_L" startx="3" priority="10"/>
   <col type="EMPTY" startx="5" priority="10"/>
   <col type="PAD" startx="6" priority="10"/>
   <col type="EMPTY" startx="7" priority="10"/>

   <row type="EMPTY" starty="0" priority="11"/>
   <row type="EMPTY" starty="15" priority="11"/>
  </fixed_layout>
```

This feels like somewhere is using a width value when it should be using a height value.

mithro, 2018-02-01:
@JohnDMcMaster said he might take a look at this bug tomorrow as an introduction to the vpr work flow.

mcmasterg, 2018-02-01:
I'll take a look tomorrow

kmurray, 2018-02-01:
Before c958c2b we were only testing square device aspect ratios, so a width/height mix-up is possible.

I fixed a (possibly) related bug in 44ab268; so probably best to make sure you have that fix when you start debugging.

mithro, 2018-02-01:
I also notice that your `zoom to fit` button in the GUI seems to have the width/height swapped too...

mcmasterg, 2018-02-01:
[image]

Red: missing elements
Green: elements normally present

kmurray, 2018-02-01:
> I also notice that your zoom to fit button in the GUI seems to have the width/height swapped too...

Should be fixed in 86bd6b1.

kmurray, 2018-02-07:
Things look consistent for me with f1db038:

```
vpr merged.xml test_eblif.eblif --device xc7a50t-test --disp on --route_chan_width 3
```
Produces:

[image]

mithro, 2018-02-07:
@kmurray - You need to change the width to be non-square -- the temporary work around was to make the architecture square.

mithro, 2018-04-15:
Looks like this still occuring in the latest vpr;
```
VPR FPGA Placement and Routing.                                                               
Version: 8.0.0-dev+vpr-7.0.5-5756-gcd10a5664-dirty                                            
Revision: vpr-7.0.5-5756-gcd10a5664-dirty                                                     
Compiled: 2018-04-14T13:16:07 (debug build)                                                   
Compiler: GNU 6.3.0 on Linux-4.9.0-4-amd64 x86_64                                             
University of Toronto                                                                         
vtr-users@googlegroups.com                                                                    
This is free open source code under MIT license.    
```

mithro, 2019-05-06:
@kmurray It appears that there are still issues around long / tall layouts.

Using the following [arch.xml](https://gist.github.com/mithro/a4a8e8c9f44a55089231cac95275d867#file-arch-xml) and [blif](https://gist.github.com/mithro/a4a8e8c9f44a55089231cac95275d867#file-dsp_combinational-test-eblif) I get weird output like;
```
Warning 3: Logic block #65 (a25) has only 1 input pin 'a25.I' -- the whole block is hanging logic that should be swept.
Warning 4: Logic block #66 (a17) has only 1 input pin 'a17.I' -- the whole block is hanging logic that should be swept.
Warning 5: Logic block #67 (a18) has only 1 input pin 'a18.I' -- the whole block is hanging logic that should be swept.
Warning 6: Logic block #68 (a19) has only 1 input pin 'a19.I' -- the whole block is hanging logic that should be swept.
Warning 7: Logic block #69 (a20) has only 1 input pin 'a20.I' -- the whole block is hanging logic that should be swept.
Warning 8: Logic block #70 (a21) has only 1 input pin 'a21.I' -- the whole block is hanging logic that should be swept.
Warning 9: Logic block #71 (a22) has only 1 input pin 'a22.I' -- the whole block is hanging logic that should be swept.
Warning 10: Logic block #72 (a23) has only 1 input pin 'a23.I' -- the whole block is hanging logic that should be swept.
Warning 11: Logic block #73 (a24) has only 1 input pin 'a24.I' -- the whole block is hanging logic that should be swept.
Warning 12: Logic block #74 (a16) has only 1 input pin 'a16.I' -- the whole block is hanging logic that should be swept.
Warning 13: Logic block #75 (a26) has only 1 input pin 'a26.I' -- the whole block is hanging logic that should be swept.
Warning 14: Logic block #76 (a27) has only 1 input pin 'a27.I' -- the whole block is hanging logic that should be swept.
Warning 15: Logic block #77 (a28) has only 1 input pin 'a28.I' -- the whole block is hanging logic that should be swept.
Warning 16: Logic block #78 (a29) has only 1 input pin 'a29.I' -- the whole block is hanging logic that should be swept.
Warning 17: Logic block #79 (a30) has only 1 input pin 'a30.I' -- the whole block is hanging logic that should be swept.
Warning 18: Logic block #80 (a31) has only 1 input pin 'a31.I' -- the whole block is hanging logic that should be swept.
Warning 19: Logic block #81 (b0) has only 1 input pin 'b0.I' -- the whole block is hanging logic that should be swept.
Warning 20: Logic block #82 (b1) has only 1 input pin 'b1.I' -- the whole block is hanging logic that should be swept.
Warning 21: Logic block #83 (a15) has only 1 input pin 'a15.I' -- the whole block is hanging logic that should be swept.
Warning 22: Logic block #84 (a14) has only 1 input pin 'a14.I' -- the whole block is hanging logic that should be swept.
Warning 23: Logic block #85 (a13) has only 1 input pin 'a13.I' -- the whole block is hanging logic that should be swept.
Warning 24: Logic block #86 (a12) has only 1 input pin 'a12.I' -- the whole block is hanging logic that should be swept.
Warning 25: Logic block #87 (a11) has only 1 input pin 'a11.I' -- the whole block is hanging logic that should be swept.
Warning 26: Logic block #88 (a10) has only 1 input pin 'a10.I' -- the whole block is hanging logic that should be swept.
Warning 27: Logic block #89 (a9) has only 1 input pin 'a9.I' -- the whole block is hanging logic that should be swept.
Warning 28: Logic block #90 (a8) has only 1 input pin 'a8.I' -- the whole block is hanging logic that should be swept.
Warning 29: Logic block #91 (a7) has only 1 input pin 'a7.I' -- the whole block is hanging logic that should be swept.
Warning 30: Logic block #92 (a6) has only 1 input pin 'a6.I' -- the whole block is hanging logic that should be swept.
Warning 31: Logic block #93 (a5) has only 1 input pin 'a5.I' -- the whole block is hanging logic that should be swept.
Warning 32: Logic block #94 (a4) has only 1 input pin 'a4.I' -- the whole block is hanging logic that should be swept.
Warning 33: Logic block #95 (a3) has only 1 input pin 'a3.I' -- the whole block is hanging logic that should be swept.
Warning 34: Logic block #96 (a2) has only 1 input pin 'a2.I' -- the whole block is hanging logic that should be swept.
Warning 35: Logic block #97 (a1) has only 1 input pin 'a1.I' -- the whole block is hanging logic that should be swept.
Warning 36: Logic block #98 (b17) has only 1 input pin 'b17.I' -- the whole block is hanging logic that should be swept.
Warning 37: Logic block #99 (m) has only 1 input pin 'm.I' -- the whole block is hanging logic that should be swept.
Warning 38: Logic block #100 (b31) has only 1 input pin 'b31.I' -- the whole block is hanging logic that should be swept.
Warning 39: Logic block #101 (b30) has only 1 input pin 'b30.I' -- the whole block is hanging logic that should be swept.
Warning 40: Logic block #102 (b29) has only 1 input pin 'b29.I' -- the whole block is hanging logic that should be swept.
Warning 41: Logic block #103 (b28) has only 1 input pin 'b28.I' -- the whole block is hanging logic that should be swept.
Warning 42: Logic block #104 (b27) has only 1 input pin 'b27.I' -- the whole block is hanging logic that should be swept.
Warning 43: Logic block #105 (b26) has only 1 input pin 'b26.I' -- the whole block is hanging logic that should be swept.
Warning 44: Logic block #106 (b25) has only 1 input pin 'b25.I' -- the whole block is hanging logic that should be swept.
[... cut ...]
Warning 66: Logic block #128 (b3) has only 1 input pin 'b3.I' -- the whole block is hanging logic that should be swept.
Warning 67: Logic block #129 (b2) has only 1 input pin 'b2.I' -- the whole block is hanging logic that should be swept.
```

And then vpr dies with;
```
/github/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:305 operator[]: Assertion 'index < this->dim_sizes_[0]' failed (Index out of range (above dimension maximum)).
```

If I change the layout to be square, everything runs fine.

kmurray, 2019-05-07:
I think this particular out-of-bounds access is fixed in fc2cf8f3.

However it occurs in the less-commonly used bi-directional RR graph construction code. Are you really modelling a bi-directional RR graph? I'd expect most modern architectures to use uni-directional routing.

mithro, 2019-05-07:
I have switched us to `unidir` routing here, still be good for this to work :-)

closed 2025-05-15 by github-actions[bot]