verilog-to-routing/vtr-verilog-to-routing #278: Issue with adding a connection making packing fail

mithro, 2018-02-02:
When I enable this line; (see [full source here](https://github.com/mithro/symbiflow-arch-defs/blob/0dc17135edf79dd4060fc540e017cd9ae68e64a7/artix7/arch/xc7a50t-roi-virt/merged.xml#L7298-L7313))
```xml 
   <direct name="WW2BEG" input="TILE_INT_L.WW2BEG" output="BLOCK_CLBLL_L_INT_L.WW2BEG" />
   <direct name="WW4BEG" input="TILE_INT_L.WW4BEG" output="BLOCK_CLBLL_L_INT_L.WW4BEG" />
   <!-- Local Interconnects -->
   <direct name="BYP_L"  input="TILE_INT_L.BYP_L"    output="TILE_CLBLL_L.BYP" />
   <!--
   <direct name="CLK_L"  input="TILE_INT_L.CLK_L"    output="TILE_CLBLL_L.CLK" />
   -->
   <direct name="CTRL_L" input="TILE_INT_L.CTRL_L"   output="TILE_CLBLL_L.CTRL"/>
   <direct name="FAN"    input="TILE_INT_L.FAN_L"    output="TILE_CLBLL_L.FAN" />
   <direct name="IMUX_L" input="TILE_INT_L.IMUX_L"   output="TILE_CLBLL_L.IMUX"/>
   <direct name="LOGIC_OUTS_L" input="TILE_CLBLL_L.LOGIC_OUTS" output="TILE_INT_L.LOGIC_OUTS_L" />
   <direct name="TEST_CLK_L"  input="BLOCK_CLBLL_L_INT_L.CLK_L"    output="TILE_CLBLL_L.CLK" />
   <!--
   <direct name="TEST_BYP_L"  input="BLOCK_CLBLL_L_INT_L.BYP_L"    output="TILE_CLBLL_L.BYP" />
   <direct name="TEST_CTRL_L" input="BLOCK_CLBLL_L_INT_L.CTRL_L"   output="TILE_CLBLL_L.CTRL"/>
   <direct name="TEST_FAN"    input="BLOCK_CLBLL_L_INT_L.FAN_L"    output="TILE_CLBLL_L.FAN" />
   <direct name="TEST_IMUX_L" input="BLOCK_CLBLL_L_INT_L.IMUX_L"   output="TILE_CLBLL_L.IMUX"/>
   <direct name="TEST_LOGIC_OUTS_L" input="TILE_CLBLL_L.LOGIC_OUTS" output="BLOCK_CLBLL_L_INT_L.LOGIC_OUTS_L" />
   -->
```

I go from packing succeeding to the following error;
```
Not enough resources expand FPGA size to (16 x 16)
Complex block 0: out:D1, type: PAD .
Passed route at end.
Packing took 0.04 seconds
Error 1: 
Type: Packing
File: /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/pack/cluster.cpp
Line: 1972
Message: Can not find any logic block that can implement molecule.
	Atom D1

Makefile:63: recipe for target 'ff.disp' failed
make: *** [ff.disp] Error 1
```

This also occurs with a .blif which only has a lut and shouldn't be using the `CLK_L` lines at all...
```
Warning 191: Block type 'INT_R' was not specified in device grid layout
Warning 192: Block type 'HCLK_L' was not specified in device grid layout
Warning 193: Block type 'HCLK_R' was not specified in device grid layout
Not enough resources expand FPGA size to (16 x 16)
Complex block 0: out:O, type: PAD .
Passed route at end.
Packing took 0.04 seconds
Error 1: 
Type: Packing
File: /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/pack/cluster.cpp
Line: 1972
Message: Can not find any logic block that can implement molecule.
	Atom O

Makefile:63: recipe for target 'lut.disp' failed
make: *** [lut.disp] Error 1
```

As far as I understand adding this line should only make it easier to do routing as it is adding *another* path...

mithro, 2018-02-07:
[This example](http://hopper.mithis.com/~tim/issue-278.tar.bz2) is a bit easier to look at (but still not minimal).

To cause the packing failure remove the command from the ` <direct name="CLK_L" input="BLK_BB-INT_L.CLK_L" output="BLK_TI-CLBLL_L.CLK" />` on line 10089.

mithro, 2018-02-07:
Should download and extract the tarball and then be able to run `make DEVICE=xc7a50t-test ff.disp`

kmurray, 2018-02-08:
I think I've narrowed the issue down and come-up with a work around.

It seems that this architecture exposes a bug in the packer's pin-counting feasibility filter, causing it to (incorrectly) state that a primitive (e.g. LUT/FF) can not be routed within a CLB, when in fact it can be.

The long-term solution is to fix the pin-counter to work correctly on this architecture. At this point I don't yet understand what the problem within the pin-counter is. However there is a work-around.

**Work-around**
I've added a command-line option in 9b00c2e allowing the the pin feasibility filter to be disabled, which allows the example circuits to pack.

For example, with the pin feasibility filter enabled (the default behaviour):
```
vpr ../arch.xml --device xc7a50t-test ../ff.blif --pack
``` 
produces:
```
Error 1: 
Type: Packing
File: /project/trees/vtr/vpr/src/pack/cluster.cpp
Line: 2012
Message: Can not find any logic block that can implement molecule.
	Atom D1 (.latch)
```

The work-around is to disable the filter with `--clustering_pin_feasibility_filter off`:
```
vpr ../arch.xml --device xc7a50t-test ../ff.blif --pack --clustering_pin_feasibility_filter off
```
which allows the design to pack successfully.

**Improved Congested/Impossible Cluster Routing Messages**
I've also improved the messages when a net is unroutable/congested during packing, to include both the netlist net and pins involved, and the architecture pins involved. This should make it easier to figure out where true connectivity problems exist.

For example in a case where no possible path exists VPR now produces:
```
No routing path from cluster-external source (LB_SOURCE) to cluster-internal sink (LB_SINK accessible via pins: ff[1].D[0]): needed for net 'n627' from pin 'n627.out[0]' to pin 'top^scevgep~18_FF_NODE.D[0]'
```
which shows both the architecture pin involved (`ff[1].D[0]`) and the associated net and connection pins in the netlist.

In the case of a congested (overused) pin, VPR will describe the architecture pins and netlist nets involved.

**Run-time Clustering Debug Flag**
I've also added a `--debug_clustering` command-line option which turns on additional clustering output to give more insight into what it is doing. This replaces the previous `#define DEBUG_FAILED_PACKING_CANDIDATES`.