verilog-to-routing/vtr-verilog-to-routing #279: Make `<fc>` and `<pinlocations>` optional in top level `<pb_type>` objects

mithro, 2018-02-02:
In my repository for SymbiFlow architecture definition repo, it has been useful to create `<pb_type>` objects recursively. 

You create the `<pb_type>` object and then load it into vpr to see the result. However, to get these definitions to load into vpr you need to make them top blocks level blocks -- which means they need to have `<fc>` and `<pinlocations>` tags.

When you later try to include this currently top level block inside another block you have to remove the `<fc>` and `<pinlocations>` tags and put them on the new including object. This makes it hard to switch back and forth between having objects at the top level and be included. 

Requiring these tags also makes it impossible to have two FPGA architectures, one with the object at the top level and one were it is included via another object. 

Lastly, when doing this both these tags have semi-reasonable defaults as well;
 * `<fc>` can just default to being connected to everything.
 * `<pinlocations>` can just default to `spread` option.

Maybe we could demote the current error to a warning?

mithro, 2018-02-02:
This is very much related to https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/272

kmurray, 2018-02-09:
Defaulting `<pinlocations>` to `spread` seems reasonable.

However it's less clear what a good `<fc>` default would be. 
I'm not a big fan of setting the default fractional `<fc>` to 1.0, since this is quite unrealistic.
FPGAs typically use a far lower Fc, but the exact value depends a lot on the architecture.

daveshah1, 2018-03-13:
Would a good compromise be specifying a default `<fc>` value inside the `device` section of the architecture, to be used if `<fc>` is not specified?

kmurray, 2018-03-13:
Setting a default in the `<device>` section seems like a good idea.

Perhaps something like:
```
<fc_default in_type="{frac|abs}" in_val="{int|float}" out_type="{frac|abs}" out_val="{int|float}"/>
```

I'd probably also drop the `default_` prefix from the `<fc>` attributes on `<pb_type>`s to avoid having multiple things named default:
```
 <fc in_type="{frac|abs}" in_val="{int|float}" out_type="{frac|abs}" out_val="{int|float}">
```
the existing `<fc_override>` sub-tags could remain the same.

Internally I'd still keep the existing per-top-level pb_type data structures and just initialize them to the device default if no `<fc>` tag is specified; this should restrict the code changes to libarchfpga (VPR would be unchanged).

As a methodology note, we have a lot of existing architecture files that we don't want to break. So whenever we change the format we should add a transformation pass to [upgrade_arch.py](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/scripts/upgrade_arch.py) to automatically upgrade legacy architecture files.

daveshah1, 2018-03-24:
This can be closed now that #308 has been merged.

closed 2018-03-26 by kmurray