verilog-to-routing/vtr-verilog-to-routing #283: Support xml forms of `.route` and `.place` files

mithro, 2018-02-06:
Currently vpr produces XML output in the `.net` file. The `.route` and `.place` files use a custom format.

It seems like it would make sense to support XML forms (**in addition to** the existing format) of `.route` and `.place` files.

The reasoning is;
 * Make it easier for programs to read that output.
 * Only have one output format style further processing tools need to deal with.
 * Match the `.net` file.
 * Make it easier to extend the format with extra info as needed.

mithro, 2018-02-06:
I think the first step would be to write up a little python script which converts the existing files into XML forms and then we can all agree on what that form the XML should look like and test the tools which do the ingestion. Once we are all happy with that we can look at extending vpr to write that format directly.

How does that sound?

kmurray, 2018-02-09:
Having the `.route` and `.place` files be easier to work with programmatically makes sense, and would be a good enhancement.

Starting with a script is a good to experiment and settle on good formats.

However after discussion with @vaughnbetz I think the preference would be to replace the existing `.route`/`.place` file formats completely with XML based once they are ready, rather than support both going forward.

mithro, 2018-02-09:
I agree that once we have settled on good formats, `vpr` should output that format directly rather then needing a conversion script.

On the vpr output side, I think you should use a generic serialization engine which can output xml, json, protobuf, human readable text, etc. `vpr` shouldn't really care much about the file format, just hand the right data structures to the serialization code. With this system all the serialization code can also be automatically generated from the structures that `vpr` is handing over. This would allow a bunch of code to be deleted from vpr and dramatically reducing the possibility of accidentally getting it wrong (plus you import and nice documentation for free too!).

closed 2025-05-15 by github-actions[bot]