verilog-to-routing/vtr-verilog-to-routing #285: Allow starting the vpr gui with the architecture description without doing pnr for a design

mithro, 2018-02-07:
#### Expected Behaviour

Able to load an architecture description into vpr and view the resulting description in the GUI (preferable with the routing graph shown).

(This will probably require setting a channel width?)

#### Current Behaviour

You currently need to have a design to pnr. If that design is not able to be placed or routed then the GUI doesn't pop up. This makes it hard to view inspect the architecture visually when things are broken.

kmurray, 2018-02-09:
This would be a good enhancement.

Some relevant history: VPR's clustering engine was originally a separate tool, which is why graphics opens only at placement. 

Also, since VPR searches for the minimum channel width by default there is no well defined channel width before routing (which is why the RR graph isn't drawn during placement). None of these are fundamental limitations.

mithro, 2018-03-10:
@kmurray What approach would you take to make this work?

kmurray, 2018-03-12:
It depends on what level of detail you want in graphics during the initial pre-packing stage. 

The actual drawing code should be easily re-useable. However it need two pieces of information which are not (generally) available until later in the design flow:
- Device grid dimensions (generally not known until after packing when the numbers/types of placeable clusters are known)
- RR Graph (generally not built until routing)

VPR basically has two graphics drawing modes:
- Placement mode (requires device grid, does not show RR graph)
- Routing mode (requires device grid and RR graph)

The main issue is that before packing we don't know what the device grid should be, since the mapping from netlist primitives to clustered blocks is non-trivial.

One potential approach would be to use a lower-bound estimate on the number/types of clustered blocks to come up with an initial device grid (e.g. based on the maximum number of sites for primitives within each cluster type), and use that size the initial device grid. It would then be straight forward to call the 'placement mode' graphics before packing, and have the graphics be updated during packing whenever the device is resized.

mithro, 2018-03-12:
@kmurray Could we maybe just support this only on [`fixed layout`](http://docs.verilogtorouting.org/en/latest/arch/reference/#tag-fpga-grid-layout-fixed_layout) descriptions? That way the grid size is already known?

kmurray, 2018-03-13:
If the choice is unambiguous that seems reasonable.

However keep in mind that potentially multiple `<fixed_layout>`s could be specified, and choosing which one to use before packing falls into the same problem.

closed 2025-05-15 by github-actions[bot]