verilog-to-routing/vtr-verilog-to-routing #288: Build failure on Arch Linux

daveshah1, 2018-02-12:
I am getting a build failure in Arch Linux, related to libtatum and its tbb dependency.

```[ 10%] Built target libtatum
[ 11%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/analysis/timing_reports.cpp.o
In file included from /home/david/symbiflow/vtr-verilog-to-routing/vpr/src/analysis/timing_reports.cpp:10:
In file included from /home/david/symbiflow/vtr-verilog-to-routing/vpr/src/timing/timing_info.h:6:
In file included from /home/david/symbiflow/vtr-verilog-to-routing/libs/EXTERNAL/libtatum/libtatum/tatum/analyzer_factory.hpp:10:
In file included from /home/david/symbiflow/vtr-verilog-to-routing/libs/EXTERNAL/libtatum/libtatum/tatum/graph_walkers.hpp:9:
In file included from /home/david/symbiflow/vtr-verilog-to-routing/libs/EXTERNAL/libtatum/libtatum/tatum/graph_walkers/ParallelLevelizedWalker.hpp:13:
In file included from /usr/include/tbb/combinable.h:24:
/usr/include/tbb/enumerable_thread_specific.h:621:55: error: called object type 'int' is not a function or function pointer
            void construct(void* where) {new(where) T(f());}
                                                      ^
/usr/include/tbb/enumerable_thread_specific.h:676:30: note: in instantiation of member function 'tbb::interface6::internal::construct_by_finit<unsigned long,
      int>::construct' requested here
                Constructor::construct(where);
                             ^
/usr/include/tbb/enumerable_thread_specific.h:661:34: note: in instantiation of member function 'tbb::interface6::internal::callback_leaf<unsigned long,
      tbb::interface6::internal::construct_by_finit<unsigned long, int> >::construct' requested here
            template<typename X> callback_leaf( const X& x ) : Constructor(x) {}
                                 ^
/usr/include/tbb/enumerable_thread_specific.h:689:35: note: in instantiation of function template specialization
      'tbb::interface6::internal::callback_leaf<unsigned long, tbb::interface6::internal::construct_by_finit<unsigned long, int> >::callback_leaf<int>' requested
      here
                return new(where) callback_leaf(x);
                                  ^
/usr/include/tbb/enumerable_thread_specific.h:863:80: note: in instantiation of function template specialization
      'tbb::interface6::internal::callback_leaf<unsigned long, tbb::interface6::internal::construct_by_finit<unsigned long, int> >::make<int>' requested here
            internal::callback_leaf<T,internal::construct_by_finit<T,Finit> >::make( tbb::internal::move(finit) )
                                                                               ^
/usr/include/tbb/combinable.h:46:46: note: in instantiation of function template specialization 'tbb::interface6::enumerable_thread_specific<unsigned long,
      tbb::cache_aligned_allocator<unsigned long>, tbb::ets_key_usage_type::ets_no_key>::enumerable_thread_specific<int>' requested here
        explicit combinable( finit _finit) : my_ets(_finit) { }
                                             ^
/home/david/symbiflow/vtr-verilog-to-routing/libs/EXTERNAL/libtatum/libtatum/tatum/graph_walkers/ParallelLevelizedWalker.hpp:44:37: note: in instantiation of
      function template specialization 'tbb::combinable<unsigned long>::combinable<int>' requested here
            tbb::combinable<size_t> unconstrained_counter(0);
                                    ^
1 error generated.
make[3]: *** [vpr/CMakeFiles/libvpr.dir/build.make:63: vpr/CMakeFiles/libvpr.dir/src/analysis/timing_reports.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:1109: vpr/CMakeFiles/libvpr.dir/all] Error 2
make[1]: *** [Makefile:141: all] Error 2
make: *** [Makefile:45: all] Error 2
```

* VTR revision used: 8.0.0-dev+9b00c2e63
* Operating System and version: Arch Linux
* Compiler version: clang 5.0.1
* TBB version: 2018_20171205

daveshah1, 2018-02-12:
This seems to be related to the libtatum update in aad0e96, reverting to b9405cc allows the build to run successfully.

kmurray, 2018-02-12:
Interesting, I'm not sure what the issue is. To the best of my knowledge `tbb::combinable<size_t> unconstrained_counter(0);` should be legal. The error seems to be pretty deep within the TBB library. The code compiles on my system (g++-5.4 on Ubuntu 16.04).

Could you provide the values of the [TBB version macros](https://www.threadingbuildingblocks.org/docs/help/reference/environment/version_information.html) on your system?
Mine reports:
```
TBB_INTERFACE_VERSION: 9002
TBB_INTERFACE_VERSION_MAJOR: 9
TBB_COMPATIBLE_INTERFACE_VERSION: 2
```

**Workaround**
A work around would be to explicitly disable TBB in your build configuration by setting the cmake parameter `TATUM_EXECUTION_ENGINE` to `serial` (instead of `auto` or `tbb`). This will remove any TBB dependency. This will of course disable parallel processing in libtatum, VPR's static timing analyzer.

daveshah1, 2018-02-12:
Thanks for your reply. My version macros are as follows, looks to be newer than yours.
```
#define TBB_VERSION_MAJOR 2018
#define TBB_VERSION_MINOR 0
#define TBB_INTERFACE_VERSION 10002
#define TBB_INTERFACE_VERSION_MAJOR TBB_INTERFACE_VERSION/1000
#define TBB_COMPATIBLE_INTERFACE_VERSION 2
```
The workaround seems to be working, so from my point of view that's fine - I'm a long way away from doing any kind of timing analysis...

kmurray, 2018-02-12:
A closer reading of the [TBB documentation](https://software.intel.com/en-us/node/506243) shows the `tbb::combinable` initializer must be a callable (rather than a constant, as was previously used). It seems TBB v4.4 was lenient and also accepted a constant, while TBB v2018.0 does not.

I've update Tatum to use a member function to perform the initialization. VTR 173a5ad should contain the fix.

closed 2018-02-12 by kmurray