verilog-to-routing/vtr-verilog-to-routing #289: Missing documentation for the "configurable" property on switch inside rr_graph.xml

mithro, 2018-02-13:
#### Expected Behaviour

[Description about `rr_graph.xml` file in the docs](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#routing-resource-graph-file-format-xml) should match the actually produced format.

#### Current Behaviour

I'm getting the following in my `rr_graph.xml` file;
```xml
	<switches>
		<switch id="0" type="mux" name="1" buffered="1" configurable="1">
			<timing R="551" Cin="7.70000012e-16" Cout="4.00000001e-15" Tdel="5.80000006e-11"/>
			<sizing mux_trans_size="2.63073993" buf_size="27.6459007"/>
		</switch>
		<switch id="1" type="mux" name="2" buffered="1" configurable="1">
			<timing R="551" Cin="7.70000012e-16" Cout="4.00000001e-15" Tdel="5.80000006e-11"/>
			<sizing mux_trans_size="2.63073993" buf_size="27.6459007"/>
		</switch>
		<switch id="2" type="mux" name="__vpr_delayless_switch__" buffered="1" configurable="1">
			<timing R="0" Cin="0" Cout="0" Tdel="0"/>
			<sizing mux_trans_size="0" buf_size="0"/>
		</switch>
	</switches>
```

There doesn't seem to be any documentation on the `configurable="1"` property. I guess this is describing if a switch can be configured as being open / closed?

I'm assuming that `configurable="1"` has only been added very recently, so the docs haven't been updated yet?

#### Your Environment

* VTR revision used: Currently using 80c80ca28a1d2810db0e80e00799bd2acbd43588

kmurray, 2018-02-14:
Yes, it's a new feature. Thanks for pointing out the missing documentation.

I've updated the docs in 53e7530.

closed 2018-02-14 by kmurray

mithro, 2018-02-14:
@kmurray What does "configurable" actually mean in this case? Does it mean that the switch is permanently on or permanently off (which I don't think would make much sense) or something else?

kmurray, 2018-02-14:
See the discussion [here](https://docs.verilogtorouting.org/en/latest/arch/reference/#switches).

Nodes in the RR graph represent conductors (i.e. wires) while the edges are the connections between them. 

VPR has historically assumed that all RR edges are configurable switches (i.e. can be 'turned-off' to disconnect two nodes). However there are some circuit structures where this does not hold.

Specific examples an in-line buffer/inverter or an electrical short between two nodes. In these cases the edges are non-configurable since they can not be disconnected/turned-off. By adding the configurable flag (and updated the router to respect it) VPR can now model these types structures.

Most edges in a general purpose FPGA routing architecture are configurable; but non-configurable edges do appear in more specialized structures like clock networks.

The flag defaults to true (matching previous behaviour) since this is a rather niche feature.

mithro, 2018-02-14:
This is actually very much a feature I was about to request. Way to go with the mind reading :-P

mithro, 2018-02-14:
Do they work yet?

kmurray, 2018-02-14:
The basic functionality is working. For example the `strong_flyover_wires` regression test  tests non-configurable buffers and electrical shorts.

The current implementation in the router is sub-optimal and slows it down more than it should. Addressing that is on my todo list.