verilog-to-routing/vtr-verilog-to-routing #291: Rewrite existing handprinted XML output to use pugixml instead

mithro, 2018-02-14:
#### Current Behaviour

Currently much of the XML output is done using hand coded statements using `fprintf` or `iostream`s. This leads to the code being a lot more complicated (you have to pass around tab depths and stuff), error prone (IE Your not escaping XML entities correctly in all places) and longer than it needs to be.

Newer XML output code is thankfully using the [pugixml](https://pugixml.org/) library.

I believe the following files currently output XML by hand:

 - [ ] [`vpr/src/power/power.cpp`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/power/power.cpp)
 - [ ] [`vpr/src/pack/output_clustering.cpp`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/output_clustering.cpp)
 - [ ] [`vpr/src/route/rr_graph_writer.cpp`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/rr_graph_writer.cpp)
 - [ ] [`libs/EXTERNAL/libcatch/catch.hpp`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/libs/EXTERNAL/libcatch/catch.hpp)

#### Possible Solution

It would be good if all XML output was done using the [pugixml](https://pugixml.org/) library.

Open questions;

 * How to test the XML output doesn't change?
 * More?

#### Context

The Verilog to Routing code base is reasonable old and had many authors. Like all old code bases it has areas which could use cleanup, refactoring and improvement. This issue is just to track just one idea on how things can be improved.

mithro, 2018-02-14:
@kc8apf is currently working on doing the [`vpr/src/pack/output_clustering.cpp`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/output_clustering.cpp) file.

kmurray, 2018-02-14:
Makes a lot of sense. We weren't using a (reasonable) XML library when most of that code was written. Since we're already using pugixml it makes sense to use it.

The only one I don't think we should change is `libs/EXTERNAL/libcatch/catch.hpp` since it's an [externally developed library](https://github.com/catchorg/Catch2).

> How to test the XML output doesn't change?

The most obvious would be to diff the before/after files (perhaps after formatting through xmllint for consistency).

However my main check would be that none of the [regression tests](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/README.developers.md#running-tests) fail. In particular the `vtr_reg_basic` and `vtr_reg_strong` tests.

kmurray, 2018-02-14:
I suppose another potential concern is that some of the output files (e.g. `rr_graph.xml`) can get pretty large (GBs); and I think pugixml stores the whole document in memory before it writes things out -- which could significantly increase VPR's peak memory use.

closed 2025-05-15 by github-actions[bot]