verilog-to-routing/vtr-verilog-to-routing #298: VPR is failing in vtr_reg_strong during ABC's formal verification

aarongraham9, 2018-03-05:
VPR is failing in `vtr_reg_strong` during ABC's `formal verification`. The two that are failing are `k6_frac_N10_40nm/always_true` and `k6_frac_N10_40nm/always_false` in `strong_func_formal_vpr`.

PLEASE NOTE: The new ABC requires the `readline-devel`/`libreadline-dev` package.

#### Expected Behaviour

I am expecting `k6_frac_N10_40nm/always_true` and `k6_frac_N10_40nm/always_false` in `strong_func_formal_vpr` to pass successfully.

#### Current Behaviour

VPR is failing in `vtr_reg_strong` during ABC's `formal verification`. The two that are failing are `k6_frac_N10_40nm/always_true` and `k6_frac_N10_40nm/always_false` in `strong_func_formal_vpr`.

#### Possible Solution

Change VPR to output the correct `.names` don't-know-don't-care truth table values.

#### Steps to Reproduce (for bugs)

Looking at “always_true” the original file that ABC is comparing against is `always_true.blif`:
 
```
.model always_true
.inputs i0 i1 i2 i3 i4 i5
.outputs o
 
.names i0 i1 i2 i3 i4 i5 o
------ 1
 
.end
 ```

Verses vpr’s `always_true_post_synthesis.blif`:
 
```
#BLIF generated by VPR 8.0.0-dev+9c4a02b-dirty from post-place-and-route implementation
.model always_true
.inputs i0 i1 i2 i3 i4 i5 
.outputs o 
 
#IO assignments
.names o_input_0_0 o
1 1
.names i0 i0_output_0_0
1 1
.names i1 i1_output_0_0
1 1
.names i2 i2_output_0_0
1 1
.names i3 i3_output_0_0
1 1
.names i4 i4_output_0_0
1 1
.names i5 i5_output_0_0
1 1
 
#Interconnect
.names i0_output_0_0 lut_o_input_0_3
1 1
.names i1_output_0_0 lut_o_input_0_0
1 1
.names i2_output_0_0 lut_o_input_0_4
1 1
.names i3_output_0_0 lut_o_input_0_1
1 1
.names i4_output_0_0 lut_o_input_0_2
1 1
.names i5_output_0_0 lut_o_input_0_5
1 1
.names lut_o_output_0_0 o_input_0_0
1 1
 
#Cell instances
.names lut_o_input_0_0 lut_o_input_0_1 lut_o_input_0_2 lut_o_input_0_3 lut_o_input_0_4 lut_o_input_0_5 lut_o_output_0_0 
 1
 
.end
```
 
The “New” ABC’s new Unbounded Sequential Equivalence Check (`dsec`) is dying with:

``` 
ABC command line: "dsec always_true.pre-vpr.blif always_true_post_synthesis.blif".
 
Abc_SopCheck: SOP has a mismatch between its cover size (0) and its fanin number (6).
NodeCheck: SOP check for node "lut_o_output_0_0" has failed.
Io_ReadBlifMv: The network check has failed for model always_true.
Reading network from file has failed.
 ```

If I manually change the last `.names` in `always_true_post_synthesis.blif` to:
 
```
#Cell instances
.names lut_o_input_0_0 lut_o_input_0_1 lut_o_input_0_2 lut_o_input_0_3 lut_o_input_0_4 lut_o_input_0_5 lut_o_output_0_0 
------ 1
```
 
Then manually re-run the “New” ABC’s `dsec` I get:
 
ABC command line: `dsec always_true.pre-vpr.blif always_true_post_synthesis.blif"`.
 
`Error: The network has no latches. Used combinational command "cec".`
 
And then running the fallback `cec` I get:
 
ABC command line: `cec always_true.pre-vpr.blif always_true_post_synthesis.blif`.
 
`Networks are equivalent after structural hashing.`
 
Which is what I believe we should expect. `always_false` seems to be the same.

#### Context

This causes `vtr_reg_strong` to fail.

#### Your Environment

* VTR revision used: Commit [`e430cbd`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/e430cbd4d0af5818e592e6ef2f92c4533385d326) based on [`master`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/master)'s [`05da4af`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/05da4af1825deda3874c41ce2f66ed004627eade)
* Operating System and version: 

```
uname -a
Linux 81f4a1ea7f5a 4.13.0-32-generic #35~16.04.1-Ubuntu SMP Thu Jan 25 10:13:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

```
cat /etc/*release
Fedora release 27 (Twenty Seven)
NAME=Fedora
VERSION="27 (Twenty Seven)"
ID=fedora
VERSION_ID=27
PRETTY_NAME="Fedora 27 (Twenty Seven)"
ANSI_COLOR="0;34"
CPE_NAME="cpe:/o:fedoraproject:fedora:27"
HOME_URL="https://fedoraproject.org/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=27
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=27
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
Fedora release 27 (Twenty Seven)
Fedora release 27 (Twenty Seven)
```

* Compiler version:
```
gcc --version
gcc (GCC) 7.2.1 20170915 (Red Hat 7.2.1-2)
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
g++ --version                                                                                                                                                                               
g++ (GCC) 7.2.1 20170915 (Red Hat 7.2.1-2)
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
clang --version
clang version 5.0.0 (tags/RELEASE_500/final)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

Commit [`e430cbd`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/e430cbd4d0af5818e592e6ef2f92c4533385d326) from the [`odin_dev`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/odin_dev) branch.

PLEASE NOTE: The new ABC requires the `readline-devel`/`libreadline-dev` package.

kmurray, 2018-03-05:
Thanks for the detailed report! This issue should be fixed in 2144015 on odin_dev.

Once all the outstanding issues with the new ABC are resolved we should open a pull request to review and merge all the changes.

closed 2018-03-05 by kmurray

aarongraham9, 2018-03-05:
You're welcome! Thanks @kmurray!

Sounds good! Will do!

aarongraham9, 2018-03-05:
Confirmed. Fixed. Thanks again @kmurray!