verilog-to-routing/vtr-verilog-to-routing #299: ACE 2 segfaults in vtr_reg_strong during strong_power after ABC update

aarongraham9, 2018-03-05:
ACE 2 segfaults in `vtr_reg_strong` during `strong_power` after ABC update and causes the `.power` files not to be created; this causes the `vtr_reg_strong` regression test to fail.

PLEASE NOTE: The ABC source has been updated.

PLEASE NOTE: The new ABC requires the `readline-devel`/`libreadline-dev` package.

#### Expected Behaviour

ACE 2 not to segfault in `vtr_reg_strong` during `strong_power`, after ABC update, and to generate the `.power` files so the `vtr_reg_strong` regression test passes successfully.

#### Current Behaviour

ACE 2 segfaults in `vtr_reg_strong` during `strong_power` after ABC update and causes the `.power` files not to be created; this causes the `vtr_reg_strong` regression test to fail.

`vtr_reg_strong` output snippet:

```
Wildcard in filename (*.power) matched 0 files.  There must be exactly one match.
Wildcard in filename (*.power) matched 0 files.  There must be exactly one match.
regression_tests/vtr_reg_strong/strong_power...[ERROR] Different number of entries in the two files.
```

#### Possible Solution

#### Steps to Reproduce (for bugs)

Pull commit [`e430cbd`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/e430cbd4d0af5818e592e6ef2f92c4533385d326) from the [`odin_dev`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/odin_dev) branch and run the `./run_reg_test.pl vtr_reg_strong` command in the VTR root directory.

#### Context

This causes `vtr_reg_strong` to fail.

#### Your Environment

* VTR revision used: Commit [`e430cbd`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/e430cbd4d0af5818e592e6ef2f92c4533385d326) based on [`master`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/master)'s [`05da4af`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/05da4af1825deda3874c41ce2f66ed004627eade)
* Operating System and version: 

```
uname -a
Linux 81f4a1ea7f5a 4.13.0-32-generic #35~16.04.1-Ubuntu SMP Thu Jan 25 10:13:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

```
cat /etc/*release
Fedora release 27 (Twenty Seven)
NAME=Fedora
VERSION="27 (Twenty Seven)"
ID=fedora
VERSION_ID=27
PRETTY_NAME="Fedora 27 (Twenty Seven)"
ANSI_COLOR="0;34"
CPE_NAME="cpe:/o:fedoraproject:fedora:27"
HOME_URL="https://fedoraproject.org/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=27
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=27
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
Fedora release 27 (Twenty Seven)
Fedora release 27 (Twenty Seven)
```

* Compiler version:
```
gcc --version
gcc (GCC) 7.2.1 20170915 (Red Hat 7.2.1-2)
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
g++ --version                                                                                                                                                                               
g++ (GCC) 7.2.1 20170915 (Red Hat 7.2.1-2)
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
clang --version
clang version 5.0.0 (tags/RELEASE_500/final)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

Commit [`e430cbd`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/e430cbd4d0af5818e592e6ef2f92c4533385d326) from the [`odin_dev`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/odin_dev) branch.

PLEASE NOTE: The new ABC requires the `readline-devel`/`libreadline-dev` package.

aarongraham9, 2018-03-05:
Update: Please use commit [`f281978`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/f28197845bc3cc4bc35efea85b1462d55cfd0a77) from the [`odin_dev`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/odin_dev) branch now as this removes some other `vtr_reg_strong` failures related to [#298](https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/298).

jgoeders, 2018-03-05:
OK, I'll look at this sometime this week.

aarongraham9, 2018-03-05:
Perfect! Thanks @jgoeders!

jgoeders, 2018-03-06:
What version of ABC have we changed to?  Is it a stable release?  

I tracked down the bug, and it stems from the fact that there is some unexpected behavior from the ABC functions that I haven't run into before.  I get the number of objects in the network using Abc_NtkObjNum(), but then when I iterate through them using Abc_NtkForEachObj() it actually reaches higher indices than there are supposedly objects.  

I'm not sure if this is intended behavior or not.  Any ABC experts here?  If not, I will reach out to Alan Mishchenko.

aarongraham9, 2018-03-06:
The "latest" stable release [abc20160717](https://bitbucket.org/alanmi/abc/commits/53f39c11b58d) is from August 2016. So we decided to use a commit [HG hash: 3933:918e46ecc85f](https://bitbucket.org/alanmi/abc/commits/918e46ecc85f800a65a699c30a1646d9994c9acf) (2017-10-04) from their rolling releases from the ABC [Mercurial repository](https://bitbucket.org/alanmi/abc).

Ok. That certainly seems like it would cause some bad behaviour. That's good to know; I believe in `ODIN II` we are also using the `Abc_NtkForEachObj()` macro.

I guess I am the resident expert, such as it is, as I did the ABC update; but I'm not sure either. It may be a good idea to reach out to Alan. Alternatively as the code I pulled is 4 months out of date we could pull a more recent version in and see if the issue is fixed?

kmurray, 2018-03-06:
You could also look at YOSYS to see which version of ABC its using (which is presumably reasonably stable snapshot).

aarongraham9, 2018-03-06:
From [YOSYS' Makefile](https://github.com/YosysHQ/yosys/blob/6f26695d9b596fe3b221334a93c5d94678582aea/Makefile) it looks like they are using [HG hash: 6e3c24b3308a](https://bitbucket.org/alanmi/abc/commits/6e3c24b3308aeba7325a9dbfb5d5d94f84ea5d17) which is from January 5th, 2018 (2018-01-05).

aarongraham9, 2018-03-09:
FYI: After updating to [HG hash: 6e3c24b3308a](https://bitbucket.org/alanmi/abc/commits/6e3c24b3308aeba7325a9dbfb5d5d94f84ea5d17) which is from January 5th, 2018 (2018-01-05) there are still the same two failures.

```
regression_tests/vtr_reg_strong/strong_power
-----------------------------------------
Current time: Mar-06 08:09 PM.  Expected runtime of next benchmark: 3 seconds
k6_frac_N10_mem32K_40nm/ch_intrinsics...failed: ace
Current time: Mar-06 08:09 PM.  Expected runtime of next benchmark: Unknown
k6_frac_N10_mem32K_40nm/diffeq1...failed: ace
```

```
Parsing test results... 
scripts/parse_vtr_task.pl -l /workspace/vtr-verilog-to-routing/vtr_flow/tasks/regression_tests/vtr_reg_strong/task_list.txt  
Wildcard in filename (*.power) matched 0 files.  There must be exactly one match.
Wildcard in filename (*.power) matched 0 files.  There must be exactly one match.
```

```
Test 'vtr_reg_strong' had 2 run failures
```

jgoeders, 2018-03-09:
OK, I'm waiting to hear back from Alan.  At minimum I will need to fix the code because they count objects from index 1 instead of 0 now.  And I'm waiting to hear if there are supposed to be the gaps of indices that I have observed.  Once I hear back I'll fix ACE.

daveshah1, 2018-03-17:
Just FYI, I am seeing this issue too on Arch Linux, but not all the time (and it seems to happen more often when running `ace` using `time`, as happens in the regression tests). Based on the core dump, the exact location of the segfault is
```
#0  0x000055f297a40988 in st_insert (table=0x55f29a0b80e0, key=0x55f29a1c9190 "p|\020\232\362U", value=0x55f29a3076b0 "")
    at /home/david/symbiflow/vtr-verilog-to-routing/abc_with_bb_support/src/misc/st/st.c:206
206	    FIND_ENTRY(table, hash_val, key, ptr, last);
```
called from
```
#1  0x000055f2979adfbf in main (argc=9, argv=0x7fffb481c698) at /home/david/symbiflow/vtr-verilog-to-routing/ace2/ace.c:366
```
I see that `hash_val` is equal to -12, which doesn't seem right based on a quick glance at the code (but could be a result of an invalid input from elsewhere).

daveshah1, 2018-03-17:
It looks like changing `ST_PTRHASH` from
```
#define ST_PTRHASH(x,size) ((int)((unsigned long)(x)>>2)%size)
```
to
```
#define ST_PTRHASH(x,size) ((int)(((unsigned long)(x)>>2)%size))
```
This fix seemed to have been made a while ago in mainline ABC

aarongraham9, 2018-03-22:
That is correct @daveshah1. I've updated the ABC on this branch to ABC HG Hash: 918e46ecc85f from 2017-10-04. We are hoping to get this newer ABC into the master branch here in VTR **_shortly_**.

jgoeders, 2018-03-23:
This should be fixed now in 3332d6b3a7465c885dd0fe7c41e48c0ed7a1a92f

aarongraham9, 2018-03-23:
Confirmed. Fixed. Thanks @jgoeders!

jeanlego, 2018-05-31:
the Issue still occurs on Arch linux sadly.

jeanlego, 2018-05-31:
fixed