verilog-to-routing/vtr-verilog-to-routing #302: New ABC has the ReadLine(.h) Header/Package as a Dependency

aarongraham9, 2018-03-05:
The new ABC has the ReadLine(.h) header/package as a dependency. Should this be added in CMake as a required librairy?

#### Your Environment

* VTR revision used: Commit [`f281978`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/f28197845bc3cc4bc35efea85b1462d55cfd0a77) based on [`master`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/master)'s [`05da4af`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/05da4af1825deda3874c41ce2f66ed004627eade)

* Operating System and version:
```
uname -a
Linux 81f4a1ea7f5a 4.13.0-32-generic #35~16.04.1-Ubuntu SMP Thu Jan 25 10:13:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

```
cat /etc/*release
Fedora release 27 (Twenty Seven)
NAME=Fedora
VERSION="27 (Twenty Seven)"
ID=fedora
VERSION_ID=27
PRETTY_NAME="Fedora 27 (Twenty Seven)"
ANSI_COLOR="0;34"
CPE_NAME="cpe:/o:fedoraproject:fedora:27"
HOME_URL="https://fedoraproject.org/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=27
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=27
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
Fedora release 27 (Twenty Seven)
Fedora release 27 (Twenty Seven)
```

* Compiler version:
```
gcc --version
gcc (GCC) 7.2.1 20170915 (Red Hat 7.2.1-2)
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
g++ --version                                                                                                                                                                               
g++ (GCC) 7.2.1 20170915 (Red Hat 7.2.1-2)
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
clang --version
clang version 5.0.0 (tags/RELEASE_500/final)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

Commit [`f281978`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/f28197845bc3cc4bc35efea85b1462d55cfd0a77) from the [`odin_dev`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/odin_dev) branch.

aarongraham9, 2018-03-05:
The new ABC has the ReadLine(.h) header/package as a dependency. Should this be added in CMake as a required librairy? This has already been done, added as an installed package, in the Dockerfile.

@kmurray do you have any thoughts on this?

kmurray, 2018-03-05:
I think we should try to minimize the number of strictly required dependencies. Having lots of external dependencies makes it more difficult for users to get up and running.

I think ABC can be compiled without readline?
If so my preference would be to have the CMakeLists detect whether readline is installed on the system. If it is available we can build ABC with readline support; otherwise we build ABC without readline support.

aarongraham9, 2018-03-05:
Ok. Fair enough. I'll see if I can convince CMake to make the appropriate checks and call ABC with the `ABC_USE_NO_READLINE=1` flag.

aarongraham9, 2018-03-06:
Apparently ABC itself doesn't check if readline exists on the system: [readline library dependency not checked before compilation](https://bitbucket.org/alanmi/abc/issues/59/readline-library-dependency-not-checked) so I'll have to figure out how to do it for them in our build system and pass that down into their CMakeLists.txt.

aarongraham9, 2018-03-06:
(Failing because of improved, and not yet updated, QoR values)

This has been resolved in a9639a28b4697a1b2ca6aaba4d5df184289e27fc.

commit a1649d2b (2018-03-06): Have VTR's Makefile Detect the ReadLine Header/Package

Have VTR's Makefile Detect if the ReadLine Header/Package is Present
 on the System and Call ABC's CMakeLists.txt with the Appropriate Flags

Resolves #302.
commit a9639a28 (2018-03-14): Have VTR's Makefile Detect the ReadLine Header/Package

Have VTR's Makefile Detect if the ReadLine Header/Package is Present
 on the System and Call ABC's CMakeLists.txt with the Appropriate Flags

Resolves #302.
closed 2018-03-06 by aarongraham9