verilog-to-routing/vtr-verilog-to-routing #310: Segfault in alloc_and_load_track_top_pin_lookup

mithro, 2018-03-20:
Architecture file -> https://gist.github.com/mithro/ef23fac984eb229a0300bbda22fca6b6

```
tansell@tansell:~/work/catx/testing/vpr-testarch-1$ gdb --args /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/vpr arch.xml --device 2x4 wire.eblif --route_chan_width 6 --write_rr_graph wire.r
r_graph.xml                                                                                                                                                                                     
GNU gdb (GDB) 7.9-gg19                                                                                                                                                           
Copyright (C) 2015 Free Software Foundation, Inc.                                                                                                                             
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>                                                                             
This is free software: you are free to change and redistribute it.                                              
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"                                                                                                           
and "show warranty" for details.                                                                                
This GDB was configured as "x86_64-grtev4-linux-gnu".                                                                                                                                    
Type "show configuration" for configuration details.                                                    
                                                                                                                                                                                                         
<http://go/gdb-home FAQ: http://go/gdb-faq Email: c-toolchain-team IRC: gdb>                      
Find the GDB manual and other documentation resources online at:                                                                                             
<http://www.gnu.org/software/gdb/documentation/>.                                                                                                                 
For help, type "help".                                                                                                                                      
Type "apropos word" to search for commands related to "word".                                                                                      
Reading symbols from /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/vpr...done.
Unable to determine compiler version.     
Skipping loading of libstdc++ pretty-printers for now.
Non-google3 binary detected.        
(gdb) run                        
Starting program: /usr/local/google/home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/vpr arch.xml --device 2x4 wire.eblif --route_chan_width 6 --write_rr_graph wire.rr_graph.xml
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Dwarf Error: wrong version in compilation unit header (is 0, should be 2, 3, or 4) [in module /usr/lib/debug/.build-id/cb/88f1fe155855433a969d01f052923b299701b8.debug]

VPR FPGA Placement and Routing.
Version: 8.0.0-dev+vpr-7.0.5-5697-g967a755f2-dirty
Revision: vpr-7.0.5-5697-g967a755f2-dirty
Compiled: 2018-03-19T23:28:56 (debug build)
Compiler: GNU 6.3.0 on Linux-4.9.0-4-amd64 x86_64
University of Toronto
vtr-users@googlegroups.com
This is free open source code under MIT license.

VPR was run with the following command-line:
/usr/local/google/home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/vpr arch.xml --device 2x4 wire.eblif --route_chan_width 6 --write_rr_graph wire.rr_graph.xml
```

```
Resource usage...
        Netlist      0  blocks of type: EMPTY
        Architecture 0  blocks of type: EMPTY
        Netlist      1  blocks of type: BLK_IG-IBUF
        Architecture 6  blocks of type: BLK_IG-IBUF
        Netlist      1  blocks of type: BLK_IG-OBUF
        Architecture 6  blocks of type: BLK_IG-OBUF
        Netlist      1  blocks of type: BLK_TI-LUTFF
        Architecture 12 blocks of type: BLK_TI-LUTFF

Device Utilization: 0.05 (target 1.00)
        Block Utilization: 0.00 Type: EMPTY
        Block Utilization: 0.17 Type: BLK_IG-IBUF
        Block Utilization: 0.17 Type: BLK_IG-OBUF
        Block Utilization: 0.08 Type: BLK_TI-LUTFF

Placement

Starting placement delay look-up...
Starting build routing resource graph...

Program received signal SIGFPE, Arithmetic exception.
0x0000555555b6fbd6 in vtr::NdMatrixProxy<int, 2ul>::operator[] (this=0x7fffffff27b0, index=0) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:42
42                  size_t next_dim_stride = dim_stride_ / dim_sizes_[idim_ + 1];
(gdb) bt
#0  0x0000555555b6fbd6 in vtr::NdMatrixProxy<int, 2ul>::operator[] (this=0x7fffffff27b0, index=0) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:42
#1  0x0000555555b6deb7 in vtr::NdMatrixProxy<int, 2ul>::operator[] (this=0x7fffffff27b0, index=0) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:53
#2  0x0000555555b694d9 in alloc_and_load_track_to_pin_lookup (pin_to_track_map=..., Fc=..., type_width=1, type_height=1, num_pins=1, max_chan_width=6, num_seg_types=2)
    at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp:2190
#3  0x0000555555b6240d in build_rr_graph (graph_type=GRAPH_UNIDIR, L_num_types=4, types=0x555556206db8, grid=..., nodes_per_chan=0x5555561d7938 <g_vpr_ctx+248>, sb_type=WILTON, Fs=3, 
    switchblocks=std::vector of length 0, capacity 0, num_seg_types=2, num_arch_switches=2, segment_inf=0x5555562059b0, global_route_switch=1, wire_to_arch_ipin_switch=0, delayless_switch=1, 
    R_minW_nmos=6065.52002, R_minW_pmos=18138.5, base_cost_type=DELAY_NORMALIZED, trim_empty_channels=false, trim_obs_channels=false, directs=0x0, num_directs=0, wire_to_rr_ipin_switch=0x7fffffffbee4, 
    num_rr_switches=0x5555561d79b0 <g_vpr_ctx+368>, Warnings=0x7fffffff30cc) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp:544
#4  0x0000555555b613d8 in create_rr_graph (graph_type=GRAPH_UNIDIR, num_block_types=4, block_types=0x555556206db8, grid=..., nodes_per_chan=0x5555561d7938 <g_vpr_ctx+248>, num_arch_switches=2, 
    det_routing_arch=0x7fffffffbeb0, segment_inf=0x5555562059b0, base_cost_type=DELAY_NORMALIZED, trim_empty_channels=false, trim_obs_channels=false, directs=0x0, num_directs=0, 
    num_rr_switches=0x5555561d79b0 <g_vpr_ctx+368>, Warnings=0x7fffffff30cc) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp:275
#5  0x0000555555b0a304 in alloc_routing_structs (router_opts=..., det_routing_arch=0x7fffffffbeb0, segment_inf=0x5555562059b0, directs=0x0, num_directs=0)
    at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/place/timing_place_lookup.cpp:230
#6  0x0000555555b09efa in compute_delay_lookup_tables (router_opts=..., det_routing_arch=0x7fffffffbeb0, segment_inf=0x5555562059b0, chan_width_dist=..., directs=0x0, num_directs=0)
    at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/place/timing_place_lookup.cpp:110
#7  0x0000555555b09d4f in alloc_lookups_and_criticalities (chan_width_dist=..., router_opts=..., det_routing_arch=0x7fffffffbeb0, segment_inf=0x5555562059b0, directs=0x0, num_directs=0)
    at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/place/timing_place.cpp:91
#8  0x0000555555af08b3 in try_place (placer_opts=..., annealing_sched=..., chan_width_dist=..., router_opts=..., det_routing_arch=0x7fffffffbeb0, segment_inf=0x5555562059b0, directs=0x0, num_directs=0)
    at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/place/place.cpp:375
#9  0x0000555555a6abdc in vpr_place (vpr_setup=..., arch=...) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:539
#10 0x0000555555a6a9c4 in vpr_place_flow (vpr_setup=..., arch=...) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:520
#11 0x0000555555a69c50 in vpr_flow (vpr_setup=..., arch=...) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:309
#12 0x0000555555a595af in main (argc=9, argv=0x7fffffffdb38) at /usr/local/google/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/main.cpp:69
(gdb) print dim_stride_
$1 = 0
(gdb) print dim_sizes_
$2 = (const size_t *) 0x7fffffff2dd8
(gdb) print idim_
$3 = 3
(gdb) print dim_sizes_[idim_+1]
$4 = 0
(gdb) 
```

mithro, 2018-03-21:
Ending up with an rr_graph node with no inputs and outputs is the actual bug I was investigating when I hit this bug.

kmurray, 2018-03-22:
I've fixed the issue, which related to the handling of pins with no connected tracks, as of c5128ab.

This architecture used separate input/output blocks, which meant that the IBUF block only had an output pin, and the OBUF block only had an input pin. The previous code implicitly expected that each block type had at least one input and at least one output pin. 

Historically VPR only supported a single I/O block type (a restriction removed relatively recently in fb085bd). As a result all of our traditional VPR architectures use a shared input/output block (and hence always had at least one input and one output pin).

closed 2018-03-22 by kmurray