verilog-to-routing/vtr-verilog-to-routing #311: initial FPGA size based on the number of blocks

jeanlego, 2018-03-23:
I'm just curious if starting the packing process at a larger size than the default one based on a threshold on the number of blocks and nets, would it speed up the packing process?

excerpt:

After removing unused inputs...
	total blocks: 155575, total nets: 157117, total inputs: 36, total outputs: 33
Begin prepacking.
Finish prepacking.
Using inter-cluster delay: 1.36132e-09
Not enough resources expand FPGA size to (8 x 8)

[...]

Not enough resources expand FPGA size to (105 x 105)

kmurray, 2018-03-23:
The actual time spent resizing the FPGA is actually very quick. During packing the FPGA model is effectively just the device grid: a 2D matrix of block types/locations.

Re-sizing the grid is very cheap compared to actually packing a cluster (which involves expensive operations like routing connections etc.). So I don't expect a better initial solution would make much of a run-time difference.

jeanlego, 2018-03-23:
Got it! I have very little knowledge of anything past Odin so thank you!

closed 2018-03-23 by jeanlego

kmurray, 2018-03-23:
No worries!