verilog-to-routing/vtr-verilog-to-routing #318: Assert failure when writing atom_netlist.cleaned.echo.blif

mithro, 2018-04-14:
## Assert failure when writing atom_netlist.cleaned.echo.blif

Currently trying to determine if the problem is the eblif or my architecture file. My architecture file works with other blif files.

I'm getting the following assert failure when using an `eblif` input.
```
# /home/tansell/work/catx/vtr-verilog-to-routing/vpr/vpr /home/tansell/work/catx/vtr/testarch/devices/lutff/arch.merged.xml --device 2x4 --timing_analysis off rot.eblif --echo_file on --gen_post_synthesis_netlist on

....

Compress circuit   
Compress circuit took 0.00 seconds          
Verify circuit
Verify circuit took 0.00 seconds
/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/atom_netlist_utils.cpp:141 print_netlist_as_blif: Assertion 'pins.size() == 1' failed.
                                                   
Program received signal SIGABRT, Aborted.  
```

When inside gdb I get the following traceback,
```gdb
(gdb) bt
#0  0x00007ffff624afcf in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff624c3fa in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x0000555555a70ae8 in vtr::Assert::handle_assert (expr=0x555555e3296b "pins.size() == 1", file=0x555555e32890 "/home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/atom_netlist_utils.cpp", line=141, 
    function=0x555555e332e0 <print_netlist_as_blif(_IO_FILE*, AtomNetlist const&)::__func__> "print_netlist_as_blif", msg=0x0) at /home/tansell/work/catx/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_assert.h:137
#3  0x0000555555c3e512 in print_netlist_as_blif (f=0x555556218630, netlist=...) at /home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/atom_netlist_utils.cpp:141
#4  0x0000555555c3d906 in print_netlist_as_blif (filename="atom_netlist.cleaned.echo.blif", netlist=...) at /home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/atom_netlist_utils.cpp:38
#5  0x0000555555c8437f in read_and_process_circuit (circuit_format=e_circuit_format::EBLIF, circuit_file=0x7fffffffbbd8 "rot.eblif", user_models=0x0, library_models=0x55555620b680, should_absorb_buffers=true, 
    should_sweep_dangling_primary_ios=true, should_sweep_dangling_nets=true, should_sweep_dangling_blocks=true, should_sweep_constant_primary_outputs=false)
    at /home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/read_circuit.cpp:68
#6  0x0000555555a6c379 in vpr_init (argc=11, argv=0x7fffffffd9e8, options=0x7fffffffbf70, vpr_setup=0x7fffffffba60, arch=0x7fffffffbe80)
    at /home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:255
#7  0x0000555555a5c253 in main (argc=11, argv=0x7fffffffd9e8) at /home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/main.cpp:63
(gdb) p clk_net
""
(gdb) p *blk_model
{name = 0x55555620b860 ".latch", inputs = 0x55555620e0f8, outputs = 0x55555620bba8, instances = 0x0, used = 0, pb_types = 0x555556207790, next = 0x55555620b740, index = 2}

```

The eblif is generated from the following verilog file using Yosys using the command line `yosys -p "synth; opt_clean; write_blif -attr -cname -conn -impltf -param build/testarch/2x4/rot.eblif" rot.v`
```verilog
module top(input clk, output D1, output D2, output D3, output D4, output D5);

   reg ready = 0;
   reg [3:0] rot;

   always @(posedge clk) begin
      if (ready)
         begin
            rot <= {rot[2:0], rot[3]};
         end
      else
        begin
           ready <= 1;
           rot <= 4'b0001;
        end
   end

   assign D1 = rot[0];
   assign D2 = rot[1];
   assign D3 = rot[2];
   assign D4 = rot[3];
   assign D5 = 1;
endmodule // top
```

Yosys generates the following `eblif` file.
```blif
# Generated by Yosys 0.7+501 (git sha1 588ce0e3, x86_64-conda_cos6-linux-gnu-gcc 8a21ab48 -fPIC -Os)

.model top
.inputs clk
.outputs D1 D2 D3 D4 D5
.names rot[3] ready $0\rot[3:0][0]
1- 1
-0 1
.names rot[0] ready $0\rot[3:0][1]
11 1
.names rot[1] ready $0\rot[3:0][2]
11 1
.names rot[2] ready $0\rot[3:0][3]
11 1
.latch $true ready re clk 0
.latch $0\rot[3:0][0] rot[0] re clk 2
.latch $0\rot[3:0][1] rot[1] re clk 2
.latch $0\rot[3:0][2] rot[2] re clk 2
.latch $0\rot[3:0][3] rot[3] re clk 2
.conn rot[0] D1
.conn rot[1] D2
.conn rot[2] D3
.conn rot[3] D4
.conn $true D5
.end
```

#### Your Environment
```
VPR FPGA Placement and Routing.
Version: 8.0.0-dev+vpr-7.0.5-5756-gcd10a5664-dirty
Revision: vpr-7.0.5-5756-gcd10a5664-dirty
Compiled: 2018-04-14T13:16:07 (debug build)
Compiler: GNU 6.3.0 on Linux-4.9.0-4-amd64 x86_64
University of Toronto
vtr-users@googlegroups.com
This is free open source code under MIT license.
```

kmurray, 2018-04-16:
I've found the problem.

The root cause is that `$true` is un-driven in your `.eblif` input, which causes it to be swept by VPR's netlist cleaning. This then means VPR can't identify the input (D pin) of the latch when writing the atom netlist.

So the fix is to add:
```
.names $true
 1
```
to the `.eblif`.

I'll also look at seeing if I can get VPR to produce a helpful warning/error message.

mithro, 2018-04-16:
Ahh.. 

If we remove `-impltf` from the yosys command line it will add the definition for `$true` and similar.

From http://www.clifford.at/yosys/cmd_write_blif.html
>    -impltf
>        do not write definitions for the $true, $false and $undef wires.

mithro, 2018-04-16:
Should `vpr` check that all named nets have a driver?

kmurray, 2018-04-16:
The default is to sweep away un-driven nets, since it's assumed they have no meaning.

It is not uncommon to have 'imperfect' netlists, which is why VPR does some cleaning passes after loading the netlist. The end-user can control this behaviour with the various `--sweep`* options.

100c71f converts the assertion failure to a warning in this particular case (no net connected to a required `.latch` pin).

In 4a20e7b I've added a `--verbose_sweep` option to VPR which shows what is swept away by netlist cleaning, which would diagnose that `$true` was swept. Hopefully this will help diagnose these types of issues in the future.

closed 2018-04-16 by kmurray