verilog-to-routing/vtr-verilog-to-routing #325: VPR has issues when carry chain inputs are involved in a blocks internal routing

mithro, 2018-04-23:
If you have a case where the carry input is shared with another path then the packer can assign non-carry signals to the carry input and the routing will fail when trying to drive a signal from the fabric onto this signal.

An example could be an architecture like the following;
[image]

The `I[3]` pin to the LUT can be driven via either the `CIN` input or the `I[3]` pin. This is modeled with something like;
```xml
<pb_type name="TILE">
 <!-- Tile Inputs -->
 <input  name="IN  " num_pins="4" />
 <!-- Tile Outputs -->
 <output name="OUT " num_pins="1" />
 <!-- Carry pins -->
 <input  name="CIN " num_pins="1" />
 <output name="COUT" num_pins="1" />

 <pb_type blif_model=".names" class="lut" name="LUT" num_pb="1">
  <input equivalent="false" name="in" num_pins="4" port_class="lut_in"/>
  <output equivalent="false" name="out" num_pins="1" port_class="lut_out"/>
  <delay_matrix in_port="LUT.in" out_port="LUT.out" type="max">
    10e-12
    10e-12
    10e-12
    10e-12
  </delay_matrix>
 </pb_type>
 <interconnect>
  <direct name="LUT.in[0]" input="TILE.IN[0]"          output="LUT.in[0]" />
  <direct name="LUT.in[1]" input="TILE.IN[1]"          output="LUT.in[1]" />
  <direct name="LUT.in[2]" input="TILE.IN[2]"          output="LUT.in[2]" />
  <mux    name="LUT.in[3]" input="TILE.CIN TILE.IN[3]" output="LUT.in[3]" />
  <direct name="TILE.OUT"  input="LUT.out"             output="TILE.OUT"  />
  <direct name="TILE.COUT" input="LUT.out"             output="TILE.COUT" />
 </interconnect>
</pb_type>
```

The order of the `<mux>` seems to affect if `TILE.CIN` is chosen to have a signal assigned to it. `TILE.CIN TILE.IN[3]` will end up with `TILE.CIN` in the packed output. `TILE.IN[3] TILE.CIN` seems to end up with `TILE.IN[3]` being in the packed output.

#### Expected Behaviour
The carry paths are only used for carry functionality.

#### Possible Solution

@kmurray suggested that the fc values should be read from the architecture and the carry's shouldn't be marked as connected to the fabric.

#### Context

FYI This type of structure appears on the iCE40 carry chain;
[image]
and also on the carry path in the Artix-7;
[image]

mithro, 2018-04-23:
@jhol - FYI

kmurray, 2018-04-23:
Thanks for the detailed report @mithro!

Currently, the packer always assumes it can 'route outside' the cluster from any block output pin to any block input pin.  This is modelled as an (effectively infinite capacity) node in the logic block internal RR graph, which has edges to/from all input/output pins.

While this true on traditional architectures, it isn't always true with less flexible structures like carry chains. The architectures with carry-chains which VTR has targeted up till now have only connected the Cin/Cout pins to dedicated carry-chain related routing which didn't interface with the general routing network (which is why they are unaffected by this issue).

I suspect the fix is to make the packer aware that certain input pins (like Cin/Cout) are _not_ connected to the general routing. This would amount to improving the packer's external routing model to account for the Fc specifications (including `<fc_override>`s). In this case that would remove the edges between the Cin/Cout block pins, forcing the packer to only use them for carry-chain connections.

> The order of the <mux> seems to affect if TILE.CIN is chosen to have a signal assigned to it. TILE.CIN TILE.IN[3] will end up with TILE.CIN in the packed output. TILE.IN[3] TILE.CIN seems to end up with TILE.IN[3] being in the packed output.

Likely this ordering effects which pin the packer is looking at first (or perhaps how it breaks ties between pins).

mithro, 2018-04-23:
You can find a test case here -> https://gist.github.com/mithro/17a4a3b6cc22c6a340f23933a109ca5d

The [arch.xml is here](https://gist.githubusercontent.com/mithro/17a4a3b6cc22c6a340f23933a109ca5d/raw/f7b55362e2c99e7bcd21354cae52331b64a44a65/arch.xml).

`make DEVICE=2x4 lut.echo` produces;
```
Warning 46: Empty heap occurred in get_heap_head.
Warning 47: Some blocks are impossible to connect in this architecture.
Cannot route from BLK_IG-IBUF.I[0] (rr_node: 0 type: SOURCE ptc: 0 xlow: 1 ylow: 1) to BLK_TI-LUTFF.CIN[0] (rr_node: 62 type: SINK ptc: 6 xlow: 3 ylow: 3) to -- no possible path
Failed to route connection from 'I3' to 'O' for net 'I3'
Routing failed.
```

kmurray, 2018-04-24:
Thanks @mithro, I've been able to reproduce the issue.

kmurray, 2018-04-26:
As I suggested above, I've confirm that specific case can be fixed by considering <fc_overrides> when building the Packer's model of external routing.

However this correspondingly breaks the routing of _all_ carry chains, since the CIN/COUT nodes are not connected to the Packer's global external SOURCE/SINK nodes.

Likely the best approach is to have multiple distinct external SOURCE/SINK nodes (e.g. ones for global routing, ones for carry-chains; potentially others in general). 

The hard part becomes having the packer decide which external source/sink any specific connection should use.
This is challenging since as the packer is running the start/endpoints of a connection may not have been packed into a block yet -- so the packer doesn't know which source/sink to use. 

In general you can come up with architectures which require different decisions, and can't be determined solely based upon the netlist structure. My current thinking is to punt on the issue in general (it may not be decidable _a prior_) and just require `<pack_patterns>` to guide the packer to the correct solution.

mithro, 2018-04-26:
@kmurray Would this extend to clock's also being a seperate SOURCE/SINK + routing network too?

kmurray, 2018-04-27:
It depends on what the external connectivity of the clock pins. If the clock pins are accessible via the intra-block routing fabric then then a separate SOUCE/SINK shouldn't be needed. However, if they are independent the likely yes.

vaughnbetz, 2018-07-12:
As discussed in the VTR meeting, Tim to take Kevin's code in progress and run basic, strong, and nightly regressions and then run ice40 and debug various issues.  We don't know of any issues, but haven't tested fully so expect some.

kmurray, 2018-07-12:
I've pushed my current WIP to the [generalized_pack_patterns](https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/generalized_pack_patterns) branch.

Apologies in advance for the messy state of the code! The changes propagated to more code than I originally anticipated.

qaarah, 2018-07-24:
Hi, Can I  give the post-placed .blif file generated by another CAD tool to VPR for routing and final ouput?
Thanks

vaughnbetz, 2018-07-24:
Yes.  You will need to create a .net file and .place file, and pass their names to vpr on the command line, and specify --route so only the router runs.

-net_file <file>
Path to packed user circuit in net format
--place_file <file>
Path to final placement file

mithro, 2018-09-14:
For people playing along at home, the following diagrams show the various carry chain issues;
[image]

(This diagram [can be found here](https://docs.google.com/drawings/d/1qREImoaUjWDSsnbimDu-Mig3_M9hTr-6q-zbPd2VPEM/edit?usp=sharing).)

closed 2025-05-15 by github-actions[bot]