verilog-to-routing/vtr-verilog-to-routing #331: "found non-adjacent segments" in rr_graph when using bidirectional tracks

mithro, 2018-04-24:
#### Current Behaviour
```
Fc Actual Values: type = BLK_BB-VPR_PAD, pin = 1, seg = 3 (local), Fc_out = 5, Fc_in = 0.
Fc Actual Values: type = BLK_BB-VPR_PAD, pin = 1, seg = 4 (direct), Fc_out = 5, Fc_in = 0.
Warning 84: in check_rr_node: RR node: 116 type: OPIN location: (2,2) pin: 51 pin_name: BLK_TL-PLB.FCOUT[0] has no out-going edges.
Warning 85: in check_rr_node: RR node: 336 type: OPIN location: (3,2) pin: 51 pin_name: BLK_TL-PLB.FCOUT[0] has no out-going edges.
Build routing resource graph took 0.037802 seconds
Best routing used a channel width factor of 26.
Routing took 1.42 seconds

Checking to ensure routing is legal...
Error 1: 
Type: Routing
File: /home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/route/check_route.cpp
Line: 143
Message: in check_route: found non-adjacent segments in traceback while checking net 1:
  RR node: 964 type: CHANY location: (4,2) <-> (4,3) track: 23 len: 1 seg_type: direct dir: BI_DIR
  RR node: 512 type: CHANX location: (1,1) <-> (2,1) track: 3 len: 1 seg_type: global dir: BI_DIR

Makefile:94: recipe for target 'big_xor.echo' failed
make: *** [big_xor.echo] Error 1
```

#### Steps to Reproduce (for bugs)

Using the following [architecture file](https://gist.githubusercontent.com/mithro/17a4a3b6cc22c6a340f23933a109ca5d/raw/387fb462286d45ce7e06e02f371ec8533831e6dc/arch.xml) with `DEVICE=test4` should cause the failure to occur.

(You can also just do `make DEVICE=test4 lut.echo` with the [Full environment](https://gist.githubusercontent.com/mithro/17a4a3b6cc22c6a340f23933a109ca5d) @ rev 387fb462286d45ce7e06e02f371ec8533831e6dc)

#### Your Environment

```
VPR FPGA Placement and Routing.
Version: 8.0.0-dev+vpr-7.0.5-5770-g5234d7d81
Revision: vpr-7.0.5-5770-g5234d7d81
Compiled: 2018-04-24T14:56:05 (release build)
Compiler: GNU 6.3.0 on Linux-4.9.0-5-amd64 x86_64
University of Toronto
vtr-users@googlegroups.com
This is free open source code under MIT license.
```

kmurray, 2018-04-25:
I'm still digging into it, but it initially seems like the 'longline' segment length may be the issue. The longline option isn't something that is widely used which is also why I'm suspicious.

For me, not using longline allows the lut benchmark to route without error. Hopefully that will give you a work around.

mithro, 2018-04-26:
Awesome! Thank you so much.

Will test shortly.