verilog-to-routing/vtr-verilog-to-routing #332: No switch is specified for the ipin cblock

mithro, 2018-04-25:
#### Current Behaviour
```
Error 1: 
Type: Architecture file
File: /home/tansell/work/catx/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp
Line: 701
Message: No switch is specified for the ipin cblock, check if there is an error in arch file
Makefile:34: recipe for target 'wire.disp' failed
```

#### Steps to Reproduce (for bugs)
```xml
<?xml version="1.0"?>
<!-- set: ai sw=1 ts=1 sta et -->
<architecture xmlns:xi="http://www.w3.org/2001/XInclude">
  <models/>
  <layout>
    <fixed_layout height="2" name="JOHN" width="3">
      <single priority="1" type="MYIN"  x="0" y="0"/>
      <single priority="1" type="MYLUT" x="1" y="0"/>
      <single priority="1" type="MYOUT" x="2" y="0"/>
    </fixed_layout>
  </layout>
  <complexblocklist>
    <pb_type name="MYIN">
      <input name="DATIN" num_pins="2"/>
      <pb_type name="OUTPUT" blif_model=".output" num_pb="2">
        <input name="outpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="MYIN.DATIN[0]" name="O0" output="OUTPUT[0].outpad"/>
        <direct input="MYIN.DATIN[1]" name="O1" output="OUTPUT[1].outpad"/>
      </interconnect>
      <pinlocations pattern="custom">
        <loc side="right">MYIN.DATIN[0] MYIN.DATIN[1]</loc>
      </pinlocations>
      <fc in_type="frac" in_val="1.0" out_type="frac" out_val="1.0"/>
    </pb_type>

    <pb_type name="MYOUT">
      <output name="DATOUT" num_pins="1"/>
      <pb_type name="INPUT" blif_model=".input" num_pb="1">
        <output name="inpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct name="2" input="INPUT.inpad" output="MYOUT.DATOUT"/>
      </interconnect>
      <pinlocations pattern="custom">
        <loc side="left">MYOUT.DATOUT[0]</loc>
      </pinlocations>
      <fc in_type="frac" in_val="1.0" out_type="frac" out_val="1.0"/>
    </pb_type>

    <pb_type name="MYLUT">
      <input name="DATIN" num_pins="1"/>
      <output name="DATOUT" num_pins="1"/>
      <pb_type name="LUT" blif_model=".names" class="lut" num_pb="1">
        <input name="in" num_pins="1" port_class="lut_in"/>
	<output name="out" num_pins="1" port_class="lut_out"/>
	<delay_matrix in_port="LUT.in" out_port="LUT.out" type="max">10e-12</delay_matrix>
      </pb_type>
      <interconnect>
        <direct name="1" input="MYLUT.DATIN" output="LUT.in"/>
        <direct name="2" input="LUT.out" output="MYLUT.DATOUT"/>
      </interconnect>
      <pinlocations pattern="custom">
	<loc side="right">MYLUT.DATIN[0]</loc>
	<loc side="left">MYLUT.DATOUT[0]</loc>
      </pinlocations>
      <fc in_type="frac" in_val="1.0" out_type="frac" out_val="1.0"/>
[... 223 lines cut ...]
       with multiple fan-ins, so right now there's just one. May change in the future, in which case we'd need to 
       return a representative switch */
    if (switch_fanin[wire_to_arch_ipin_switch].count(UNDEFINED)) {
        /* only have one ipin cblock switch. OK. */
        (*wire_to_rr_ipin_switch) = switch_fanin[wire_to_arch_ipin_switch][UNDEFINED];
    } else if (switch_fanin[wire_to_arch_ipin_switch].size() != 0) {
        vpr_throw(VPR_ERROR_ARCH, __FILE__, __LINE__,
                "Not currently allowing an ipin cblock switch to have multiple fan-ins");
    } else {
        vpr_throw(VPR_ERROR_ARCH, __FILE__, __LINE__,
                "No switch is specified for the ipin cblock, check if there is an error in arch file");
    }
```

#### Your Environment
```
VPR FPGA Placement and Routing.
Version: 8.0.0-dev+vpr-7.0.5-5770-g5234d7d81
Revision: vpr-7.0.5-5770-g5234d7d81
Compiled: 2018-04-24T14:56:05 (release build)
Compiler: GNU 6.3.0 on Linux-4.9.0-5-amd64 x86_64
University of Toronto
vtr-users@googlegroups.com
This is free open source code under MIT license.
```

kmurray, 2018-04-25:
So the root cause of this, is that the architecture as specified never produces any connection blocks.

With the original layout:
```xml
  <layout>
    <fixed_layout height="2" name="JOHN" width="3">
      <single priority="1" type="MYIN"  x="0" y="0"/>
      <single priority="1" type="MYLUT" x="1" y="0"/>
      <single priority="1" type="MYOUT" x="2" y="0"/>
    </fixed_layout>
  </layout>
```
The blocks are all at `y=0` which puts them along the bottom edge ([perimeter](url)) of the device, causing them to have no vertical routing channels next to them:
[image]
Since the pins are assigned to the left/right sides they have no vertical channels to connect to, means no connection blocks are built (causing the error message).

A potential fix is changing the device grid so that the block pins are adjacent to routing channels by shifting them all up (`y=1`):
```xml
    <fixed_layout height="3" name="JOHN" width="3">
      <single priority="1" type="MYIN"  x="0" y="1"/>
      <single priority="1" type="MYLUT" x="1" y="1"/>
      <single priority="1" type="MYOUT" x="2" y="1"/>
    </fixed_layout>
```

which produces:
[image]

Which will build and route successfully.

I will converted error to a warning and clarified the message. This allows the RR graph checker to run (which reports numerous issues with disconnected tracks etc.), and will allow users to see the RR graph in graphics. This extra information should help them understand why this is occurring.

For example, on the original architecture VPR would now produce:
```
Attempting to route at 2 channels (binary search bounds: [-1, -1])
Starting build routing resource graph...
Warning 4: No switch found for the ipin cblock in RR graph. Check if there is an error in arch file, or if no connection blocks are being built in RR graph
Warning 5: Track 0 had no switches
Warning 6: Track 1 had no switches
Warning 7: in check_rr_node: RR node: 10 type: CHANX location: (1,0) track: 0 len: 0 seg_type: L4 dir: INC_DIR has no out-going edges.
Warning 8: in check_rr_node: RR node: 11 type: CHANX location: (1,0) track: 1 len: 0 seg_type: L4 dir: DEC_DIR has no out-going edges.
Warning 9: in check_rr_graph: fringe node 2 IPIN at (0,0) has no fanin.
         This is possible on a fringe node based on low Fc_out, N, and certain lengths.
```

kmurray, 2018-04-25:
Changes made in 0dc8802.

mithro, 2018-04-25:
@kmurray In this case I guess we need to pad the Y=1 side of the architecture with empties so that we end up with segments in the space between the `X=0 || X=1` and the `X=1 || X=2`.

mithro, 2018-04-25:
This issue would also have been very obvious if I had seen it in the GUI. How were you able to get that graphic? (It that what the converting to warnings did?)

I think this shows maybe why we need to put some effort into https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/285

kmurray, 2018-04-25:
Yes, the conversion to a warning allows things to continue to the GUI.