verilog-to-routing/vtr-verilog-to-routing #335: rr_graph reader: review unexpected channel direction edges

mcmasterg, 2018-04-27:
TLDR: we should either make the drawing correct or reject the rr_graph file

Conceptually an rr_graph is laid out in a two dimensional XY fashion with blocks/tracks occupying various rows and columns. While edges are expected to follow with this in mind, there is no requirement that they do so.

For example, see the attached case. I've created bi-directional edges (ie one with chanx => chany and one with chany => chanx) between the topmost CHANX and the center right CHANY. However, the CHANY is DEC_DIR, which confuses the GUI since I'm trying to connect it as a source. This resulted in the GUI assuming the source came from above and drawing a bad line.

Screenshot:
[image]

Sample project with this behavior: [unexpected_chan_dir_edge.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/1953714/unexpected_chan_dir_edge.zip)

#### Expected Behaviour

We should either make the drawing correct (it should appear as a bi-directional link in the sample project) or reject the rr_graph file

#### Current Behaviour

The graph is accepted without warning or error. Note: there are unrelated warnings for the unconnected nodes in the sample project.

#### Possible Solution

If the XY grid is a core concept, suggest rejecting this rr_graph

#### Steps to Reproduce (for bugs)

1. Unzip attached zip
2. cd build
3. vpr ../arch.xml --device JOHN --timing_analysis off --sweep_dangling_blocks off --route_chan_width 2 ../wire.blif --disp on --read_rr_graph rr_graph2.xml

#### Context

Testing rr_graph python library. I will probably add checks on my end for now not to generate these until I have a demonstrated need for them

#### Your Environment

* VTR revision used: 8.0.0-dev+vpr-7.0.5-5781-ga144521de
* Operating System and version:
* Compiler version:

mithro, 2018-04-29:
@kmurray Is there any reason why there are `CHANX` and `CHANY` rather than just `CHAN`s which just happens to run in `X` or `Y` directions? It feels like channels should just go from an arbitrary `(X1, Y1) -> (X2, Y2)`.

It is likely we need "diagonal" channels in the future (in fact the iCE40 has connections between `(X, Y) -> (X+1, Y+1)` and `(X, Y) -> (X-1, Y-1)`, etc).

kmurray, 2018-04-30:
> Is there any reason why there are CHANX and CHANY rather than just CHANs which just happens to run in X or Y directions?

I agree this would be more general. 

That VPR uses CHANX/CHANY since that was how it was originally coded. Doing so simplifies the code since it only needs to concern itself with the horizontal/vertical cases (rather than all possible arbitrary orientations).

The actual RR node class (`t_rr_node`) has support for unique position coordinates (xlow, ylow, xhigh, yhigh), and so could support diagonal RR nodes.

The real dependencies on CHANX/CHANY fall within the drawing code (draws differently depending the type, since it assumes CHANX/CHANY are non-zero length nodes which are horizontally/vertically aligned).

The RR graph generator (`build_rr_graph()`) likely also has some similar dependencies.

So while generalizing to a non-horizontal/vertical CHAN is possible it would require updating the above code to robustly handle such cases -- a non-trivial amount of work.

> It is likely we need "diagonal" channels in the future (in fact the iCE40 has connections between (X, Y) -> (X+1, Y+1) and (X, Y) -> (X-1, Y-1), etc).

I'd recommend modeling this as two RR nodes (one CHANX, one CHANY) with a _non-configurable edge_ between them. 

For example, something like:

```xml
<switch id="1" name="electrical_short" buffered="0" configurable="0">
  <timing R="0" Cin="0" Cout="0" Tdel="0"/>
</switch>
...
<node id="0" type="CHANX" direction="INC_DIR" capacity="1">
  <loc xlow="X" ylow="Y" xhigh="X+1" yhigh="Y"/>
</node>
<node id="1" type="CHANY" direction="INC_DIR" capacity="1">
  <loc xlow="X+1" ylow="Y" xhigh="X+1" yhigh="Y+1"/>
</node>
...
<edge src_node="0" sink_node="1" switch_id="1">
```

Since the switch is non-configurabel (`configurable="0"`), the router will effectively treat the above as a single RR node from (X,Y) -> (X+1, Y+1).

This L-shaped implementation is also more aligned with the physical layout, since modern process technologies do not allow arbitrary routing, but restrict you to either 1D or 2D X/Y routing.

closed 2025-05-15 by github-actions[bot]