verilog-to-routing/vtr-verilog-to-routing #350: ODIN II and constants in module instantiation

gorski123, 2018-06-06:
While running a Verilog file through Odin, Odin kept crashing. I ran it in a debugger and tracked it down to an oassert statement on line 676 of ast_util.cpp. The line is as follows:

oassert(bit < var_node->types.number.binary_size);

It aborts here since bit and types.number.binary_size are actually equal.

While I am not sure exactly what the code is attempting to do here, I was able to determine that the problem is related to a module instantiation in my verilog file, specifically this one:

bramctrlsimple
	#(
		.SIZE(8),
		.ADDR_WIDTH(3),
		.DATA_WIDTH(6)
	)
	mem15(
		.clk(clk),
		.mem_done(mem15_0_mem_done),
	        .mem_access(1'b0),
                .mem_data_out(mem15_0_mem_data_in),
	        .mem_addr_in(mem15_0_mem_addr_out),
	        .mem_data_in(6'b000000),
		.mem_req(mem15_0_mem_req)
	);

The issue appears to be the line where I attempt to map the constant 6'b000000 to mem_data_in . I am pretty sure this is the offending line for a few reasons: 

first of all the variables in the oassert statement (bit and types.number.binary_size) both equal the length of my constant here (6). If I change the constant length, the variables in the oassert change in line. 

second, and probably more telling, if I remove the mapping to a constant, then ODIN goes on to crash on the next constant mapping in my verilog file. 

So my question is does ODIN support mapping to constants directly in the module instantiation? I only ask as the verilog file I am attempting to convert is automatically generated and thousands of lines long and it will be a bit of a chore to remove the constant mappings.

I am running in Ubuntu 16.04.1 LTS

Thanks so much,
Jason

kmurray, 2018-06-06:
Thanks for the detailed bug report!

@jeanlego @PanosNB @aarongraham9 Odin devs, any thoughts?

jeanlego, 2018-06-12:
seems to be related to issue #213 where the binary value is not being sized appropriately. I think the commit fixing this was reverted a while back since some benchmarks broke when doing so. I'm looking into this. 

this reproduces the bug

[v_runGA.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/2092608/v_runGA.v.txt)

jeanlego, 2018-06-16:
should be fixed! tested with v_runGA.txt and it compiles.
please report back if it works for you @gorski123  or you have any issues so I can close this.

also tested with Odin reg_micro, full, syntax and vtr reg basic and strong
thanks. :)

closed 2018-06-16 by jeanlego