verilog-to-routing/vtr-verilog-to-routing #351: How to model "mutually exclusive" fan out in a rr_graph.xml

mithro, 2018-06-06:
In the iCE40 architecture, we have the following;
[image]

This means that `sp4_v_t47` can be connected to `sp4_h_r_10` **or** `sp4_h_l_47` **or** `sp4_b_b_10` and so forth.

This all works fine if the connections here were unidirectional, IE `sp4_h_r_10` drives `sp4_h_l_47` (as once you drive a signal onto `sp4_h_l_47` you can't drive a second signal onto it). 

**However**, in the iCE40 cases these are **pass gate**. Thus when we create the edges we end up with an edge which goes from `sp4_h_r_10->sp4_h_l_47` **and** `sp4_h_r_10<-sp4_h_l_47`. This leads to the problem that vpr will try to connect `sp4_v_t_47` to `sb_v_b_10` via `sp4_h_r_10`. This does not work as `sp4_h_r_10` can only be connected to **one** of these other wires at a time.

What we need is a way to specify a group of  fan out switches that are mutually exclusive. IE You can only select **one** in the group.

I tried to come up with a way to represent this in the current `rr_graph.xml` format, but just was not able to figure out how to do it.

Do you have any ideas?

If not -- any suggestions on how we should implement this in VPR?

mithro, 2018-06-06:
@kmurray - FYI

vaughnbetz, 2018-06-07:
Kevin and I just discussed this. We think it is doable with an rr-graph that has an extra node between the wires connected by these (shared configuration) switches.  That node also needs to a new tag of "don't re-expand" after it is used to route one connection on a multi-fanout net.  It would look like:
```
sp4_v_t_47 -------> special node ------> sp4_h_r_10
                      ^
 ....                 |
sp4_h_l_47 -----------+
```

The special node needs to be marked as not re-expandable as listed above.  We think we can do that with a new tag specified on wires (optional, re-expandable, default = true) and a small amount of code in the router.  Then the rr-graph you create and would have these nodes in it, and the architecture file would have that special wire.  We think it could be a length 0 wire, but haven't tested if length 0 wires are allowed (if not we could make them allowed).

daveshah1, 2018-06-10:
Really sorry to have wasted everyone's time here. It turns out this was ultimately a mistake in icestorm due to a misunderstanding of Lattice's tool output, and it appears all of the "routing" bits are actually unidirectional after a bit of testing.

Not sure if other FPGAs might have this kind of architecture, and/or whether some of the "routing" bits really are bidirectional switches, though.

mithro, 2018-06-10:
I think we probably do want to support this feature in the future. As it is not longer on the critical path it is probably a low priority.

closed 2025-04-29 by vaughnbetz