verilog-to-routing/vtr-verilog-to-routing #355: Clean up get_rr_node_index a bit

vaughnbetz, 2018-06-07:
get_rr_node_index  has some odd code that should be cleaned up.

1. For rr_node index lookups of CHANX types, the x and y index is swapped when rr_node_indices is loaded:

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/rr_graph2.cpp#L1255,L1264

Then when the lookup is done in get_rr_node_index, the indices are swapped again. 

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/rr_graph2.cpp#L1354,L1356

We should stop doing that, and delete the comment about it.

2. We allow lookups by side for CHANX and CHANY, even though side makes no sense for them.  We should at least always go to side 0, rather than the enumerated type at sides[0] as that will break if sides[0] is not equal to 0.

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/rr_graph2.cpp#L1338,L1341

#### Proposed Behaviour
Same as current, but less confusing code under the hood.

#### Current Behaviour
None.

#### Possible Solution
See above.

#### Context
Code cleanup.

vaughnbetz, 2018-06-07:
@kmurray

vaughnbetz, 2025-04-29:
@soheilshahrouz : I think you just fixed this -- is that right?

soheilshahrouz, 2025-05-16:
Closing the issue as #3042 was merged.

closed 2025-05-16 by soheilshahrouz