verilog-to-routing/vtr-verilog-to-routing #359: Two copies of the blif output code?

mithro, 2018-06-09:
There seems to be blif output code in two places under `vpr/src`;

in `vpr/src/base/atom_netlist_utils.cpp`

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/base/atom_netlist_utils.cpp#L39-L365

in `vpr/src/pack/output_blif.cpp`
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/output_blif.cpp#L85-L352

The code looks very similar, is there any reason we need both?

HackerFoo, 2019-12-06:
`vpr/src/pack/output_blif.cpp` was removed in https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/280634e5cc9789bd8addf071eaac45cb32e9dd01

HackerFoo, 2019-12-06:
The remaining BLIF writers seem to vary in what they support:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/bd6a0acffd7d8521db5ea6d0f7a3c88d1c5b35d3/vpr/src/base/atom_netlist_utils.cpp#L193-L208
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/bd6a0acffd7d8521db5ea6d0f7a3c88d1c5b35d3/vpr/src/base/netlist_writer.cpp#L1235-L1238

And `print_netlist_as_blif()` adds artificial buffers:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/bd6a0acffd7d8521db5ea6d0f7a3c88d1c5b35d3/vpr/src/base/atom_netlist_utils.cpp#L96-L111

closed 2025-05-15 by github-actions[bot]