verilog-to-routing/vtr-verilog-to-routing #362: Convert rr_edge index to StrongId

mithro, 2018-06-12:
#### Current Behaviour

Currently the rr_edge values taken and returned by edge methods on the rr_node object are the basic type `short`. This makes it easy to accidentally use the value in bad ways.

#### Proposed Behaviour

It would be good if the value was an "opaque" type. IE A value which you can get and hand back to the rr_node class, but you have no idea how it is implemented.

When debug is enabled, it could even check that the edge is for the right rr_node object, while in release it would internally still be a short.

#### Possible Solution

VtR already has the idea of ["StrongId" class](libs/libvtrutil/src/vtr_strong_id.h) which prevents implicit casting. This can probably be used here too.

Another option is something like this;
```cpp
#ifndef DEBUG
        struct t_edge_idx {
            unsigned short i;
        };
        static_assert (sizeof(t_edge_idx) == sizeof(short), "t_edge_idx size broken");
#else
        struct t_edge_idx {
            t_rr_node* owner;
            unsigned short i;
        };
#endif
        constexpr static t_edge_idx edge_invalid { std::numeric_limits<unsigned short>::max() };
        static_assert (std::is_trivial<t_edge_idx>::value, "t_edge_idx is not trivial object!");
        static_assert (std::is_pod<t_edge_idx>::value, "t_edge_idx is not plain old data type!");
```

#### Context

It should be easier to accidently use the edge index value in the wrong way.

This would also make it much easier to change how edges are handled in the future.

kmurray, 2018-06-12:
This would be an excellent code clean-up.

To be consistent with the rest of the code base I think using a `vtr::StrongId` is the preferred approach.

In addition to the rr-edge IDs I'd also suggest converting the rr-node IDs to vtr::StrongId.

The main code changes would be:
- Define `vtr::StrongId` types for RR edges and nodes (see `atom_netlist_fwd.h` for an example). Something like:
```cpp
struct rr_edge_id_tag;
typedef vtr::StrongId<rr_edge_id_tag,short> RREdgeId;

struct rr_node_id_tag;
typedef vtr::StrongId<rr_node_id_tag,int> RRNodeId;
```
- Update `DeviceContext::rr_nodes` to use a `vtr::vector` indexed by `RRNodeId`
- Update `t_rr_node` to use `RREdgeId`s instead of plain `shorts` 
- Update the relevant code which uses plain ints/shorts for node/edge Ids to use `RRNodeId`/`RREdgeId` (probably most of the work).

**Unique Edge IDs**
It is likely out-of-scope for this change, but in the longer term it would be good to update the RR graph data structures to use unique IDs for edges (instead of having them indexed relative to a specific node).

This would be similar to how the Netlist data structures work. I've found that using a unique ID for each element in the graph is conceptually simpler, more convenient and leads to cleaner code.

mustafabbas, 2018-06-13:
I agree that this is a good idea for rr_nodes as well. And I work on converting rr_nodes to use StrongIds and for the std::vector I created to use the vtr::vector instead.

closed 2025-05-15 by github-actions[bot]