verilog-to-routing/vtr-verilog-to-routing #368: Floating point exception in get_bidir_opin_connections with multi-tile blocks

mithro, 2018-06-19:
When using blocks which have a height > 1, I get the following exception;
```
Program received signal SIGFPE, Arithmetic exception.
0x0000555555bfcd68 in vtr::NdMatrix<std::vector<int, std::allocator<int> >, 4ul>::operator[] (this=0x555556352070, index=6)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:294
294                 size_t dim_stride = this->size() / this->dim_size(0);
(gdb) bt
#0  0x0000555555bfcd68 in vtr::NdMatrix<std::vector<int, std::allocator<int> >, 4ul>::operator[] (this=0x555556352070, index=6)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:294
#1  0x0000555555c07183 in get_bidir_opin_connections (i=3, j=4, ipin=6, from_rr_node=24, rr_edges_to_create=std::set with 0 elements, 
    opin_to_track_map=std::vector of length 4, capacity 4 = {...}, L_rr_node_indices=std::vector of length 7, capacity 7 = {...}, seg_details=0x555556309be0)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/rr_graph2.cpp:779
#2  0x0000555555bf306a in build_bidir_rr_opins (i=3, j=3, side=RIGHT, L_rr_node=0x555556349ba8, L_rr_node_indices=std::vector of length 7, capacity 7 = {...}, 
    opin_to_track_map=std::vector of length 4, capacity 4 = {...}, Fc_out=std::vector of length 4, capacity 4 = {...}, created_rr_edges=std::set with 40 elements = {...}, 
    seg_details=0x555556309be0, grid=..., directs=0x0, num_directs=0, clb_to_clb_directs=0x0, num_seg_types=2)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp:1209
#3  0x0000555555bf2994 in alloc_and_load_rr_graph (num_nodes=532, L_rr_node=0x555556349ba8, num_seg_types=2, seg_details=0x555556309be0, chan_details_x=..., chan_details_y=..., 
    track_to_pin_lookup=std::vector of length 4, capacity 4 = {...}, opin_to_track_map=std::vector of length 4, capacity 4 = {...}, switch_block_conn=..., sb_conn_map=0x0, grid=..., Fs=3, 
    sblock_pattern=0x0, Fc_out=std::vector of length 4, capacity 4 = {...}, Fc_xofs=..., Fc_yofs=..., L_rr_node_indices=std::vector of length 7, capacity 7 = {...}, max_chan_width=10, 
    wire_to_ipin_switch=0, delayless_switch=1, directionality=BI_DIRECTIONAL, Fc_clipped=0x7fffffff280f, directs=0x0, num_directs=0, clb_to_clb_directs=0x0)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp:1107
#4  0x0000555555befdbf in build_rr_graph (graph_type=GRAPH_BIDIR, L_num_types=4, types=0x5555562f9218, grid=..., nodes_per_chan=0x5555562c8978 <g_vpr_ctx+248>, sb_type=WILTON, Fs=3, 
    switchblocks=std::vector of length 0, capacity 0, num_seg_types=2, num_arch_switches=2, segment_inf=0x5555562fca40, global_route_switch=1, wire_to_arch_ipin_switch=0, 
    delayless_switch=1, R_minW_nmos=6065.52002, R_minW_pmos=18138.5, base_cost_type=DELAY_NORMALIZED, trim_empty_channels=false, trim_obs_channels=false, directs=0x0, num_directs=0, 
    wire_to_rr_ipin_switch=0x7fffffffbd24, num_rr_switches=0x5555562c89f0 <g_vpr_ctx+368>, Warnings=0x7fffffff2ecc)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp:574
#5  0x0000555555beeb0e in create_rr_graph (graph_type=GRAPH_BIDIR, num_block_types=4, block_types=0x5555562f9218, grid=..., nodes_per_chan=0x5555562c8978 <g_vpr_ctx+248>, 
    num_arch_switches=2, det_routing_arch=0x7fffffffbcf0, segment_inf=0x5555562fca40, base_cost_type=DELAY_NORMALIZED, trim_empty_channels=false, trim_obs_channels=false, directs=0x0, 
    num_directs=0, num_rr_switches=0x5555562c89f0 <g_vpr_ctx+368>, Warnings=0x7fffffff2ecc)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp:278
#6  0x0000555555b94844 in alloc_routing_structs (router_opts=..., det_routing_arch=0x7fffffffbcf0, segment_inf=0x5555562fca40, directs=0x0, num_directs=0)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/place/timing_place_lookup.cpp:229
#7  0x0000555555b9446c in compute_delay_lookup_tables (router_opts=..., det_routing_arch=0x7fffffffbcf0, segment_inf=0x5555562fca40, chan_width_dist=..., directs=0x0, num_directs=0)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/place/timing_place_lookup.cpp:110
#8  0x0000555555b942c1 in alloc_lookups_and_criticalities (chan_width_dist=..., router_opts=..., det_routing_arch=0x7fffffffbcf0, segment_inf=0x5555562fca40, directs=0x0, num_directs=0)
    at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/place/timing_place.cpp:91
#9  0x0000555555b7a247 in try_place (placer_opts=..., annealing_sched=..., chan_width_dist=..., router_opts=..., det_routing_arch=0x7fffffffbcf0, segment_inf=0x5555562fca40, directs=0x0, 
    num_directs=0) at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/place/place.cpp:377
#10 0x0000555555aea581 in vpr_place (vpr_setup=..., arch=...) at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:555
#11 0x0000555555aea369 in vpr_place_flow (vpr_setup=..., arch=...) at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:536
#12 0x0000555555ae9580 in vpr_flow (vpr_setup=..., arch=...) at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:317
#13 0x0000555555ad70af in main (argc=11, argv=0x7fffffffdb48) at /home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/main.cpp:69
(gdb) 
```

It seems to have something to do with having output pins not on the 0x0 offset. If I *just* have input pins on 0x1 offset, then things work fine.

Command line;
```
vpr arch.xml --device 2x4 wire.eblif --route_chan_width 10 --min_route_chan_width_hint 1 --write_rr_graph rr_graph.virt.xml
```

arch.xml
```xml
<?xml version="1.0"?>
<!-- set: ai sw=1 ts=1 sta et -->
<architecture xmlns:xi="http://www.w3.org/2001/XInclude">
  <models/>
  <layout>
    <fixed_layout height="3" name="1x1" width="7">
      <!--
      0123456
    2 .......
    1 .I.X.O.
    0 .......
    -->
      <col priority="10" startx="0" type="EMPTY"/>
      <col priority="10" startx="1" type="BLK_IG-IBUF"/>
      <col priority="10" startx="2" type="EMPTY"/>
      <col priority="10" startx="3" type="BLK_TI-TILE"/>
[... 242 lines cut ...]
    <switch_block fs="3" type="wilton"/>
    <chan_width_distr>
      <x distr="uniform" peak="1.000000"/>
      <y distr="uniform" peak="1.000000"/>
    </chan_width_distr>
  </device>
  <switchlist>
    <switch Cin=".77e-15" Cout="4e-15" R="551" Tdel="58e-12" buf_size="27.645901" mux_trans_size="2.630740" name="mux" type="mux"/>
  </switchlist>
  <segmentlist>
    <segment Cmetal="22.5e-15" Rmetal="101" freq="0.250000" length="4" name="span" type="bidir">
      <sb type="pattern">1 1 1 1 1</sb>
      <cb type="pattern">1 1 1 1</cb>
      <wire_switch name="mux"/>
      <opin_switch name="mux"/>
    </segment>
    <segment Cmetal="22.5e-15" Rmetal="101" freq="0.750000" length="1" name="local" type="bidir">
      <sb type="pattern">1 1</sb>
      <cb type="pattern">1</cb>
      <wire_switch name="mux"/>
      <opin_switch name="mux"/>
    </segment>
  </segmentlist>
</architecture>
```

mithro, 2018-06-19:
When I do a `export CMAKE_PARAMS="-DVTR_ASSERT_LEVEL=4"` then I get the following;
```
/home/tansell/github/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:288 operator[]: Assertion 'this->dim_size(0) > 0' failed (Can not index into size zero dimension).
```

mithro, 2018-06-19:
FYI - This is with `vpr-7.0.5-5991-g18ff5e715`

I've also tried doing;
```xml
 <pinlocations pattern="custom">
  <loc side="right" xoffset="0" yoffset="0">BLK_TI-TILE.CLK[0] BLK_TI-TILE.CLK[1] BLK_TI-TILE.OUT[0] BLK_TI-TILE.OUT[1] BLK_TI-TILE.IN[0] BLK_TI-TILE.IN[1] BLK_TI-TILE.IN[2] BLK_TI-TILE.IN[3] BLK_TI-TILE.IN[4] BLK_TI-TILE.IN[5]</loc>
  <loc side="right" xoffset="0" yoffset="1">BLK_TI-TILE.CLK[0] BLK_TI-TILE.CLK[1] BLK_TI-TILE.OUT[0] BLK_TI-TILE.OUT[1] BLK_TI-TILE.IN[0] BLK_TI-TILE.IN[1] BLK_TI-TILE.IN[2] BLK_TI-TILE.IN[3] BLK_TI-TILE.IN[4] BLK_TI-TILE.IN[5]</loc>
  <loc side="left" xoffset="0" yoffset="0">BLK_TI-TILE.CLK[0] BLK_TI-TILE.CLK[1] BLK_TI-TILE.OUT[0] BLK_TI-TILE.OUT[1] BLK_TI-TILE.IN[0] BLK_TI-TILE.IN[1] BLK_TI-TILE.IN[2] BLK_TI-TILE.IN[3] BLK_TI-TILE.IN[4] BLK_TI-TILE.IN[5]</loc>
  <loc side="left" xoffset="0" yoffset="1">BLK_TI-TILE.CLK[0] BLK_TI-TILE.CLK[1] BLK_TI-TILE.OUT[0] BLK_TI-TILE.OUT[1] BLK_TI-TILE.IN[0] BLK_TI-TILE.IN[1] BLK_TI-TILE.IN[2] BLK_TI-TILE.IN[3] BLK_TI-TILE.IN[4] BLK_TI-TILE.IN[5]</loc>
  <loc side="top" xoffset="0" yoffset="0">BLK_TI-TILE.CLK[0] BLK_TI-TILE.CLK[1] BLK_TI-TILE.OUT[0] BLK_TI-TILE.OUT[1] BLK_TI-TILE.IN[0] BLK_TI-TILE.IN[1] BLK_TI-TILE.IN[2] BLK_TI-TILE.IN[3] BLK_TI-TILE.IN[4] BLK_TI-TILE.IN[5]</loc>
  <loc side="top" xoffset="0" yoffset="1">BLK_TI-TILE.CLK[0] BLK_TI-TILE.CLK[1] BLK_TI-TILE.OUT[0] BLK_TI-TILE.OUT[1] BLK_TI-TILE.IN[0] BLK_TI-TILE.IN[1] BLK_TI-TILE.IN[2] BLK_TI-TILE.IN[3] BLK_TI-TILE.IN[4] BLK_TI-TILE.IN[5]</loc>
  <loc side="bottom" xoffset="0" yoffset="0">BLK_TI-TILE.CLK[0] BLK_TI-TILE.CLK[1] BLK_TI-TILE.OUT[0] BLK_TI-TILE.OUT[1] BLK_TI-TILE.IN[0] BLK_TI-TILE.IN[1] BLK_TI-TILE.IN[2] BLK_TI-TILE.IN[3] BLK_TI-TILE.IN[4] BLK_TI-TILE.IN[5]</loc>
  <loc side="bottom" xoffset="0" yoffset="1">BLK_TI-TILE.CLK[0] BLK_TI-TILE.CLK[1] BLK_TI-TILE.OUT[0] BLK_TI-TILE.OUT[1] BLK_TI-TILE.IN[0] BLK_TI-TILE.IN[1] BLK_TI-TILE.IN[2] BLK_TI-TILE.IN[3] BLK_TI-TILE.IN[4] BLK_TI-TILE.IN[5]</loc>
 </pinlocations>
```

kmurray, 2018-06-19:
Should be fixed in 7af6420.

closed 2018-06-19 by kmurray