verilog-to-routing/vtr-verilog-to-routing #376: docs.verilogtorouting.org yields HTTP 403, Cloudflare Error 1014

p--b, 2018-07-23:
Apologies if this isn't the best place for this issue; wasn't sure if the verilog-to-routing.githib.io repo would be more appropriate...

#### Expected Behaviour
Loading https://docs.verilog-to-routing.org should yield documentation

#### Current Behaviour
Loading https://docs.verilogtorouting.org yields a cloudflare error

#### Possible Solution
Serve website directly without using cloudflare

#### Steps to Reproduce
```
$ curl docs.verilogtorouting.org -I
HTTP/1.1 403 Forbidden
Date: Mon, 23 Jul 2018 00:59:49 GMT
Content-Type: text/html; charset=UTF-8
Connection: close
Set-Cookie: __cfduid=df77d7fc2f70fe6174e1809e8a33a31201532307589; expires=Tue, 23-Jul-19 00:59:49 GMT; path=/; domain=.verilogtorouting.org; HttpOnly
Cache-Control: max-age=15
Expires: Mon, 23 Jul 2018 01:00:04 GMT
X-Frame-Options: SAMEORIGIN
Server: cloudflare
CF-RAY: 43ea34e3735bbc50-LHR
$ lynx docs.verilogtorouting.org -dump

   Please enable cookies.

Error 1014 Ray ID: 43ea360ae4b106be • 2018-07-23 01:00:36 UTC

CNAME Cross-User Banned

What happened?

   You've requested a page on a website that is part of the Cloudflare
   network. The host is configured as a CNAME across accounts on
   Cloudflare, which is prohibited by security policy.

What can I do?

   If you are interested in learning more about Cloudflare, please visit
   our website.

   Cloudflare Ray ID: 43ea360ae4b106be • Your IP: 146.169.136.132 •
   Performance & security by [1]Cloudflare

References

   1. https://www.cloudflare.com/5xx-error-landing?utm_source=error_footer
```

#### Context
Attempting to view documentation

#### Your Environment
The internet

mithro, 2018-07-23:
FYI - You can view the docs at http://vtr-verilog-to-routing.readthedocs.io/en/latest/ while the problem is fixed.

kmurray, 2018-07-31:
We have been using Cloudflare to provide https access to `docs.verilogtorouting.org`, which was CNAMEed to `readthedocs.io` (who host the actual documentation).

It seems that Read The Docs recently switched to using Cloudflare for readthedocs.io with the aim of providing https for custom domains (the same reason we were using Cloudflare). However, for security reasons Cloudflare only allows one-level of proxying, which is why things broke. For full details see [here](https://github.com/rtfd/readthedocs.org/issues/4395).

The recommendation seems to be to set our `docs.verilogtorouting.org` CNAME to `cloudflare-to-cloudflare.readthedocs.io` which allows us to keep settings like http -> https redirects in place.

At some point RTD plans on enabling http -> https redirects globally. Once that is done we can likely disable the Cloudflare proxying on our end, and set the CNAME back to `readthedocs.io`.

In the meantime, I've updated the CNAME to `cloudflare-to-cloudflare.readthedocs.io` and docs.verilogtorouting.org should now be served correctly.

closed 2018-07-31 by kmurray