verilog-to-routing/vtr-verilog-to-routing #378: No driver and Memory corruption problem

qaarah, 2018-08-01:
Hi,
I used a yosys generated  blif file to input Arachne-pnr for place and route. I got outputs that are .asc and txt format. Can I get .blif file from Arachne after Place and route, Just like VPR? Is there any possibility to get a blif file from arachne-pnr after place and route.
VPR Version: VPR 7.0 Full Release, May 18, 2013
On Linux
Thanks

kmurray, 2018-08-01:
I'd suggest trying the latest version of VTR from the trunk. 

We've substantially upgrade the robustness of the blif handling since the VTR 7.0 release, and I suspect the trunk would handle this with no issues.

kmurray, 2018-08-01:
That means your compiler doesn't have C++ 14 support which is required to build the latest VTR.

Compilers that should work are:
* GCC >= 4.9
* Clang >= 3.6

You should be able to install one of these with your Linux distribution's package manager.

qaarah, 2018-08-02:
Thanks, I am working on it to install. (One of them). I tried all the questions that are posted on different forums but after installation of GCC , my compiler could not be updated.

kmurray, 2018-08-02:
That sounds like a computer setup related issue, which is unrelated to VTR. I'd suggest checking your Linux Distro's documentation.

I'm going to close this issue for now. If you get things sorted out and still run into the no driver/memory corruption message with the latest VTR feel free to re-open it.

closed 2018-08-02 by kmurray

qaarah, 2018-08-03:
Hi @kmurray . 
I am tryinig to synthesize using new VTR. I am using the sample_arch.xml file and also tried the xml file of previous version but I get the following error. 
"Found 1 'timing' child node(s) of 'device' (expected 0). <timing> has been replaced with the <switch_block> tag. Please upgrade your architecture file."
I have gone through the timing and <switch_block> but could not solve the problem. 
Is there any special xml. for new versions?
Thanks for help.