verilog-to-routing/vtr-verilog-to-routing #380: Timing and switch block error in VTR 8. 0

qaarah, 2018-08-06:
Hi,
I am trying to palce and route using VPR the blif generated by yosys (synth_ice40). I face the following error.
Architecture file: sample_arch.xml
Circuit name: Example_LUT

Warning 1: Model 'multiply' input port 'b' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 2: Model 'multiply' input port 'a' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 3: Model 'multiply' output port 'out' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
Warning 4: Model 'single_port_ram' input port 'data' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 5: Model 'single_port_ram' input port 'addr' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 6: Model 'single_port_ram' input port 'we' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 7: Model 'single_port_ram' output port 'out' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
Warning 8: Model 'dual_port_ram' input port 'data2' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 9: Model 'dual_port_ram' input port 'data1' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 10: Model 'dual_port_ram' input port 'addr2' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 11: Model 'dual_port_ram' input port 'addr1' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 12: Model 'dual_port_ram' input port 'we2' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 13: Model 'dual_port_ram' input port 'we1' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 14: Model 'dual_port_ram' output port 'out2' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
Warning 15: Model 'dual_port_ram' output port 'out1' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
Error 1: sample_arch.xml:129 Unexpected attribute 'auto' found on node 'layout'.
When I do solve this issue by giving new sample architecture provided on https://vtr-verilog-to-routing.readthedocs.io/en/latest/vpr/file_formats.html#fpga-architecture-xml, I  get the following error.
Error 1: Arch_VPR8.xml:94 Found 1 'timing' child node(s) of 'device' (expected 0). has been replaced with the <switch_block> tag. Please upgrade your architecture file.
I also tried to play with switch_block but I do not know, what up-gradation is required. 
Thanks

kmurray, 2018-08-07:
Thanks for filing a separate issue for this. That's typically better than adding comments to older ones (unless the older issue directly matches what you are facing). I've included my previous response below:

It looks like you are using an out-of-date architecture file (perhaps a VTR7 architecture file with the latest trunk code?).

The architecture files which are included the the trunk code should all be up-to-date.

If you have your own architecture file which you need to upgrade then you can use the vtr_flow/scripts/upgrade_arch.py script. However I'd suggest starting out with one of the architecture files from the trunk first.

qaarah, 2018-08-07:
@kmurray Thank you very much. I used the architecture file from the truck, but I still needed to make some changes that I have made and it works now. 

Changes made:
1. timing block in device model is deleted 
2. in sizing 3rd parameter is deleted. 
3. Connection block is added. connection_block input_switch_name="0"/>
4. <io width="1.000000"/> Removed
5. Fc _in is replaced with  <fc in_type="frac" in_val="0.15" out_type="frac" out_val="0.10"/>
Thanks

qaarah, 2018-08-21:
Hi, 
While routing with VPR 8.0m it seems that it does routing with different iterations, What does it mean? For instance "Completed net delay value cross check successfully.
Critical path: 0.95277 ns
Successfully routed after 6 routing iterations."
After that it begins with fifty iterations and there is a message in log file. routing failed, but the the end its shows routing successfully completed. Does it skip the routing where connection was not possible?  Just like I got an error here:
"Cannot route from clb.I[1] (rr_node: 241 type: SOURCE ptc: 1 xlow: 1 ylow: 1) to io[7].outpad[0] (rr_node: 428 type: SINK ptc: 21 xlow: 2 ylow: 1) to -- no possible path
Failed to route connection from 'io_0_5_0' to 'out:io_13_11_1' for net 'io_8_0_0'
Routing failed." 
If someone could explain.
Thanks

kmurray, 2018-08-21:
By default VPR attempts to find the minimum channel width via a binary search on minimum channel width.  Some channel widths will be un-routable (which is information needed to perform the search).

Usually you care about what the actual minimum channel width is, which VPR reports at the end as the 'Best channel width'.

I'd suggest looking at some of the VPR related references if you need more background information about what it is doing. For example:
 - https://docs.verilogtorouting.org/en/latest/references/#betz-vpr
 - https://docs.verilogtorouting.org/en/latest/references/#betz-arch-cad

closed 2018-08-21 by kmurray

qaarah, 2018-08-21:
Yes, That make sense. Got it. 
Thanks