verilog-to-routing/vtr-verilog-to-routing #381: Ubuntu 18

raminrasoulinezhad, 2018-08-07:
make command doesn't work properly on the 7.0 version.
#### Expected Behaviour

CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find BISON (missing: BISON_EXECUTABLE)

#### Current Behaviour

#### Possible Solution

#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

qaarah, 2018-08-07:
Install Bison 
sudo apt-get install bison
You can also install using PPA

kmurray, 2018-08-07:
The VTR 7 release is rather old, so it was never tested on more recent versions of Ubuntu.

I'd recommend trying the latest master/trunk code, which should build OK. However you do need to install the required dependencies. A list (which should be up to date) is here: http://docs.verilogtorouting.org/en/latest/dev/building/#debian-ubuntu

closed 2018-08-07 by kmurray

raminrasoulinezhad, 2018-08-08:
Thanks, there was a problem with required preinstallation files. Now it's working :)