verilog-to-routing/vtr-verilog-to-routing #384: Incorrect Post-ABC Clock Restoration

kmurray, 2018-08-17:
#### Expected Behaviour

The VTR flow should handle circuits with no, single, or multiple clocks correctly.

#### Current Behaviour

There are actually two (inter-related) bugs here:

A) Clock restoration fails in certain cases, due to a bug in the clock restoration script. This occurs when the .latches are re-ordered in the ABC output.

B) More seriously, fixing (A) reveals that clock restoration is currently incorrect in many cases, and *no warning/error is generated*. This has been passing regression tests since the script incorrectly uses the first clock for all latches; and we've only tested these code-paths with single-clock circuits (which coincidentally produces the correct results).

#### Steps to Reproduce

A) Clock restoration fails if `.latch`es are re-ordered (thanks to Dipika Badri for the initial report!)
1. Run the command:
```
$VTR_ROOT/vtr_flow/scripts/run_vtr_flow.pl  $VTR_ROOT/vtr_flow/benchmarks/vtr_benchmarks_blif/stereovision3.blif  $ARCH/power/k6_N10_I40_Fi6_L4_frac0_ff1_22nm.xml  -power -cmos_tech $VTR_ROOT/vtr_flow/tech/PTM_22nm/22nm.xml -starting_stage abc
```
2. Observe the failed output:
```
k6_N10_I40_Fi6_L4_frac0_ff1_22nm/stereovision3                                   failed: to restore multi-clock latch info (took 0.31 seconds)
```
3. Move to the run directory and manually run the clock restoration script:
```
$VTR_ROOT/vtr_flow/scripts/restore_multiclock_latch_information.pl stereovision3.odin.blif stereovision3.raw.abc.blif stereovision3.abc.blif
```
Observe the following error:
```
Latch top^FF_NODE~50 not found in Odin II BLIF!
.latch       n363 top^FF_NODE~50  0
```
However, the .latch top^FF_NODE~50 does in fact exist in both the reference and post-ABC netlists.

B) Fix file seeking behaviour in restore_multiclock_latch_information.pl. Run `vtr_reg_basic` and observe new cock restoration failures on previously passing circuits.

#### Context

ABC strips away clock information when it performs synthesis. This information is required at later stages of the design flow (e.g. VPR) so that multi-clock circuits are handled correctly.

We previously beleived that simply name-matching the `.latch`es would be sufficient to uniquely identify    `.latch`es in both the pre and post ABC netlists. However this is not always the case.

For instance, before ABC:
```
$ grep .latch mkPktMerge.odin.blif 
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~310^MUX_2~4257 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^wp~0_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~370^MUX_2~2877 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^full_r_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~357^MUX_2~2906 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^gb2_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~390^MUX_2~2787 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^empty_r_FF_NODE re top^CLK 3
.latch top^MULTI_PORT_MUX~2257^MUX_2~3776 top^fi0HasPrio_FF_NODE re top^CLK 3
.latch top^MULTI_PORT_MUX~2257^MUX_2~3775 top^fi0Active_FF_NODE re top^CLK 3
.latch top^MULTI_PORT_MUX~2257^MUX_2~3777 top^fi1Active_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~327^MUX_2~2943 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^rp~0_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~327^MUX_2~2944 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^rp~1_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~327^MUX_2~2945 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^rp~2_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~327^MUX_2~2946 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^rp~3_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^MULTI_PORT_MUX~796^MUX_2~3519 top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^rp~0_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^MULTI_PORT_MUX~859^MUX_2~3363 top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^empty_r_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^MULTI_PORT_MUX~826^MUX_2~3482 top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^gb2_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^MULTI_PORT_MUX~839^MUX_2~3453 top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^full_r_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^MULTI_PORT_MUX~796^MUX_2~3520 top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^rp~1_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^MULTI_PORT_MUX~796^MUX_2~3521 top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^rp~2_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^MULTI_PORT_MUX~796^MUX_2~3522 top.arSRLFIFO_b+fi1.generic_fifo_sc_b+fifo_1^rp~3_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^MULTI_PORT_MUX~1248^MUX_2~3951 top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^wp~0_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^MULTI_PORT_MUX~1308^MUX_2~3810 top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^full_r_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^MULTI_PORT_MUX~1295^MUX_2~3892 top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^gb2_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^MULTI_PORT_MUX~1328^MUX_2~3863 top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^empty_r_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^MULTI_PORT_MUX~1248^MUX_2~3952 top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^wp~1_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^MULTI_PORT_MUX~1248^MUX_2~3953 top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^wp~2_FF_NODE re top^CLK 3
.latch top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^MULTI_PORT_MUX~1248^MUX_2~3954 top.arSRLFIFO_c+fo.generic_fifo_sc_c+fifo_1^wp~3_FF_NODE re top^CLK 3
[... 38 lines cut ...]

As shown above, it seems that in some instances ABC completely re-names the connections to/from .latches, which breaks the fundamental assumption that the clock restoration script makes (i.e. that it can match .latches based on output signal names).

It seems to me that the best course of action is to convert all `.latch`es to blackboxes before passing netlists through ABC, and then restoring them to `.latch`es after ABC. This would ensure all clock information was preserved, but at the cost of disabling sequential optimizations within ABC (we already disable most sequential optimizations like re-timing anyway).

For example, given an initial `.latch`:
```
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~310^MUX_2~4257 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^wp~0_FF_NODE re top^CLK 3
```
a script would convert it to a blackbox as it was sent through ABC:
```
.subckt latch_re_3 in=top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~310^MUX_2~4257 out=top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^wp~0_FF_NODE
control=top^CLK
```
where the subckt type (`latch_re_3`) encodes that it is rising-edge trigged with an initial value set to don't care (`3`).
Since ABC leaves blackboxes alone, a script could then convert the blackbox back into a latch without loosing any information:
```
.latch top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^MULTI_PORT_MUX~310^MUX_2~4257 top.arSRLFIFO_a+fi0.generic_fifo_sc_a+fifo_1^wp~0_FF_NODE re top^CLK 3
```

#### Your Environment

* VTR revision used:  b59b97c
* Operating System and version: Ubunut 16.04
* Compiler version: GCC-5

kmurray, 2018-08-17:
@vaughnbetz @KennethKent @jeanlego @aarongraham9: Any thoughts?

It seems to we have only two options:

i) Convert `.latch`es to black-boxes as described above.
- This is a fairly trivial change which should be simple to make
- But it may reduce the optimizations ABC can perform

ii) Synthesize each clock-domain through ABC independently, and then re-stich the netlist together into one netlist later.
- Splitting by clock domain is likely more work. Additionally, this would need to be done anywhere we use ABC (e.g. for logic optimization, but also for equivalence checking), or by tools which use ABC as a library (e.g. ACE) -- this seems like a fair amount of work
- However, it would allow ABC to perform more sequential optimizations.

aarongraham9, 2018-08-21:
Looking into this @jeanlego and @kmurray.

I'd like to try and see if the restoration script can be improved and we can fix it that way. I'd rather not Black-Box them if possible as ABC isn't able to optimize just about anything and the QoR values are very bad in comparison to VTR 7 let alone with the updated ABC/VTR 8.

jeanlego, 2018-08-21:
I have pushed a commit that should fix this if you can try it!

vaughnbetz, 2018-08-23:
1. Aaron, can you post the QoR impact of using black box latches?  It sounds like you have this data from a prior run; can you attach a spreadsheet and summary?  If you don't have complete data, I think it would be good to rerun this so we have a precise QoR impact estimate, as the black box solution definitely seems easiest.  If the QoR is not too bad, maybe it will be the winning candidate.

2. I think the options are:

- black box the latches:  simplest, but hurts QoR.  Need to see by how much.
- For circuits with more than one clock, split the circuit into multiple clock domains, and synthesize each separately in ABC.  This is essentially a netlist covering/colouring problem I think.  One sub-netlist per clock domain (all src and dst registers in same clock domain along with the logic between them) plus one netlist for all the inter-clock domain logic and logic reachable / reaching primary I/Os (in this netlist, all flops should be converted to primary inputs and outputs to prevent sequential optimizations, which are unsafe between clock domains).
- Fix the current script. Kevin doesn't think this is doable, as the latches are being renamed in unpredictable ways; it's not clear how we can match them.

vaughnbetz, 2018-08-23:
With the new approach you mention above (black box all but one domain at a time) it seems like you'll get duplicate synthesis of the logic between clock domains.  How do you ensure you get only one copy in the end?

Interestingly, none of these approaches, except black boxing all latches, is fully safe for cross-clock domain paths -- the registers involved in those paths shouldn't be duplicated or retimed as they can form synchronizers etc.  Since abc doesn't understand that, I think fixing it is beyond the scope of what we want to do at least for the foreseeable future.

jeanlego, 2018-08-25:
I've opened a PR with a way to test all three options, it also includes simulation to verify the blif pre vs post ABC

jeanlego, 2018-10-21:
view PR #408 
I've included in the flow a simulation before and after abc.
for 1 case odin could not simulate
fpu_hard_block_arch:      hard_fpu_arch_timing.xml/mm3.v/common                 (failed to include simulation) OK
but this is because the odin siulator doesn't have the hardblock built for the simulator
 
for 2 cases, odin failed siulation, but that is because they have no input pins, I'll make a workaround so that it generates N cycles for the simulator derived from the ouput vector file length.

here are the culprit:
func_formal_flow:         k6_frac_N10_40nm.xml/const_true.blif/common            (failed: odin Simulation)  OK
func_formal_flow:         k6_frac_N10_40nm.xml/const_false.blif/common           (failed: odin Simulation)  OK

But all other cases pass odin simulation. The good news is that the iterative approach seem to yield better result in a lot of cases! 

here are the result of vtr_reg_strong

# VTR_REG_STRONG
># 122 qor test failures
> # 0 run failures

## regression_tests vtr_reg_strong strong_fpu_hard_block_arch
`pass`
## regression_tests vtr_reg_strong strong_fracturable_luts
`pass`
## regression_tests vtr_reg_strong strong_mcnc
`pass`
## regression_tests vtr_reg_strong strong_no_timing
`pass`
## regression_tests vtr_reg_strong strong_timing
`pass`
## regression_tests vtr_reg_strong strong_titan
`pass`
## regression_tests vtr_reg_strong strong_power
`pass`
## regression_tests vtr_reg_strong strong_func_formal_flow

arch|circuit|changes|golden|result
--|--|--|--|--
k6_frac_N10_40nm | always_true.blif | num_pre_packed_nets: | 1 | 7
k6_frac_N10_40nm | always_true.blif | num_post_packed_nets: | 1 | 7
k6_frac_N10_40nm | always_true.blif | placed_CPD_est: | nan | 0.734653
k6_frac_N10_40nm | always_true.blif | placed_setup_TNS_est: | 0 | -0.734653
k6_frac_N10_40nm | always_true.blif | placed_setup_WNS_est: | 0 | -0.734653
k6_frac_N10_40nm | always_true.blif | routed_wirelength: | 0 | 7
k6_frac_N10_40nm | always_true.blif | num_io: | 0 | 6
k6_frac_N10_40nm | always_false.blif | num_pre_packed_nets: | 1 | 7
k6_frac_N10_40nm | always_false.blif | num_post_packed_nets: | 1 | 7
k6_frac_N10_40nm | always_false.blif | placed_CPD_est: | nan | 0.734653
k6_frac_N10_40nm | always_false.blif | placed_setup_TNS_est: | 0 | -0.734653
k6_frac_N10_40nm | always_false.blif | placed_setup_WNS_est: | 0 | -0.734653
k6_frac_N10_40nm | always_false.blif | routed_wirelength: | 0 | 7
k6_frac_N10_40nm | always_false.blif | num_io: | 0 | 6
k6_frac_N10_40nm | false_path_mux.blif | num_pre_packed_nets: | 4 | 7
k6_frac_N10_40nm | false_path_mux.blif | num_pre_packed_blocks: | 6 | 8
k6_frac_N10_40nm | false_path_mux.blif | num_post_packed_nets: | 4 | 5
k6_frac_N10_40nm | false_path_mux.blif | placed_CPD_est: | 0.708653 | 1.37865
k6_frac_N10_40nm | false_path_mux.blif | placed_setup_TNS_est: | -0.708653 | -1.37865
k6_frac_N10_40nm | false_path_mux.blif | placed_setup_WNS_est: | -0.708653 | -1.37865
k6_frac_N10_40nm | false_path_mux.blif | num_io: | 3 | 4
k6_frac_N10_40nm | mult_4x4.blif | num_post_packed_nets: | 21 | 27
k6_frac_N10_40nm | mult_5x5.blif | num_pre_packed_nets: | 47 | 43
k6_frac_N10_40nm | mult_5x5.blif | num_pre_packed_blocks: | 57 | 53
k6_frac_N10_40nm | mult_5x5.blif | logic_block_area_used: | 215576 | 161682
k6_frac_N10_40nm | mult_5x5.blif | num_clb: | 4 | 3
k6_frac_N10_40nm | rca_3bit.blif | num_pre_packed_nets: | 12 | 13
k6_frac_N10_40nm | rca_3bit.blif | num_pre_packed_blocks: | 16 | 17
k6_frac_N10_40nm | rca_4bit.blif | num_pre_packed_nets: | 15 | 17
k6_frac_N10_40nm | rca_4bit.blif | num_pre_packed_blocks: | 20 | 22
k6_frac_N10_40nm | rca_4bit.blif | placed_CPD_est: | 1.04365 | 1.71365
k6_frac_N10_40nm | rca_4bit.blif | placed_setup_TNS_est: | -4.54826 | -6.55826
[... 148 lines cut ...]
## regression_tests vtr_reg_strong strong_custom_switch_block
`pass`
## regression_tests vtr_reg_strong strong_custom_grid
`pass`
## regression_tests vtr_reg_strong strong_flyover_wires
`pass`
## regression_tests vtr_reg_strong strong_minimax_budgets

arch|circuit|changes|golden|result
--|--|--|--|--
k6_frac_N10_frac_chain_mem32K_40nm | stereovision3.v | logic_block_area_used: | 970092 | 700622
k6_frac_N10_frac_chain_mem32K_40nm | stereovision3.v | num_clb: | 18 | 13

## regression_tests vtr_reg_strong strong_scale_delay_budgets

arch|circuit|changes|golden|result
--|--|--|--|--
k6_frac_N10_frac_chain_mem32K_40nm | stereovision3.v | num_clb: | 18 | 15

## regression_tests vtr_reg_strong strong_verify_rr_graph
`pass`
## regression_tests vtr_reg_strong strong_analysis_only
`pass`
## regression_tests vtr_reg_strong strong_route_only
`pass`

kmurray, 2018-10-23:
Thanks @jeanlego for putting this all together, it looks like a lot of work!

**Regression Failures**
> for 1 case odin could not simulate
> fpu_hard_block_arch: hard_fpu_arch_timing.xml/mm3.v/common (failed to include simulation) OK
> but this is because the odin siulator doesn't have the hardblock built for the simulator

Seems like disabling simulation is the right approach here.

**ABC Latch Classes**
> the thought is to use ABC in two passes,
> before the first pass register all the clock domains in a file and assign them a class
> rewrite the the blif file to use the classes for the latches
> run a sequential only optimization through ABC
> recover the clock domains from the classes file
> second pass is to run the iterative approach with combinational optimization only.

Provided the class feature works correctly, why would we need to go through ABC in two passes? Wouldn't one pass be sufficient (would could then restore the clocks from the saved class information)?

I will note that at one point I did look into ABC's latch class feature, but when I tried it the class information was not included in ABC's output BLIF (i.e. it was dropped just like the regular clocks...).

**QoR Failures**
Taking a look at the QoR failures you've included they all seem OK.

They mostly look like improvements caused by fewer netlist primitives (fewer pre-packed blocks and pre-packed nets) for the `stereovision3` benchmark.

**failed to include simulation**
I'm seeing a large number of `failed to include simulation     OK` lines when I use run_vtr_flow on this branch.

Can you clarify what this means? I think it means ODIN is not performing simulation to verify that the ODIN and ABC BLIF's are equivalent.

If that is the case, and we don't want to run simulation, then we probably should only print `OK` (since skipping the simulation is expected).  Any thoughts?

**Verification**
Can you clarify what testing this branch has gone through?

How have you verified that the different flow types are functionally correct?

How have you verified that the correct clocks are being restored?

vaughnbetz, 2018-10-25:
Thanks @jeanlego for pulling this together.

Some questions:
- In the QoR data, what is the difference between vanilla, blanket and iterative?  Is vanilla running ABC with no latch black boxes (old flow), while blanket is all black boxes and iterative is one clock domain at a time black boxes?
- Kevin generated some QoR results (see his follow up post) that show reasonable differences between vanilla and iterative on the vtr circuits, which are almost all single clock so the differences seem a bit bigger than expected.

kmurray, 2018-10-25:
Here are the results I generated from the mulipass_abc_flow branch on the VTR benchmarks.

[image]

As mentioned, `iterative` seems to produce fewer primitives, lower channel width, and wirelength but high critical path delay.

However since all the VTR benchmarks (except the tiny `stereovision3` benchmark) are all single clock, we would expect the results to be the same as `vanilla` -- since for the single clock case we'd expect `iterative` to reduce down to `vanilla`.

jeanlego, 2018-10-25:
The difference could be that i force an initial value on ABC using the init command, and I now force Odin to default to an initial value of 0 rather than X for .latch unless specified otherwise.

I had to do these for the simulator.

I defined "clock domains" to be a triplet of driver, edge sensitivity, and initial value.
Each run includes a report.clk providing the different "clock domain" used to itterate and also a file that display the number of latches within this clock domain.

I also had to modify Odin as it used to only display rising edge for latches but now as support for all 5 types. Some result discrepancy could come from there if you compared with the old version.

jeanlego, 2018-10-25:
Vanilla is running with the patched restoration script ( typo prevented the script from rewinding the file (that is original) ) 
Blanket approach is when we black box all the latches and itteration is one "clock domain" triplet (described above)
But the biggest takeaway here is that all other than itteration fails to match initial vs post ABC simulation

jeanlego, 2019-02-18:
Should this be closed based on 411

KennethKent, 2019-02-18:
Yes, please :)

Dr. Kenneth B. Kent, P.Eng.
Professor & Director, IBM Centre for Advanced Studies - Atlantic
Faculty of Computer Science, University of New Brunswick
540 Windsor Street, Fredericton, New Brunswick, E3B 5A3
ph. (506) 451-6971
fax. (506) 453-3566

Honorary Professor, Department of Computer Science and Institute of Visual Computing
Bonn-Rhein-Sieg University of Applied Sciences
Grantham-Allee 20, 53757 Sankt Augustin, Germany

jeanlego, 2019-02-18:
closing as PR 411 fixed the issue

closed 2019-02-18 by jeanlego