verilog-to-routing/vtr-verilog-to-routing #386: Place and Route file formats generated by VPR 8 compatibility with VPR7

qaarah, 2018-08-29:
Hi, 
I am using ZUMA flow where I use VPR 8 for place and route. In my old flow I was using VPR 7 that works fine with the scripts. I can see the change in the formats of files (.place .net .route) a bit. How can I make these formats compatible/similar to VPR7 formats. 

* VTR revision used: 8.0 
* Operating System and version: Ubuntu 14
* Compiler version: GCC11

kmurray, 2018-08-29:
*As a side note, VTR 8 is still in development and has not yet been released*

Maintaining forward/backward compatibility for VPR's file formats across versions is a *non-goal*. We don't try to break the formats, but we do change the file formats as needed to support new/more complete features.

Since the FPGA Architecture XML file is usually human specified we do maintain [a script](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/scripts/upgrade_arch.py) to upgrade it's format.
For tool generated files like the .net/.place/.route files we don't provide any support for upgrading/downgrading their formats.

That being said, I believe .net/.place/.route files have seen only minor changes since the VTR 7 release. All the information which was in the VTR 7 formats should be there in essentially the same form in the current VTR code base.

My suggestion would be to upgrade any scripts you are using which parse the .net/.place/.route files to support the new formats. Most likely they can ignore any of the new fields/information in the VTR 8 formats and get the same information as the VTR 7 formats.

closed 2018-08-29 by kmurray

qaarah, 2018-08-29:
Thanks, 
I have already upgraded the FPGA Architecture XML file according to the VTR 8.0. 
Yes there are minor changes, that could be seen, i.e. in the first line of the placement file that is used for routing.  While calling ZUMA, I get the following errors.
"File "/home/qazi/Desktop/Qazi/ZUMA/source/zuma_build.py", line 80, in <module>
    **ReadRouting.read_routing(route_file)
  File "/home/qazi/Desktop/Qazi/ZUMA/source/ReadRouting.py", line 26, in read_routing
    x = int(igot[0].split(' ')[2])+1
ValueError: invalid literal for int() with base 10: 'Placement_ID:'"**
And here is the routing file that I get from VPR (8.0)
"Placement_File: place.p Placement_ID: **SHA256:ebbb3c5981395b5a848e6e4bbb7d812ed516a40c3cec4ce79355022c94aad7e7**
Array size: 5 x 5 logic blocks.

Routing:

Net 0 (io_0_5_0)

Node:	718	SOURCE (3,1)  Class: 34  Switch: 1
Node:	755	  OPIN (3,1)  Pin: 34   clb.O[6] Switch: 0
Node:	1036	 CHANX (3,0)  Track: 4  Switch: 0
Node:	1114	 CHANY (3,1) to (3,3)  Track: 4  Switch: 0
Node:	987	  IPIN (4,2)  Pad: 3  Switch: 1
Node:	969	  SINK (4,2)  Pad: 3  Switch: -1

Net 1 (io_0_11_1): global net connecting:

Block io_13_11_1 (#17) at (3,1), Pin class 28.
Block out:io_0_11_1 (#1) at (3,4), Pin class 0.
Block out:io_4_0_1 (#2) at (3,4), Pin class 3.
Block out:io_13_6_0 (#13) at (0,2), Pin class 3.
Block out:io_13_15_1 (#16) at (2,0), Pin class 3.

Net 2 (io_5_0_1)

Node:	717	SOURCE (3,1)  Class: 33  Switch: 1
Node:	754	  OPIN (3,1)  Pin: 33   clb.O[5] Switch: 0
Node:	1112	 CHANY (3,1) to (3,2)  Track: 2  Switch: 0
Node:	945	  IPIN (4,1)  Pad: 0  Switch: 1
Node:	936	  SINK (4,1)  Pad: 0  Switch: -1

Net 3 (io_5_17_1)

Node:	716	SOURCE (3,1)  Class: 32  Switch: 1
Node:	753	  OPIN (3,1)  Pin: 32   clb.O[4] Switch: 0
Node:	1048	 CHANX (3,1)  Track: 6  Switch: 0
Node:	1111	 CHANY (3,1)  Track: 1  Switch: 0
Node:	957	  IPIN (4,1)  Pad: 3  Switch: 1
Node:	939	  SINK (4,1)  Pad: 3  Switch: -1

In Command in line option there should be Architecture file in the first line instead of Placement ID. 
Up gradation of scripts which parse .net/.place/.route, you mean, ReadRouting.py etc files. but those files should be xml to be ugraded, if I use above script  ? right? How to upgrade these files. How to get rid of this error.  
Thanks

kmurray, 2018-08-29:
The script I mentioned only applies to FPGA architecture files. As I mentioned we provide _no support_ for upgrading/downgrading any other VTR related file formats.

I'd suggest directing your questions about ZUMA and it's scripting infrastructure to the ZUMA developers, as ZUMA is not part of VTR.

qaarah, 2018-08-29:
Thanks @kmurray.