verilog-to-routing/vtr-verilog-to-routing #387: Unable to run run_vtr_flow.pl after upgrading to Ubuntu 18.04

bonifus, 2018-09-06:
After upgrading to Ubuntu 18.04, while on running the script run_vtr_flow.pl I am getting the following error
"Can't use string ("") as a HASH ref while "strict refs" in use at ./vtr_flow/scripts/run_vtr_flow.pl line 1354."

#### Expected Behaviour
VTR flow should run for given benchmark circuit and architecture file

#### Current Behaviour
Instead of executing the flow, getting the response: "Can't use string ("") as a HASH ref while "strict refs" in use at ./vtr_flow/scripts/run_vtr_flow.pl line 1354."
#### Possible Solution
May be change in syntax as Ubuntu 18.04 LTS uses GCC 7.3

#### Steps to Reproduce
1. Go to VTR_ROOT
2. Run run_vtr_flow.pl script using any architecture and benchmark. I tried with k4_N4_90nm.xml, alu4.blif and k6_frac_N10_mem32K_40nm.xml, diffeq1.v
Command used:
./vtr_flow/scripts/run_vtr_flow.pl vtr_flow/arch/timing/k6_frac_N10_mem32K_40nm.xml vtr_flow/benchmarks/verilog/diffeq1.v

#### Context
Earlier with Ubuntu 16.04 LTS same command was working fine. But now after upgrading to Ubuntu 18.04 LTS, run_vtr_flow.pl is throwing the error "Can't use string ("") as a HASH ref while "strict refs" in use at ./vtr_flow/scripts/run_vtr_flow.pl line 1354."
All other scripts which I have checked is working to be fine.

#### Your Environment

* VTR revision used: VTR development trunk downloaded on 05-Sep-2018
* Operating System and version: Ubuntu 18.04 LTS
* Compiler version: gcc 7.3.0

kmurray, 2018-09-06:
Thanks for the report @bonifus!

I've looked into it and the issue is that run_vtr_flow.pl expects the first argument to be the circuit, and the second argument to be the architecture file. So for your example the command should be:
```
./vtr_flow/scripts/run_vtr_flow.pl vtr_flow/benchmarks/verilog/diffeq1.v vtr_flow/arch/timing/k6_frac_N10_mem32K_40nm.xml 
```
(which has the arguments swapped).

I've updated the script (as of 45f719b) to give a better error message if the arguments are in the wrong order:
```
$ ./vtr_flow/scripts/run_vtr_flow.pl vtr_flow/arch/timing/k6_frac_N10_mem32K_40nm.xml vtr_flow/benchmarks/verilog/diffeq1.v
Error: Expected circuit file as first argument (was vtr_flow/arch/timing/k6_frac_N10_mem32K_40nm.xml), and FPGA Architecture file as second argument (was vtr_flow/benchmarks/verilog/diffeq1.v)
```

closed 2018-09-06 by kmurray

bonifus, 2018-09-06:
Thank you for your prompt response.