verilog-to-routing/vtr-verilog-to-routing #391: VPR packing fails when number of nets to route > number of LUTs available

lvezier, 2018-09-19:
The design contains only "wires", ie:
.names x y
1 1

It routes all north to south, south to north, east to west and west to east.

It fails as if it needs to consumes one LUT per wire whereas it should only uses routing resources.

#### Expected Behaviour

#### Current Behaviour

Absorbed logical nets 0 out of 958 nets, 958 nets not absorbed.
FPGA sized to 9 x 10 (T8X7)
Device Utilization: 5.65 (target 1.00)
	Block Utilization: 1.00 Type: io
	Block Utilization: 9.98 Type: clb

Packing took 0.32 seconds (max_rss 27.2 MiB, delta_rss +17.6 MiB)
Error 1: 
Type: Other
File: vtr_repository/vpr/src/pack/pack.cpp
Line: 243
Message: Failed to find device which **satisifies** resource requirements required: io: 958, clb: 479 (available io: 960, clb: 48)

(PS: you can note the typo on "satisfies" )

#### Possible Solution

#### Steps to Reproduce

1. Run example provided in attachment

[vpr_issue_only_net.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/2397286/vpr_issue_only_net.zip)

#### Context

This example is used internally to check IO assignment, not really representative of a real use case.

#### Your Environment

* Version: 8.0.0-dev+bb7c4f1-dirty
* Revision: bb7c4f1-dirty
* Operating System and version: RHEL 7.5
* Compiler version: GNU 5.3.1 on Linux-4.8.13-100.fc23.x86_64 x86_64

kmurray, 2018-09-19:
Thanks for the report.

While VPR can generally sweep away these types of 'buffer' LUTs, it can **not** do so it if they connect to both primary inputs and outputs (in such cases it's unclear to VPR what the net connecting the primary inputs/outputs should be called). This is fundamentally a limitation the BLIF format, which has no notion of logical assignment (i.e. comparable to Verilog's `assign y = x;`).

Due to this and other limitations of traditional BLIF, VPR does support an extended BLIF format which supports [additional features](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#extended-blif-eblif).

I'd suggest trying extended BLIF's `.conn` statement (which mimics Verilog's `assign`).

So, for example you could use:
```
.conn x y
```
which should be equivalent to:
```
.names x y
1 1
```
but will not generate a buffer LUT (which VPR can not remove).

lvezier, 2018-09-19:
Thanks for the answer, I will check if I can move to extended blif format in our flow.

closed 2018-09-19 by lvezier