verilog-to-routing/vtr-verilog-to-routing #393: Segmentation fault (core dumped)

qaarah, 2018-09-28:
Hi, 
While using VPR 7 , I get the following error. 
Segmentation fault (core dumped).
I am using these commands
    --net_file netlist.net \
    --place_file place.p \
    --route_file route.r \
    --blif_file abc_out.blif \
    --timing_analysis off \
    --nodisp \
    --route_chan_width 26 \
    --echo_file on
What is the reason of segmentation fault (core dumped) error? 
Thanks

kmurray, 2018-09-28:
Thanks for the report. A segmentation fault indicates an invalid memory access (i.e. a bug) in VPR.

Can you try re-running your design with the latest version of VPR (i.e. master branch)?

All our development efforts focus on the master branch, we do not at this point make fixes to the old VTR 7 code base.

qaarah, 2018-09-28:
@kmurray Thanks.

closed 2018-09-28 by qaarah