verilog-to-routing/vtr-verilog-to-routing #409: Read The Docs: Incorrect cross-file references for Markdown files when built as dirhtml

kmurray, 2018-10-31:
#### Expected Behaviour

Some of the .md files in the root of the source tree (README.developers.md, CONTRIBUTING.md, BUILDING.md) are linked into the sphinx/Read The Docs documentation.

All cross-file references should resolve to valid links on https://docs.verilogtorouting.org.

#### Current Behaviour

Cross-file links resolve correctly when build as `html`, but incorrectly when built as `dirhtml`.

Since ReadTheDocs builds using `dirhtml` such links are broken in the documentation.

For instance, the 'Submitting Code to VTR' link under [here](https://docs.verilogtorouting.org/en/latest/dev/developing/#for-external-developers) currently resolves to:
```
https://docs.verilogtorouting.org/en/latest/dev/developing/contributing.html#submitting-code-to-vtr
```
which does not exist.

It should resolve to:
```
https://docs.verilogtorouting.org/en/latest/dev/contributing/#submitting-code-to-vtr
```

#### Possible Solution

The extra `developing/` should be removed in the link resolver when building as `dirhtml`.

#### Steps to Reproduce

1. Above link resolves correctly when built as `html`
```
#From the VTR root
$ cd docs
$ make html
```
2. Above link fails to resolve correctly when build as `dirhtml`
However when build as `dirhtml`:
```
#From the VTR root
$ cd docs
$ make dirhtml
```

#### Your Environment

* VTR revision used: 11035e9
* Operating System and version: Ubuntu 16.04, or ReadTheDocs

kmurray, 2019-05-10:
@tinayang3024 @vaughnbetz FYI this is the existing issue for the broken documentation links

vaughnbetz, 2019-05-10:
Tim, does anyone on your team know how to fix this?  It's assigned to you currently.
If not, we can reassign to Tina, and she can fix the broken links if possible. If she can't figure out how to do that, I suggest the next best action is to move the various .md documents into the /doc subdirectories and leave behind only brief pointers to those more detailed documents.

mithro, 2019-05-10:
Is there any reason you want to use `make dirhtml` (which fails) verse `make html` (which works)?

kmurray, 2019-05-13:
I believe `dirhtml` is what readthedocs uses to build the docs at [docs.verilogtorouting.org](https://docs.verilogtorouting.org).

mithro, 2019-05-13:
Okay, I'll take a look.

mithro, 2019-05-16:
So it turns out **both** html and dirhtml are broken. 

The issue is that `markdown_code_symlinks.py` script was using the `__enable_auto_doc_ref__` functionality which was [removed from recommonmark here](https://github.com/rtfd/recommonmark/commit/8baf238562c2f7c9bea603e352c3a94f51905a3e). 

Despite that commit happening back in March 2017, it was only **released** as part of [Tag 0.5.0](https://github.com/rtfd/recommonmark/releases/tag/0.5.0) which only happened 8th Jan 2019.

When I revert that the recommonmark commit the html version goes back to working and only the dirhtml output is borked.

I'm pretty sure I know how to fix this for both, but it will take me a bit longer to get to that...

mithro, 2019-05-21:
I have a fix now but it'll take me a few more days to clean it up and send a pull request.

kmurray, 2019-09-16:
Should be fixed with #943

closed 2019-09-16 by kmurray