verilog-to-routing/vtr-verilog-to-routing #413: Assertion "max_type" failed

GinnYu, 2018-11-27:
Hi,

**VPR gives me the follow error when running with command "vpr customize_arch_11211025.xml ncl_blob_merge.blif" (all command set as default). And I am currently using the latest version which is update from github, any idea what could happen with the assertion failed?**

Packing
Begin packing 'ncl_blob_merge.blif'.

After removing unused inputs...
        total blocks: 61423, total nets: 61283, total inputs: 264, total outputs: 140
Begin prepacking.
Finish prepacking.
Using inter-cluster delay: 7.256e-11
/home/e15913/project/vtr-verilog-to-routing/vpr/src/util/vpr_utils.cpp:775 find_most_common_block_type: Assertion 'max_type' failed.
Aborted (core dumped)

**If I change the code from "VTR_ASSERT(max_type);" to "VTR_ASSERT_DEBUG(max_type);", a segmentation falut will be occured.**

Packing
Begin packing 'ncl_blob_merge.blif'.

After removing unused inputs...
        total blocks: 61423, total nets: 61283, total inputs: 264, total outputs: 140
Begin prepacking.
Finish prepacking.
Using inter-cluster delay: 7.256e-11
Segmentation fault (core dumped)

#### Your Environment

*VPR FPGA Placement and Routing.
*Version: 8.0.0-dev+75e2ad7-dirty
*Revision: 75e2ad7-dirty
*Compiled: 2018-11-27T10:59:37 (release build)
*Compiler: GNU 5.4.0 on Linux-4.4.0-31-generic x86_64

kmurray, 2018-11-27:
Thanks for the report.

Can you provide the netlist and architecture files to reproduce the issue?

GinnYu, 2018-11-27:
Hi  @kmurray ,

Attached is the blif netlist and the architecture file. I am trying to place and route an asynchronous circuit with VPR. And the error shows up when I use the VPR command.

[netlist+arch_file.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/2622048/netlist%2Barch_file.zip)

Regards,
Jing