verilog-to-routing/vtr-verilog-to-routing #417: vpr fails during routing with "check_route: found non-adjacent segments"

gsomlo, 2018-12-18:
I have a custom architecture file (see [arch_old_git_e039db84c.xml.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/2692570/arch_old_git_e039db84c.xml.txt)) that used to work fine with an older version of VtR (git e039db84c), using binops.v as an example:
```
  ./vtr_flow/scripts/run_vtr_flow.pl \
    ./ODIN_II/FULL_REGRESSION_TESTS/full/binops.v \
    ../arch_old_git_e039db84c.xml
```
I tried upgrading the architecture file for the current VtR (git 5273d4095) starting with "`vtr_flow/scripts/upgrade_arch.py`" followed by some extra manual intervention (see [arch_new_git_5273d4095.xml.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/2692571/arch_new_git_5273d4095.xml.txt)).

Both architecture files (old and new) have been carefully re-indented so that only the relevant changes would show up nicely using `diff`.

The current version of vpr, using the upgraded architecture file, generates a routing failure. I used a similar command line to the above, adjusted for the new location of the binops.v sample file:
```
  ./vtr_flow/scripts/run_vtr_flow.pl \
    ./ODIN_II/regression_test/benchmark/full/binops.v \
    ../arch_new_git_5273d4095.xml
```
The (hopefully) relevant portion of vpr.out is included below:
```
...
Restoring best routing
Critical path: 2.62149 ns
Successfully routed after 21 routing iterations.
Router Stats: total_nets_routed: 439 total_connections_routed: 979 total_heap_pushes: 176547 total_heap_pops: 84607
Build routing resource graph
Warning 62: in check_rr_graph: fringe node 18 IPIN at (0,1) has no fanin.
	 This is possible on a fringe node based on low Fc_out, N, and certain lengths.
Build routing resource graph took 0.01 seconds (max_rss 10.9 MiB, delta_rss +0.0 MiB)
Best routing used a channel width factor of 26.
Error 2: 
Type: Routing
File: /home/somlo/ECE_VtR/vtr-verilog-to-routing/vpr/src/route/check_route.cpp
Line: 143
Message: in check_route: found non-adjacent segments in traceback while checking net 0:
  RR node: 3605 type: CHANX location: (1,0) <-> (2,0) track: 5 len: 1 longline: 0 seg_type: L4 dir: DEC_DIR
  RR node: 4066 type: CHANY location: (1,1) <-> (1,3) track: 0 len: 2 longline: 0 seg_type: L4 dir: INC_DIR

Routing took 1.57 seconds (max_rss 10.9 MiB, delta_rss +0.4 MiB)
...
```
The versioning information from vpr.out is shown below:
```
VPR FPGA Placement and Routing.
Version: 8.0.0-dev+5273d4095
Revision: 5273d4095
Compiled: 2018-12-17T10:54:12 (Release build)
Compiler: GNU 8.2.1 on Linux-4.19.3-200.fc28.x86_64 x86_64
University of Toronto
vtr-users@googlegroups.com
This is free open source code under MIT license.
```
This _might_ be related to issue #331 but since I'm not 100% sure I decided to open a separate bug. Please feel free to mark as duplicate if appropriate.

vaughnbetz, 2018-12-31:
Nothing looks obviously wrong in the architecture xml file to me, and in any case it the error above is an internal consistency check error in VPR (wires that do not seem close enough to connect are in fact being connected by the router).  Can you generate and check the rr_graph.echo file to see if those two segments do indeed connect to each other (have an edge between them in the proper direction)?  

It seems like either the rr-graph generator has created a strange edge, or the router illegally connected two segments that don't have an edge. So the next step is figuring out which of those two cases it is.

gsomlo, 2019-01-01:
I ran
```
./bin/vpr ../arch_new_git_5273d4095.xml \
          temp/binops.pre-vpr.blif \
          --write_rr_graph temp/binops_rr_graph.xml
```
and there doesn't appear to be an edge between nodes 3605 and 4066. Here's the attached 
[binops_rr_graph.xml.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/2719440/binops_rr_graph.xml.txt) file for the record.

kmurray, 2019-01-08:
Haven't had a chance to look in detail, but there were some changes to how the custom switchblock specification is interpreted recently (e.g. 23cd16116, 69d9e62bc) which could be responsible.

vaughnbetz, 2019-01-09:
Thanks for the investigation.  If you have the time, checking out a version of the code from early December, and from early November, and seeing if they work would be very helpful.  There was a set of changes (Nov. 20/21) that Kevin made to improve memory efficiency in the rr-graph generator and then to improve the router in that timeframe.  Mustafa has been making changes to make global clock routing possible, which involved some rr-graph changes, in December.  So by trying an early November and an early December version you'll help us eliminate or point at one of those changes as the culprit.

gsomlo, 2019-01-11:
Kevin,

On Fri, Jan 11, 2019 at 09:29:26AM -0800, kmurray wrote:
> I suspect I've found and fixed the root cause of this issue, which should be
> fixed as of ab5125d. (If it isn't please re-open this issue).

Thanks for the fix, this indeed takes care of my issue. Sorry I didn't
say anything sooner (or participate in the debugging) - I just managed
to dig myself half-way out of may winter-vacation inbox accumulation :)

Thanks again,
--Gabriel

> For details see ab5125d.