verilog-to-routing/vtr-verilog-to-routing #480: Net which drives LUT is removed if driven by .conn

lvezier, 2019-03-06:
All nets driven by a .conn (ex:` .conn reg_src wire0`) are removed by VPR with following messages:
Warning 3: Net 'wire0' has no driver and will be removed
Even if they feed a .names...

A workaround is to replace the . conn by a buffer using .names:
```
.names reg_src wire0
1 1 
```

#### Expected Behaviour

VPR should not remove net which fed .names

#### Current Behaviour

#### Possible Solution

#### Steps to Reproduce

1.  Extended blif content:
```
.model test
.inputs\
    fluclk_g\
    sig_in
.outputs\
    wire0_inv

.latch \
    sig_in\
    reg_src\
    re \
    fluclk_g\
    0

.conn reg_src wire0

.names wire0 wire0_inv
0 1

.end
```

2.  VPR command:

vpr flu.xml flu_core_content.vtr --device T222-1005 --circuit_file conn_issue.eblif --circuit_format eblif --timing_analysis on --route_chan_width 74 --cluster_seed_type timing --max_router_iterations 100 --constant_net_method global --const_gen_inference comb --sweep_constant_primary_outputs on --verbose_sweep on --disp off --place_algorithm path_timing_driven

#### Context

#### Your Environment

* VTR revision used:
Version: 8.0.0-dev+dc56def-dirty
Revision: dc56def-dirty
* Operating System and version: Linux RHEL 7.4
* Compiler version: GNU 5.3.1 on Linux-4.8.13-100.fc23.x86_64 x86_64

kmurray, 2019-03-06:
Does this result in a functional correctness issue, or is it just the warning which is a problem?

The way VPR handles .conn is to merge the two nets (e.g. reg_src and wire0) together into a single net internally. This probably means one of the nets is effectively empty and gets swept (resulting in the warning). 

(This is also effectively the same thing that happens with using a .names buffer with --absorb_buffer_luts on).

Could your re-run with the option `--echo_file on` and attach the generated `atom_netlist.orig.blif` and `atom_netlist.cleaned.blif` files? They will show the resulting netlist before and after all of VPR's netlist cleaning/modifications steps.

lvezier, 2019-03-07:
I am not just annoyed by warnings, VPR leads to a failing design bitstream due to these connection removal.

Please find attached echo files from VPR.
[conn_issue.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/2940379/conn_issue.zip)

kmurray, 2019-03-07:
Thanks for clarifying.

I've identified the issue. In your example the `.conn reg_src wire0` is declared before the `.names wire0 wire0_inv`. As a result VPR merges the `reg_src` and `wire0` nets *before* `wire0` has any sinks. As a result reg_src ends up with no sinks, and wire0 with no driver causing them to be swept as you saw.

The fix is to defer the net merging until all the sources and sinks have been declared.

commit a7bbcca2 (2019-03-07): vpr: Fix issue with eblif .conn statements delcared before net sinks

Previously VPR processed the .conn (merging the two associated nets) at
the point where the .conn was delcared.

This is incorrect in the case where the target net has not yet had all
it's sinks declared.

To handle such cases correctly we now record the pairs of nets which
need to be merged to resolve all .conn's and process them at the end of
the current .model. This ensures that all drivers/sinks have been associated
with their respective nets.

Closes #480
closed 2019-03-07 by kmurray