verilog-to-routing/vtr-verilog-to-routing #504: VPR duplicated branch warning in clock network construction code

kmurray, 2019-03-14:
#### Expected Behaviour
VPR clock construction code should compile with out warnings with gcc-8

#### Current Behaviour

With gcc-8 the following warning is produced 
```
/project/trees/vtr/vpr/src/route/clock_connection_builders.cpp: In member function ‘virtual void ClockToPinsConnection::create_switches(const ClockRRGraphBuilder&)’:
/project/trees/vtr/vpr/src/route/clock_connection_builders.cpp:252:28: warning: this condition has identical branches [-Wduplicated-branches]
                     } else if (y == grid.height()-1) {
```

You can see the relevant code [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/clock_connection_builders.cpp#L252,L256), which shows that both the else-if and else bodies have the same code.

#### Possible Solution

I'm not familiar with what this code is doing, so I don't know whether this was the intended behaviour or a copy-paste bug.

If the code is supposed to be duplicated then we could just remove the `else if (y == grid.height()-1)` to eliminate the warning.

#### Steps to Reproduce

```
$ CC=gcc-8 CXX=g++-8 make vpr
```
#### Your Environment

* VTR revision used: 3e14d80fb
* Operating System and version: Ubuntu 18.04
* Compiler version: gcc-8

kmurray, 2019-03-22:
@mustafabbas Was this was closed with  0ef9c1c?

mustafabbas, 2019-03-29:
> @mustafabbas Was this was closed with [0ef9c1c](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/0ef9c1c721a53b662f83d08502ba9daf0937c3ac)?

Yes this has been implemented.

closed 2019-03-29 by mustafabbas