verilog-to-routing/vtr-verilog-to-routing #521: 2-Stage Routing for Dedicated Routing Networks

kmurray, 2019-03-29:
#### Proposed Behaviour

For some signals (like clocks) there are dedicated routing networks available to distribute signals throughout a device.

To use such dedicated resources it would be beneficial to route in two stages:
1) From clock source (e.g. input pin/PLL) to clock network root
2) From clock network route to block pins

By marking the stub generated in (1) as not to be re-expended this ensures the dedicated network is used to reach block pins.

#### Current Behaviour

If asked to route such a signal, the router will attempt to route as usually. It may (or may not) find the dedicated routing network. If it does find the dedicated network the signal may end-up split between dedicated and regular routing.

#### Context

This is primarily motivated by dedicated clock networks, but also shows up with other dedicated routing resources (e.g. vcc/gnd as in #520).

kmurray, 2019-03-29:
Creating an issue on this topic since it is of interest to others (@litghost)

@vaughnbetz @mustafabbas Feel free to add any comments based on your discussions.

litghost, 2019-03-29:
This is timely, because @mithro and myself were discussing this problem also in relation to input pin to clock network root.

vaughnbetz, 2019-03-29:
Discussed with Mustafa today.  A few notes:

1. pre-route to clock driver nodes (from anywhere).  Mustafa will add a special "clock sink" that is attached to all the clock driver nodes (root of clock tree).  By routing to this sink we'll get the first part of the route.   
     Note: in the future we could have multiple sinks for multiple kinds of clock trees -- global vs. quadrant etc. -- but that is beyond Mustafa's work.  It should fit nicely into his framework though so shouldn't be too hard an extension.
2. In the route_tree, mark all the nodes except the clock driver node as "re_expand = false".  Now they won't go back on the heap and won't be branch points.

Mustafa to decide if the "virtual sink" representing the fact that you need to get to a clock driver goes into the route_tree / routing_traceback or not.  You can do what seems most logical and requires the fewest code updates.  I think Kevin believes not putting the sink in the route_tree / traceback is easier to understand and I tend to agree, so I lean towards that.

vaughnbetz, 2025-04-29:
This issue was completed.

closed 2025-04-29 by vaughnbetz