verilog-to-routing/vtr-verilog-to-routing #526: Router generating "stub" or "antenna" nets

litghost, 2019-04-02:
#### Expected Behaviour

Router only enables muxes on paths to sinks

#### Current Behaviour

Router generates some "stub" or "antenna" nets that do not connect to a sink.

#### Possible Solution

I haven't investigated too deeply, first step was to create an issue describing the problem.

#### Steps to Reproduce
1. Unzip [vpr_bug3.zip](https://ufile.io/a6jpf)
2. Run:
```
vpr \
 arch.unique_pack.xml \
 top.eblif \
 --device xc7a50t-basys3-test --read_rr_graph rr_graph_xc7a50t-basys3_test.rr_graph.real.xml \
 --min_route_chan_width_hint 100 --max_criticality 0.0 --max_router_iterations 500 --routing_failure_predictor off --router_high_fanout_threshold -1 \
 --constant_net_method route --route_chan_width 500 --clock_modeling route --place_algorithm bounding_box --enable_timing_computations off \
 --allow_unrelated_clustering on --route
```

#### Analysis

Example stub is at rr node 433885.  Analysis that shows its a stub:

```
# Pretty print rr graph for ease of grepping
$ xmllint --format rr_graph_xc7a50t-basys3_test.rr_graph.real.xml > t.xml
# Relevant node
$ grep -C 3 INT_L_X14Y124.SE6BEG1.LOGIC_OUTS_L23 t.xml
    </edge>
    <edge sink_node="433885" src_node="421075" switch_id="2">
      <metadata>
        <meta name="fasm_features">INT_L_X14Y124.SE6BEG1.LOGIC_OUTS_L23</meta>
      </metadata>
    </edge>
    <edge sink_node="1094822" src_node="421075" switch_id="2">
# Find node in route
$ grep -n 433885 top.route
2574:Node:	433885	 CHANX (40,27) to (44,27)  Track: 110  Switch: 3
# Walk graph following node 433885 to sinks.  switch_id="3" are shorts
$ grep src_node=\"433885 t.xml
    <edge sink_node="433884" src_node="433885" switch_id="3"/>
$ grep src_node=\"433884 t.xml
    <edge sink_node="433886" src_node="433884" switch_id="3"/>
    <edge sink_node="433885" src_node="433884" switch_id="3"/>
# So 433884, 433885 and 433886 are shorted to each other
# 433886 contains the edges to other nodes in the graph
$ grep src_node=\"433886 t.xml
    <edge sink_node="433884" src_node="433886" switch_id="3"/>
    <edge sink_node="636697" src_node="433886" switch_id="2">
    <edge sink_node="636705" src_node="433886" switch_id="2">
    <edge sink_node="1101336" src_node="433886" switch_id="2">
    <edge sink_node="1101342" src_node="433886" switch_id="2">
    <edge sink_node="1101386" src_node="433886" switch_id="2">
    <edge sink_node="637940" src_node="433886" switch_id="2">
    <edge sink_node="1038942" src_node="433886" switch_id="2">
    <edge sink_node="1101401" src_node="433886" switch_id="2">
    <edge sink_node="1101280" src_node="433886" switch_id="2">
    <edge sink_node="635773" src_node="433886" switch_id="2">
    <edge sink_node="1101287" src_node="433886" switch_id="2">
    <edge sink_node="1101156" src_node="433886" switch_id="2">
    <edge sink_node="1100588" src_node="433886" switch_id="2">
    <edge sink_node="433614" src_node="433886" switch_id="2">
    <edge sink_node="433906" src_node="433886" switch_id="2">
    <edge sink_node="433927" src_node="433886" switch_id="2">
# Grep each downstream node and see if it part of the same net
$ grep -n 636697 top.route 
$ grep -n 636705 top.route 
11852:Node:	636705	 CHANX (44,31) to (53,31)  Track: 33  Switch: 2
$ grep -n 1101336 top.route 
$ grep -n 1101342 top.route 
11777:Node:	1101342	 CHANX (44,31) to (45,31)  Track: 206  Switch: 2
$ grep 1101386 top.route 
Node:	1101386	 CHANY (44,30) to (44,31)  Track: 237  Switch: 2
Node:	1101386	 CHANY (44,30) to (44,31)  Track: 237  Switch: 3
$ grep -n 1101386 top.route 
11728:Node:	1101386	 CHANY (44,30) to (44,31)  Track: 237  Switch: 2
11734:Node:	1101386	 CHANY (44,30) to (44,31)  Track: 237  Switch: 3
$ grep -n 637940 top.route 
$ grep -n 1038942 top.route 
$ grep -n 1101401 top.route 
$ grep -n 1101280 top.route 
11595:Node:	1101280	 CHANY (44,31) to (44,32)  Track: 254  Switch: 2
11602:Node:	1101280	 CHANY (44,31) to (44,32)  Track: 254  Switch: 3
$ grep -n 635773 top.route 
$ grep -n 1101287 top.route 
$ grep -n 1101156 top.route 
$ grep -n 1100588 top.route 
$ grep -n 433614 top.route 
$ grep -n 433906 top.route 
$ grep -n 433927 top.route 
```

All downstream nodes are members of different nodes, therefore shorted node group (433884, 433885, 433886) is an stub or antenna.

#### Your Environment
VTR built from https://github.com/SymbiFlow/vtr-verilog-to-routing/commit/d8c2e526c3bbc09172ca5b6f3bb2e3cf9524c20b

kmurray, 2019-04-08:
Looks like the files you attached don't include the .net/.place/.route files. Can you attach them as well?

litghost, 2019-04-10:
Sure, attached: 
[outputs.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3066171/outputs.zip)

litghost, 2019-06-06:
@renrichard This is the route stub issue, this PR in theory should detect this error, and I believe it is failing on the CI.

https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/535

litghost, 2019-09-18:
@renrichard Did root cause for this issue get identified?  I'm planning on working on this issue, as it is affecting my work.

litghost, 2019-09-18:
@kmurray: I'm looking into this issue, and see something "smells bad"

The ``t_heap`` data has a linked list from the leaf node back to the source node:

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/91384efd78e7e1bca599cb7b667e88acac12dc87/vpr/src/route/route_common.h#L28-L29

Which is updated during expansion here:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/91384efd78e7e1bca599cb7b667e88acac12dc87/vpr/src/route/route_timing.cpp#L1939-L1942

and yet the traceback update doesn't use this data, see here:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/91384efd78e7e1bca599cb7b667e88acac12dc87/vpr/src/route/route_common.cpp#L593-L594

If the `route_ctx.rr_node_route_inf[inode].prev_edge` / `route_ctx.rr_node_route_inf[inode].prev_node` were ever out of sync with the `t_heap` `u.prev` data it might explain this bug.

The reason I say this code smells, is why have the data in two places?  Which is the "true" path?  From my perspective, the `t_heap` linked list is the "true" data, and so `traceback_branch` oaught to follow the `t_heap` linked list.

kmurray, 2019-09-18:
@litghost I think we do actually need both pieces of information.

* The values in `route_ctx.rr_node_route_inf`correspond to the actual *best* path (so far) to each a node found by the router. Once we've found the target, the 'best so far' is the actual 'best' path found. That is why it is used by the traceback routines to figure out what the actual routing path is.

* In contrast, each t_heap corresponds to a partial (potentially not 'best') path to the node it is associated with. That is, there may be multiple t_heap elements for any particular RR node, corresponding to different ways to reach that node (e.g. via different incoming edges). 

I think key bit you are missing is:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/91384efd78e7e1bca599cb7b667e88acac12dc87/vpr/src/route/route_timing.cpp#L1469-L1472
where we update the 'best' path in `route_ctx.rr_node_route_inf` from the lowest cost element just popped of the heap, if the cost is lower. This maintains the key invariant that  `route_ctx.rr_node_route_inf` stores the 'best' path to all explored nodes.

Furthermore, the t_heap doesn't really form a linked list back to the source. Each t_heap only stores a single step back (prev node and prev edge). So you can't actually trace things back via the t_heap. (Which is one of the reasons for `route_ctx.rr_node_route_inf`to store the 'best' way to reach each node from the source, allowing us to traceback the actual path). 

For more details on the path finding invariants etc, you could checkout something like CLRS's chapter on shortest paths.

litghost, 2019-09-18:
@kmurray : Agreed, I realized my mistake a little bit later.

I think I have found the answer, but I'd like a sanity check.  It looks like `update_traceback` just appends a traceback, but doesn't cleanup old traces for the same net. I added some debug prints to the `update_traceback`, and I see the following:

The first time the net is routed it uses inodes `610147`/`610148`:
```
Net: 461 sink: 288287
Previous node: 288356, via edge: 0
Previous node: 1041409, via edge: 0
Previous node: 638850, via edge: 0
Previous node: 967652, via edge: 9
Previous node: 967644, via edge: 1
Previous node: 621862, via edge: 0
Previous node: 610147, via edge: 1
Previous node: 610148, via edge: 0
Previous node: 610130, via edge: 13
```

Later that connect is ripped up, and the new path is:
```
Net: 461 sink: 288287
Previous node: 288356, via edge: 0
Previous node: 1041409, via edge: 0
Previous node: 638850, via edge: 0
Previous node: 967652, via edge: 9
Previous node: 967644, via edge: 1
Previous node: 967539, via edge: 1
Previous node: 610482, via edge: 7
Previous node: 960151, via edge: 3
Previous node: 610130, via edge: 15
```

No other nets use inodes `610147`/`610148`, and yet their connection to inode `610130` is preserved:
```
Node:	960155	 CHANY (58,17) to (58,23)  Track: 111  Switch: 145
Node:	960592	 CHANY (58,16) to (58,17)  Track: 143  Switch: 130
Node:	960645	 CHANY (58,16)  Track: 219  Switch: 128
Node:	960649	 CHANY (58,16)  Track: 215  Switch: 131
Node:	611598	 CHANY (58,16)  Track: 257  Switch: 7
Node:	1032999	 CHANY (58,16)  Track: 180  Switch: 204
Node:	236638	  IPIN (59,16)  Pin: 23   BLK-TL-SLICEL.D1[0] Switch: 0
Node:	236547	  SINK (59,16)  Class: 23  Switch: -1
Node:	610130	 CHANX (58,23) to (60,23)  Track: 202  Switch: 144
Node:	610148	 CHANX (58,22) to (64,22)  Track: 122  Switch: 3
Node:	610147	 CHANY (63,19) to (63,23)  Track: 298  Switch: -1
Node:	610130	 CHANX (58,23) to (60,23)  Track: 202  Switch: 155
Node:	492032	 CHANY (57,19) to (57,23)  Track: 259  Switch: 3
Node:	492033	 CHANX (52,19) to (58,19)  Track: 64  Switch: 150
Node:	956878	 CHANY (52,13) to (52,19)  Track: 77  Switch: 145
Node:	957332	 CHANY (52,12) to (52,13)  Track: 140  Switch: 131
Node:	603369	 CHANY (52,12)  Track: 259  Switch: 7
Node:	1029040	 CHANY (52,12)  Track: 182  Switch: 206
Node:	200740	  IPIN (53,12)  Pin: 25   BLK-TL-SLICEL.D3[0] Switch: 0
Node:	200645	  SINK (53,12)  Class: 25  Switch: -1
```

This seems to be a cleanup issue around how the traceback is updated? Am I missing something?

litghost, 2019-09-18:
VPR stdout will debug printing, and route file:
[holes.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3628666/holes.tar.gz)

litghost, 2019-09-18:
I think I found the bug?

```
          //      Since the node was reached non-configurably it is illegal to   
          //      remove it independently of any other nodes which are non-      
          //      configurably connected to it.                                      
          //                                                                         
          //      As a result we only remove a non-configurably connected node                                                                                                                                                                                                            
          //      if force_prune is set.            
```

kmurray, 2019-09-19:
> I think I found the bug?
>
> ```
>           //      Since the node was reached non-configurably it is illegal to
>           //      remove it independently of any other nodes which are [...]

If I recall correctly, the idea is that we can't prune part of a set of non-configurably connected nodes. If any node is used in a non-configurably connected set then *all* the nodes must be used. (This is the fundamental definition/invariant imposed by non-configurable edges).

As I mentioned in a bug somewhere previously, the only case I would expect there to be routing stubs is the case of non-configurably connected nodes. So if that is the case then these stubs are expected.

litghost, 2019-09-19:
> If I recall correctly, the idea is that we can't prune part of a set of non-configurably connected nodes. If any node is used in a non-configurably connected set then _all_ the nodes must be used. (This is the fundamental definition/invariant imposed by non-configurable edges).

Sure, but if the collection is tracking (e.g. https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/969), then pruning is possible.

litghost, 2019-09-19:
> As I mentioned in a bug somewhere previously, the only case I would expect there to be routing stubs is the case of non-configurably connected nodes. So if that is the case then these stubs are expected.

It isn't mentioned in this issue.  Regardless, if something like #969 is no good, shouldn't VPR do a cleanup after routing is complete?

kmurray, 2019-09-19:
> Sure, but if the usage of the non-configurable set tracked (e.g. #969), then pruning is possible.

I guess my point is that it is illegal to prune nodes which are part of a non-configurably connected node set. Non-configurable edges form sets of nodes which *must* be used together. Pruning nodes from the set is assuming that such edges are *configurable* which is not the case. 

Non-configurable edges are used to model logical connections which [can not be turned off](https://docs.verilogtorouting.org/en/latest/arch/reference/#switches), like electrical shorts and fixed buffers, and hence can't be pruned.

litghost, 2019-09-19:
> Non-configurable edges are used to model logical connections which [can not be turned off](https://docs.verilogtorouting.org/en/latest/arch/reference/#switches), like electrical shorts and fixed buffers, and hence can't be pruned.

But if there is no outbound connection from the group, the whole group _should_ be pruned.

PR #969 (2019-10-25): Track non-configurable edge usage to enable pruning.
#### Description

The previous pruning code left non-configurable sets unpruned.  The new code tracks outbound configurable edges from non-configurable sets, and prunes them when the count reaches zero.

Fixes #526

#### Related Issue

- #526

#### Motivation and Context

#### How Has This Been Tested?

 - [x] Check route has been updated to detect stubs
 - [x] Symbiflow designs that were failing with the new check route now do not error
 - [ ] CI is green

#### Types of changes

- [x] Bug fix (change which fixes an issue)
- [x] New feature  [... cut]
commit 7573a0fb (2019-10-25): Track non-configurable edge usage to enable pruning.

Fixes #526
commit 98113529 (2019-11-25): Track non-configurable edge usage to enable pruning.

Fixes #526
closed 2019-10-25 by kmurray