verilog-to-routing/vtr-verilog-to-routing #530: [21/1067] Linking CXX executable vpr/test_vpr FAILED: vpr/test_vpr

jeanlego, 2019-04-05:
It seems that test_vpr cannot compile on archlinux (gcc 8.2.1 and clang 7)

if i comment out the following lines, 
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/CMakeLists.txt#L113-L116

```
add_test(NAME test_vpr
    COMMAND test_vpr --use-colour=yes
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}/test
    )
```

the code proceed to compile properly

## error
```
[21/1067] Linking CXX executable vpr/test_vpr
FAILED: vpr/test_vpr 
: && /usr/bin/c++  -std=c++14  -Wall -Wextra -Wpedantic -Wcast-qual -Wcast-align -Wshadow -Wformat=2 -Wno-format-nonliteral -Wlogical-op -Wmissing-declarations -Wmissing-include-dirs -Wredundant-decls -Wswitch-default -Wundef -Wunused -Wunused-variable -Wunused-parameter -Wdisabled-optimization -Wnoexcept -Woverloaded-virtual -Wctor-dtor-privacy -Wnon-virtual-dtor -Wduplicated-cond -Wduplicated-branches -Wnull-dereference -Wuninitialized -Winit-self -Wcatch-value=3 -Wextra-semi      -O3 -DNDEBUG   vpr/CMakeFiles/test_vpr.dir/test/main.cpp.o vpr/CMakeFiles/test_vpr.dir/test/test_vpr.cpp.o  -o vpr/test_vpr  libs/EXTERNAL/libcatch/liblibcatch.a ../vpr/libvpr.a ../libs/libarchfpga/libarchfpga.a ../libs/libvtrutil/libvtrutil.a ../libs/liblog/liblog.a ../libs/libpugiutil/libpugiutil.a ../libs/EXTERNAL/libsdcparse/libsdcparse.a ../libs/EXTERNAL/libblifparse/libblifparse.a ../libs/libeasygl/libeasygl.a ../libs/EXTERNAL/libtatum/libtatum.a /usr/lib/libtbb.so -ltbbmalloc_proxy libs/EXTERNAL/libargparse/libargparse.a ../libs/EXTERNAL/libpugixml/libpugixml.a && :
lto1: internal compiler error: in add_symbol_to_partition_1, at lto/lto-partition.c:155
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.archlinux.org/> for instructions.
lto-wrapper: fatal error: /usr/bin/c++ returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
```
## uname -a
```
Linux jp-arch 5.0.5-arch1-1-ARCH #1 SMP PREEMPT Wed Mar 27 17:53:10 UTC 2019 x86_64 GNU/Linux
```
## gcc -v
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/8.2.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-cet=auto
Thread model: posix
gcc version 8.2.1 20181127 (GCC) 
```

ld -v
```
GNU ld (GNU Binutils) 2.31.1
```

kmurray, 2019-04-08:
Looks like this a bug in the compiler (gcc-8) related to link-time optimization. So I'm not sure this is really a VTR specific issue. (For me everything works fine with gcc 7).

My recommended work around would be to set the cmake option `VPR_IPO_BUILD` to `off` which will disable link-time optimization.

jeanlego, 2019-04-16:
thanks! that fixed it :)

closed 2019-04-16 by jeanlego